FreeCAD/FreeCAD #18537: Model crashes when opened with 1.1.0dev

furgo16, 2024-12-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I created this quick model with a pre-1.0 development version, but I've recently opened and saved it with FreeCAD 1.0.

[Beam joint barn 3.zip](https://github.com/user-attachments/files/18142340/Beam.joint.barn.3.zip)

When opening the file, FreeCAD crashes with the following stacktrace:

<details><summary>Stacktrace</summary>

```
$ freecad
FreeCAD 1.1.0, Libs: 1.1.0devR39649 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Adding snap-specific PYTHONPATH to sys.path: /home/dpm/snap/freecad/common/.local/lib/python3.10/site-packages:/snap/freecad/1226/lib/python3.10/site-packages:/snap/freecad/1226/usr/lib/python3/dist-packages
Qt: Session management error: Could not open network socket
connect failed: No such file or directory
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7ff1ed123520]
#1  0x7ff190dadbe7 in Part::Feature::getExportElementName(Part::TopoShape, char const*) const from /snap/freecad/1226/usr/lib/Part.so+0x1657
#2  0x7ff190daedf5 in Part::Feature::getElementName(char const*, App::GeoFeature::ElementNameType) const from /snap/freecad/1226/usr/lib/Part.so+0xa5
#3  0x7ff1ef5f19f8 in App::GeoFeature::resolveElement(App::DocumentObject const*, char const*, App::ElementNamePair&, bool, App::GeoFeature::ElementNameType, App::DocumentObject const*, char const**, App::GeoFeature**) from /snap/freecad/1226/usr/lib/libFreeCADApp.so+0x428
#4  0x7ff1ef6cc2fb in App::PropertyLinkBase::_registerElementReference(App::DocumentObject*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, App::ElementNamePair&) from /snap/freecad/1226/usr/lib/libFreeCADApp.so+0xdb
#5  0x7ff1ef6cc9df in App::PropertyLinkSubList::setValues(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> >&&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&&, std::vector<App::ElementNamePair, std::allocator<App::ElementNamePair> >&&) from /snap/freecad/1226/usr/lib/libFreeCADApp.so+0x33f
#6  0x7ff1ef6ccaec in App::PropertyLinkSubList::setValues(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<App::ElementNamePair, std::allocator<App::ElementNamePair> >&&) from /snap/freecad/1226/usr/lib/libFreeCADApp.so+0xcc
#7  0x7ff1ef6d0cf0 in App::PropertyLinkSubList::Restore(Base::XMLReader&) from /snap/freecad/1226/usr/lib/libFreeCADApp.so+0x450
#8  0x7ff1ef686d86 in App::PropertyContainer::Restore(Base::XMLReader&) from /snap/freecad/1226/usr/lib/libFreeCADApp.so+0x1826
#9  0x7ff1ef4d18c5 in App::Document::readObjects(Base::XMLReader&) from /snap/freecad/1226/usr/lib/libFreeCADApp.so+0x765
#10  0x7ff1ef4dae08 in App::Document::Restore(Base::XMLReader&) from /snap/freecad/1226/usr/lib/libFreeCADApp.so+0x2c8
#11  0x7ff1ef4dd102 in App::Document::restore(char const*, bool, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) from /snap/freecad/1226/usr/lib/libFreeCADApp.so+0x622
#12  0x7ff1ef74d0b0 in App::Application::openDocumentPrivate(char const*, char const*, char const*, bool, bool, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&&) from /snap/freecad/1226/usr/lib/libFreeCADApp.so+0x350
#13  0x7ff1ef750ba0 in App::Application::openDocuments(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >*, bool) from /snap/freecad/1226/usr/lib/libFreeCADApp.so+0x820
#14  0x7ff1ef752a01 in App::Application::openDocument(char const*, bool) from /snap/freecad/1226/usr/lib/libFreeCADApp.so+0xf1
#15  0x7ff1ef76ec3b in App::Application::sOpenDocument(_object*, _object*, _object*) from /snap/freecad/1226/usr/lib/libFreeCADApp.so+0xfb
#16  /snap/freecad/1226/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x128023) [0x7ff1eeac7023]
#17  /snap/freecad/1226/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyObject_MakeTpCall+0x8c) [0x7ff1eea7eadc]
#18  /snap/freecad/1226/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9dbc) [0x7ff1eea1aa1c]
#19  /snap/freecad/1226/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7ff1eeb613af]
#20  /snap/freecad/1226/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyEval_EvalCode+0xbe) [0x7ff1eeb5c3de]
#21  /snap/freecad/1226/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x20c4cd) [0x7ff1eebab4cd]
#22  /snap/freecad/1226/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyRun_StringFlags+0x79) [0x7ff1eebac4f9]
#23  0x7ff1ef0d124c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /snap/freecad/1226/usr/lib/libFreeCADBase.so+0x6c
#24  0x7ff1f0045c7c in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /snap/freecad/1226/usr/lib/libFreeCADGui.so+0x8c
#25  0x7ff1f0045e07 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /snap/freecad/1226/usr/lib/libFreeCADGui.so+0xe7
#26  0x7ff1eff19c27 in Gui::Application::open(char const*, char const*) from /snap/freecad/1226/usr/lib/libFreeCADGui.so+0x4c7
#27  0x7ff1f0059d60 in StdCmdOpen::activated(int) from /snap/freecad/1226/usr/lib/libFreeCADGui.so+0x6f0
#28  0x7ff1f004c566 in Gui::Command::_invoke(int, bool) from /snap/freecad/1226/usr/lib/libFreeCADGui.so+0x246
#29  0x7ff1f004ca26 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /snap/freecad/1226/usr/lib/libFreeCADGui.so+0x126
#30  /snap/freecad/1226/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2f48f4) [0x7ff1ed8748f4]
#31  0x7ff1ee3cfc26 in QAction::triggered(bool) from /snap/freecad/1226/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x46
#32  0x7ff1ee3d291c in QAction::activate(QAction::ActionEvent) from /snap/freecad/1226/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xac
#33  /snap/freecad/1226/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x2f4d42) [0x7ff1ee55ed42]
#34  /snap/freecad/1226/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x2fc919) [0x7ff1ee566919]
#35  0x7ff1ee41994e in QWidget::event(QEvent*) from /snap/freecad/1226/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x20e
#36  0x7ff1ee3d6763 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /snap/freecad/1226/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#37  0x7ff1ee3de3a4 in QApplication::notify(QObject*, QEvent*) from /snap/freecad/1226/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xa84
#38  0x7ff1effe6200 in Gui::GUIApplication::notify(QObject*, QEvent*) from /snap/freecad/1226/usr/lib/libFreeCADGui.so+0xd0
#39  0x7ff1ed83ce4a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /snap/freecad/1226/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#40  0x7ff1ee3dce87 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /snap/freecad/1226/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1b7
#41  /snap/freecad/1226/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1c9d2b) [0x7ff1ee433d2b]
[... cut ...]
</details>

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39649 (Git) Snap 1226
Build type: Release
Branch: main
Hash: e458526279e3cf440a8f14f2424e707983de2fc1
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-12-15:
Confirmed
```
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3d1d0) [0x7fc23f24c1d0]
#1  0x7fc1fd8cfe45 in Part::Feature::getExportElementName(Part::TopoShape, char const*) const from /tmp/.mount_FreeCANGKpEP/usr/lib/./Part.so+0x1585
#2  0x7fc1fd8d18e6 in Part::Feature::getElementName(char const*, App::GeoFeature::ElementNameType) const from /tmp/.mount_FreeCANGKpEP/usr/lib/./Part.so+0xa6
#3  0x7fc2413218f7 in App::GeoFeature::resolveElement(App::DocumentObject const*, char const*, App::ElementNamePair&, bool, App::GeoFeature::ElementNameType, App::DocumentObject const*, char const**, App::GeoFeature**) from /tmp/.mount_FreeCANGKpEP/usr/bin/../lib/libFreeCADApp.so+0x4a7
#4  0x7fc2413eec7e in App::PropertyLinkBase::_registerElementReference(App::DocumentObject*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, App::ElementNamePair&) from /tmp/.mount_FreeCANGKpEP/usr/bin/../lib/libFreeCADApp.so+0xde
#5  0x7fc2413ef37f in App::PropertyLinkSubList::setValues(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> >&&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&&, std::vector<App::ElementNamePair, std::allocator<App::ElementNamePair> >&&) from /tmp/.mount_FreeCANGKpEP/usr/bin/../lib/libFreeCADApp.so+0x33f
#6  0x7fc2413ef4ec in App::PropertyLinkSubList::setValues(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<App::ElementNamePair, std::allocator<App::ElementNamePair> >&&) from /tmp/.mount_FreeCANGKpEP/usr/bin/../lib/libFreeCADApp.so+0x12c
#7  0x7fc2413f3124 in App::PropertyLinkSubList::Restore(Base::XMLReader&) from /tmp/.mount_FreeCANGKpEP/usr/bin/../lib/libFreeCADApp.so+0x4e4
#8  0x7fc2413a6840 in App::PropertyContainer::Restore(Base::XMLReader&) from /tmp/.mount_FreeCANGKpEP/usr/bin/../lib/libFreeCADApp.so+0x1730
#9  0x7fc2412512b7 in App::Document::readObjects(Base::XMLReader&) from /tmp/.mount_FreeCANGKpEP/usr/bin/../lib/libFreeCADApp.so+0x7f7
#10  0x7fc241262064 in App::Document::Restore(Base::XMLReader&) from /tmp/.mount_FreeCANGKpEP/usr/bin/../lib/libFreeCADApp.so+0x2e4
#11  0x7fc24126c473 in App::Document::restore(char const*, bool, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) from /tmp/.mount_FreeCANGKpEP/usr/bin/../lib/libFreeCADApp.so+0x633
#12  0x7fc241464add in App::Application::openDocumentPrivate(char const*, char const*, char const*, bool, bool, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&&) from /tmp/.mount_FreeCANGKpEP/usr/bin/../lib/libFreeCADApp.so+0x31d
#13  0x7fc2414675d1 in App::Application::openDocuments(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >*, bool) from /tmp/.mount_FreeCANGKpEP/usr/bin/../lib/libFreeCADApp.so+0x7a1
#14  0x7fc241469488 in App::Application::openDocument(char const*, bool) from /tmp/.mount_FreeCANGKpEP/usr/bin/../lib/libFreeCADApp.so+0xc8
#15  0x7fc241474d86 in App::Application::sOpenDocument(_object*, _object*, _object*) from /tmp/.mount_FreeCANGKpEP/usr/bin/../lib/libFreeCADApp.so+0x106
#16  /tmp/.mount_FreeCANGKpEP/usr/bin/../lib/libpython3.11.so.1.0(+0x20603d) [0x7fc241a0603d]
#17  /tmp/.mount_FreeCANGKpEP/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x7fc2419e5b5c]
#18  /tmp/.mount_FreeCANGKpEP/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7fc2419f1532]
#19  /tmp/.mount_FreeCANGKpEP/usr/bin/../lib/libpython3.11.so.1.0(+0x2abd1d) [0x7fc241aabd1d]
#20  /tmp/.mount_FreeCANGKpEP/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7fc241aab794]
#21  /tmp/.mount_FreeCANGKpEP/usr/bin/../lib/libpython3.11.so.1.0(+0x2cabf3) [0x7fc241acabf3]
#22  /tmp/.mount_FreeCANGKpEP/usr/bin/../lib/libpython3.11.so.1.0(+0x2c669a) [0x7fc241ac669a]
#23  /tmp/.mount_FreeCANGKpEP/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x7fc241aba9c9]
#24  0x7fc240d473dc in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCANGKpEP/usr/bin/../lib/libFreeCADBase.so+0x6c
[... cut ...]
#55  /usr/lib/libc.so.6(+0x25e08) [0x7fc23f234e08]
#56  /usr/lib/libc.so.6(__libc_start_main+0x8c) [0x7fc23f234ecc]
#57  /tmp/.mount_FreeCANGKpEP/usr/bin/freecad(+0x7c2a) [0x560c4dae0c2a]
```

```
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39642 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: c80bb958f377e4d43c6a4075da173a79d153db68
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

tarman3, 2024-12-16:
FreeCAD 1.0.0.39109 - **Open**
FreeCAD 1.1.0dev.39560 - **Crash**

Ventusfahrer, 2024-12-16:
confirmed. 

revision 39416 is still loading the related project without problems.

The problem starts to happen with revision 39560 and continues with the latest:

OS: Ubuntu 24.10 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.39642 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: c80bb958f377e4d43c6a4075da173a79d153db68
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * fasteners 0.5.29

Interesting detail: if I remove in revision 39416 six Objects created with the fasteners workbench, the project did load fine in 39560 and 39462.  I can add those fasteners using 39642, save and open the file again without crashing.

Please give me a short feedback when you need the FreeCAD project file (5.6 MB).

<details>

<summary>stacktrace</summary>

```Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45250) [0x751983c45250]
#1  0x751953acfe45 in Part::Feature::getExportElementName(Part::TopoShape, char const*) const from /tmp/.mount_FreeCAHnoPek/usr/lib/./Part.so+0x1585
#2  0x751953ad18e6 in Part::Feature::getElementName(char const*, App::GeoFeature::ElementNameType) const from /tmp/.mount_FreeCAHnoPek/usr/lib/./Part.so+0xa6
#3  0x751985f218f7 in App::GeoFeature::resolveElement(App::DocumentObject const*, char const*, App::ElementNamePair&, bool, App::GeoFeature::ElementNameType, App::DocumentObject const*, char const**, App::GeoFeature**) from /tmp/.mount_FreeCAHnoPek/usr/bin/../lib/libFreeCADApp.so+0x4a7
#4  0x751985feec7e in App::PropertyLinkBase::_registerElementReference(App::DocumentObject*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, App::ElementNamePair&) from /tmp/.mount_FreeCAHnoPek/usr/bin/../lib/libFreeCADApp.so+0xde
#5  0x751985ff77db in App::PropertyXLink::setSubValues(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&&, std::vector<App::ElementNamePair, std::allocator<App::ElementNamePair> >&&) from /tmp/.mount_FreeCAHnoPek/usr/bin/../lib/libFreeCADApp.so+0x2cb
#6  0x751985ff7e90 in App::PropertyXLink::setValue(App::DocumentObject*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&&, std::vector<App::ElementNamePair, std::allocator<App::ElementNamePair> >&&) from /tmp/.mount_FreeCAHnoPek/usr/bin/../lib/libFreeCADApp.so+0x3f0
#7  0x751985ffb990 in App::PropertyXLink::Restore(Base::XMLReader&) from /tmp/.mount_FreeCAHnoPek/usr/bin/../lib/libFreeCADApp.so+0x410
#8  0x751985fa6840 in App::PropertyContainer::Restore(Base::XMLReader&) from /tmp/.mount_FreeCAHnoPek/usr/bin/../lib/libFreeCADApp.so+0x1730
#9  0x751985e512b7 in App::Document::readObjects(Base::XMLReader&) from /tmp/.mount_FreeCAHnoPek/usr/bin/../lib/libFreeCADApp.so+0x7f7
#10  0x751985e62064 in App::Document::Restore(Base::XMLReader&) from /tmp/.mount_FreeCAHnoPek/usr/bin/../lib/libFreeCADApp.so+0x2e4
#11  0x751985e6c473 in App::Document::restore(char const*, bool, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) from /tmp/.mount_FreeCAHnoPek/usr/bin/../lib/libFreeCADApp.so+0x633
#12  0x751986064add in App::Application::openDocumentPrivate(char const*, char const*, char const*, bool, bool, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&&) from /tmp/.mount_FreeCAHnoPek/usr/bin/../lib/libFreeCADApp.so+0x31d
#13  0x7519860675d1 in App::Application::openDocuments(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >*, bool) from /tmp/.mount_FreeCAHnoPek/usr/bin/../lib/libFreeCADApp.so+0x7a1
#14  0x751986069488 in App::Application::openDocument(char const*, bool) from /tmp/.mount_FreeCAHnoPek/usr/bin/../lib/libFreeCADApp.so+0xc8
#15  0x751986074d86 in App::Application::sOpenDocument(_object*, _object*, _object*) from /tmp/.mount_FreeCAHnoPek/usr/bin/../lib/libFreeCADApp.so+0x106
#16  /tmp/.mount_FreeCAHnoPek/usr/bin/../lib/libpython3.11.so.1.0(+0x20603d) [0x75198660603d]
#17  /tmp/.mount_FreeCAHnoPek/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x7519865e5b5c]
#18  /tmp/.mount_FreeCAHnoPek/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7519865f1532]
[... cut ...]
#55  0x7519842bd212 in QCoreApplication::exec() from /tmp/.mount_FreeCAHnoPek/usr/bin/../lib/libQt5Core.so.5+0x82
#56  0x751986ecdb3d in Gui::Application::runApplication() from /tmp/.mount_FreeCAHnoPek/usr/bin/../lib/libFreeCADGui.so+0x89d
#57  /tmp/.mount_FreeCAHnoPek/usr/bin/freecad(+0x78b7) [0x5eaad0cc48b7]
#58  /lib/x86_64-linux-gnu/libc.so.6(+0x2a3b8) [0x751983c2a3b8]
#59  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x751983c2a47b]
#60  /tmp/.mount_FreeCAHnoPek/usr/bin/freecad(+0x7c2a) [0x5eaad0cc4c2a]
```
</details>

furgo16, 2024-12-16:
> Please give me a short feedback when you need the FreeCAD project file (5.6 MB).

If this is directed to me, I don't specifically need the file. I can also open it with 1.0.0, but I appreciate the offer.

wwmayer, 2024-12-16:
Crash must be caused by this https://github.com/FreeCAD/FreeCAD/pull/16803
In line 368 `while (searchShape.getShape().ShapeType() < TopAbs_FACE )` a null shape is accessed and calling ShapeType() on it leads to a segfault.

PR #18548 (2024-12-16): Part: Fix segmentation fault in Feature::getExportElementName
Calling ShapeType() on a null shape is not allowed and causes a segmentation fault. So, check beforehand if the shape is not null.

Fixes #18537
commit bbb5b451 (2024-12-16): Part: Fix segmentation fault in Feature::getExportElementName

Calling ShapeType() on a null shape is not allowed and causes a segmentation fault. So, check beforehand if the shape is not null.

Fixes #18537
closed 2024-12-16 by chennes