FreeCAD/FreeCAD #18542: Sketcher in combination with Spreadsheet "scambles" object dimensions

malei255, 2024-12-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Please excuse my poor English since I am not a native speaker.

I found a strange behavior in FreeCAD 1.0 when using a spreadsheet to update a sketch. The demonstration file is attached to this report:  

Simple example using the following spreadsheet:
[image]

The result in the sketch is this (which is OK):
[image]

The position of the two objects is defined by the formulas:
[image]
and
[image]

The part looks like this -> perfect!
[image]

BUT

If you change the multiplier cell in the spreadsheet from 2 to e.g. 8
[image]

and reopen the sketch then the positions and dimensions are mixed up and out of order:
[image]

Point 1 and Point 2 (both supposed to be next to the center line) suddenly have moved to the other side and the sketch is messed up because the positions of the two points are no longer as intended.

Hope my explanation is understandable but feel free to get back to me in case you need additional input.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-12-16:
This is a classic case of what is known as 'flipping'. A sketch that is fully constrained can still have multiple solutions.

To mitigate the issue try one these options.

#### Switch to the LevenbergMarquardt solver:

1. In the [Sketcher preferences](https://wiki.freecad.org/Sketcher_Preferences) check the `Show section 'Advanced solver control'` option.
2. Restart FreeCAD.
3. Open the sketch in edit mode.
4. In the Advanced solver control section of the [Sketcher Dialog](https://wiki.freecad.org/Sketcher_Dialog) change the default solver to `LevenbergMarquardt`.

#### Use angular constraints:

1. Instead of using [horizontal](https://wiki.freecad.org/Sketcher_ConstrainHorizontal) and [vertical](https://wiki.freecad.org/Sketcher_ConstrainVertical) constraints use [angular](https://wiki.freecad.org/Sketcher_ConstrainAngle) constraints.

maxwxyz, 2024-12-16:
closing this then.

closed 2024-12-16 by maxwxyz

malei255, 2024-12-17:
@Roy-043 
Thank you very much for your explanation (and sorry for my late feedback - I've been traveling).
I tried your proposed solution and changed the sketcher to the LevenbergMarquardt solver. 
Now it is working as intended.
Are there other potential (positive or negative) side effects by switching to the LevenbergMarquardt solver?

Roy-043, 2024-12-17:
> Are there other potential (positive or negative) side effects by switching to the LevenbergMarquardt solver?

I do not know the answer to that question, suggest you start a topic in the [forum](https://forum.freecad.org/).