FreeCAD/FreeCAD #18545: Py_UNICODE and Python 3.13

Flast, 2024-12-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hi, I'm trying to build FreeCAD on my laptop, Fedora release 41 with Python 3.13, and I'm facing massive of warnings about `warning: ‘Py_UNICODE’ is deprecated` like the following.

```
/home/***/FreeCAD/src/CXX/Python3/Objects.hxx:1787:31: warning: ‘Py_UNICODE’ is deprecated [-Wdeprecated-declarations]                                                                                                                        
 1787 |     typedef std::basic_string<Py_UNICODE> unicodestring;                                                                                                                                                                              
      |                               ^~~~~~~~~~                                                                       
In file included from /usr/include/python3.13/unicodeobject.h:1014,                                                    
                 from /usr/include/python3.13/Python.h:79,                                                             
                 from /home/***/FreeCAD/src/CXX/WrapPython.h:58,                                                       
                 from /home/***/FreeCAD/src/CXX/Objects.hxx:37:                                                                                                                                                                               
/usr/include/python3.13/cpython/unicodeobject.h:10:37: note: declared here                                             
   10 | Py_DEPRECATED(3.13) typedef wchar_t Py_UNICODE;                                                                                                                                                                                       
      |                                     ^~~~~~~~~~                                                                                                                                                                                        /home/***/FreeCAD/src/CXX/Python3/Objects.hxx:1788:44: warning: ‘Py_UNICODE’ is deprecated [-Wdeprecated-declarations]                                                                                                                         1788 |     extern Py_UNICODE unicode_null_string[1];                                                                                                                                                                                         
      |                                            ^                                                                   
/usr/include/python3.13/cpython/unicodeobject.h:10:37: note: declared here                                                                                                                                                                    
   10 | Py_DEPRECATED(3.13) typedef wchar_t Py_UNICODE;                                                                                                                                                                                       
      |                                     ^~~~~~~~~~
/home/***/FreeCAD/src/CXX/Python3/Objects.hxx:1983:28: warning: ‘Py_UNICODE’ is deprecated [-Wdeprecated-declarations]
 1983 |         Char( Py_UNICODE v )                                                                                                                                                                                                          
      |                            ^                                                                                                                                                                                                          
/usr/include/python3.13/cpython/unicodeobject.h:10:37: note: declared here                                                                                                                                                                       10 | Py_DEPRECATED(3.13) typedef wchar_t Py_UNICODE;                                                                                                                                                                                       
      |                                     ^~~~~~~~~~                                                                                                                                                                                        
/home/***/FreeCAD/src/CXX/Python3/Objects.hxx:2029:39: warning: ‘Py_UNICODE’ is deprecated [-Wdeprecated-declarations] 
 2029 |         Char &operator=( Py_UNICODE v )                                                                                                                                                                                               
      |                                       ^                                                                        
/usr/include/python3.13/cpython/unicodeobject.h:10:37: note: declared here                                                                                                                                                                    
   10 | Py_DEPRECATED(3.13) typedef wchar_t Py_UNICODE; 
```

I'd like to create a PR for addressing this issue but I don't know about minimum Python version requirements.

I searched in wiki page and got three instructions:
- https://wiki.freecad.org/Compile_on_Linux#Compile_FreeCAD
- https://wiki.freecad.org/Compile_on_Windows
- https://wiki.freecad.org/Compile_on_MacOS
Those said nothing about Python requirement except Linux (>= 3.6).

On the other hand, the source code rejects Python 3.2 and below implicitly since the code uses `Py_UCS4` without any condition and it was introduced in Python 3.3.
https://github.com/FreeCAD/FreeCAD/blob/e698c73a3a1ec36bf6b735b2cd681d43b95ee0c1/src/CXX/Python3/Objects.hxx#L1790
https://docs.python.org/3.13/c-api/unicode.html#unicode-type

My question is what is the best way to get rid of the warnings:
1. Insert ifdef switch to just disable `Py_UNICODE` interface after Python 3.13: the most conservative approach
2. Remove use of `Py_UNICODE` interface: removing support and below more and more
3. Or any other suggestions?

### Full version info

```shell
$ git log -1
commit e698c73a3a1ec36bf6b735b2cd681d43b95ee0c1 (HEAD -> main, origin/main, origin/HEAD)
Author:     wmayer <wmayer@freecad.org>
AuthorDate: Sat Dec 14 17:25:04 2024 +0100
Commit:     wwmayer <werner.wm.mayer@gmx.de>
CommitDate: Sun Dec 15 16:20:30 2024 +0100

    Part: Fix crash due to infinite recursion
$ cat /etc/redhat-release
Fedora release 41 (Forty One)
$ python3 -V
Python 3.13.0
$ rpm -qi python3-devel
Name        : python3-devel
Version     : 3.13.0
Release     : 1.fc41
Architecture: x86_64
Install Date: Tue 05 Nov 2024 09:47:03 JST
Group       : Unspecified
Size        : 1854017
License     : Python-2.0.1 AND MIT
Signature   : RSA/SHA256, Wed 09 Oct 2024 07:50:20 JST, Key ID d0622462e99d6ad1
Source RPM  : python3.13-3.13.0-1.fc41.src.rpm
Build Date  : Tue 08 Oct 2024 23:45:43 JST
Build Host  : buildvm-x86-17.iad2.fedoraproject.org
Packager    : Fedora Project
Vendor      : Fedora Project
URL         : https://www.python.org/
Bug URL     : https://bugz.fedoraproject.org/python3.13
Summary     : Libraries and header files needed for Python development
Description :
This package contains the header files and configuration needed to compile
Python extension modules (typically written in C or C++), to embed Python
into other programs, and to make binary distributions for Python libraries.
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-16:
@adrianinsaval could you help out here?

adrianinsaval, 2024-12-16:
Currently our minimum version is python 3.8, but we could consider moving up to 3.10 if required

luzpaz, 2024-12-18:
@Flast can you link to the upstream ticket when you get a chance ?  
What's the status of this ticket, should we keep it open ?

Flast, 2024-12-19:
> @Flast can you link to the upstream ticket when you get a chance ?

Sorry, I still haven't created. I'll post the link when I opened.

> What's the status of this ticket, should we keep it open ?

Because the implementation is hard copied to this repo, the issue still exists even if the upstream fixes this issue. Therefore I think it should be open until importing upstream fix.

3x380V, 2025-06-04:
You might want to pick @wwmayer's [solution](https://github.com/wwmayer/FreeCAD/commits/issue_18545/).

PR #23803 (2025-09-25): Py3.13: fix warnings
Fixes (by @wwmayer):
```
In file included from src/3rdParty/PyCXX/CXX/cxxsupport.cxx:42:
src/3rdParty/PyCXX/CXX/Python3/cxxsupport.cxx:49:33: warning: ‘Py_UNICODE’ is deprecated [-Wdeprecated-declarations]
   49 | Py_UNICODE unicode_null_string[1] = { 0 };
      |                                 ^
In file included from /usr/include/python3.13/unicodeobject.h:1014,
                 from /usr/include/python3.13/Python.h:79,
                 from src/3rdParty/PyCXX/CXX/WrapPython.h:58,
                 from src/3rdParty/PyCXX/CXX/cxxsupport.cxx:37:
/usr/include/python3.13/cpy [... cut]
closed 2025-09-25 by chennes