FreeCAD/FreeCAD #18550: Opening Help throws python error

Artim96, 2024-12-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Selecting Help ->Help from the menu bar consitently will throw this error

```
15:48:58  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\user\AppData\Local\Programs\FreeCAD 1.0\Mod\Help\Help.py", line 107, in show
    location, _pagename = get_location(page)
                          ^^^^^^^^^^^^^^^^^^
  File "C:\Users\user\AppData\Local\Programs\FreeCAD 1.0\Mod\Help\Help.py", line 208, in get_location
    location, pagename = location_url(WIKI_URL + "/" + page + suffix, WIKI_URL + "/" + page)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user\AppData\Local\Programs\FreeCAD 1.0\Mod\Help\Help.py", line 170, in location_url
    with urllib.request.urlopen(req) as response:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user\AppData\Local\Programs\FreeCAD 1.0\bin\Lib\urllib\request.py", line 216, in urlopen
    return opener.open(url, data, timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user\AppData\Local\Programs\FreeCAD 1.0\bin\Lib\urllib\request.py", line 519, in open
    response = self._open(req, data)
               ^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user\AppData\Local\Programs\FreeCAD 1.0\bin\Lib\urllib\request.py", line 536, in _open
    result = self._call_chain(self.handle_open, protocol, protocol +
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user\AppData\Local\Programs\FreeCAD 1.0\bin\Lib\urllib\request.py", line 496, in _call_chain
    result = func(*args)
             ^^^^^^^^^^^
  File "C:\Users\user\AppData\Local\Programs\FreeCAD 1.0\bin\Lib\urllib\request.py", line 1391, in https_open
    return self.do_open(http.client.HTTPSConnection, req,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\user\AppData\Local\Programs\FreeCAD 1.0\bin\Lib\urllib\request.py", line 1351, in do_open
    raise URLError(err)
<class 'urllib.error.URLError'>: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: CA signature digest algorithm too weak (_ssl.c:1006)>
```

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE) [ OS: German/Germany (de_DE) ]
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Artim96, 2024-12-16:
> @Roy-043 related to changes here #18502 ?

No idea, I'm new to this program

> I cannot reproduce on Windows or macOS. @Artim96 does it happen if you Help>Restart in safe mode?

Yes, identical error message.

Roy-043, 2024-12-16:
> @Roy-043 related to changes here #18502 ?

Unrelated.

closed 2025-09-16 by maxwxyz