FreeCAD/FreeCAD #18555: Missing tags

Lin1Jay1, 2024-12-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In CAM I have found that Tags often (always?) don't appear when ramp angle set at  75 degrees.  I changed to 90 degrees and problem fixed.  In my case I had increased to 10 tags to see if that was a problem.  I ended up with 2 vertical on left side and 2 vertical on right side.  The 6 horizontal ones were missing

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

CAM

### Anything else?

[image]
75 degrees

[image]
90 deg.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-16:
Can you provide a demo file? Please also test with the stable 1.0 version as RC2 had some bugs. If possible, also check out the file in the dev build: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Lin1Jay1, 2024-12-16:
It’s the job  Outer profile with tags 4mmEM

If you see the dressup Tag and change the angle to 75 deg. You will see some of the tags don’t appear in the simulation.

luzpaz, 2024-12-18:
@Lin1Jay1 please click on https://github.com/FreeCAD/FreeCAD/issues/18555 and respond on github. When replying via email, all attachments are dropped.

Lin1Jay1, 2024-12-18:
Do you still need the demo file?

maxwxyz, 2024-12-19:
yes please

Lin1Jay1, 2024-12-19:
I tied to attach a .FCStd file and it says we don't support this file type.  So I have no idea what you want or how to send it to you.

maxwxyz, 2024-12-19:
Zip it and drag the .ZIP file in here.

luzpaz, 2024-12-19:
@Lin1Jay1 we didn't receive the attachment because when you respond via email to a github issue, it strips any attachments and just leaves the text. 
1. Please click on https://github.com/FreeCAD/FreeCAD/issues/18555
2. Zip up your attachment file
3. Drag-drop it into the github textbox

**Result:** the zip will auto-attach to this ticket.

Lin1Jay1, 2024-12-20:
[Tray12x18-Missing Tags.zip](https://github.com/user-attachments/files/18213578/Tray12x18-Missing.Tags.zip)
I have decided to abandon my attempt to use FreeCAD for CAM toolpaths.
I’m getting  
[image]
 and I have no idea how to fix this problem.  My latest download of RC04 is now completely non-functional re: being able to edit tool bit dimensions.  Previously I wasted a day trying to create tool paths using 1.0.0.  When I went back to RC02 I had no problem creating TPs in minutes.  When I went back to RC02 all my tool bits were gone.
I've spent a lot of time on this and I was making good progress until FC became non-functional.
[Tray-BirdsEye 12.15.24.pdf](https://github.com/user-attachments/files/18213723/Tray-BirdsEye.12.15.24.pdf)
I had all the TPs created for the frame and the insert.  I was just in the process of doing test cuts and optimizing the gcode when my FreeCAD became unusable.  I'm going to have to revert back to Swx 2005 and Carveco Maker I guess.
Still willing to respond to anything related to my inquiry.  I'd really like to help but I'm just stuck and I have projects I need gcode for.
Jay H

luzpaz, 2024-12-22:
Totally understand. Sorry to hear that you're encountering these issues.

> Still willing to respond to anything related to my inquiry.

Thanks. That is helpful. I'm keeping this ticket open since it's still an issue for you and someone else will most likely run into this.

Lin1Jay1, 2025-09-16:
Fine with me.  I've given up on Free CAD for CNC.  Will try again sometime in future.  Using Carveco and simplifies my life.Jay