FreeCAD/FreeCAD #18556: [part design] strange behavior on too large chamfers

mgth, 2024-12-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When chamfer is too large at top of a cylinder, old versions used to report an error, but in current version it compute an invalid body
[image]
Does the same with more complex shapes containing curved edges.

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39646 (Git) Conda
Build type: Release
Branch: main
Hash: b25b83c1a58bf0ae95cbd502d0ccae27e8127cad
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: OpenDark.qss/FreeCAD Classic/Qt default
Installed mods: 
  * Assembly3 0.12.2
  * CurvesWB 0.6.12
  * Defeaturing 1.2.2
  * ExtremeProDark 2.6.4
  * OpenDark 2024.3.13
  * parts_library
  * parts_library.bak
  * workfeature
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-16:
Please attach file

mgth, 2024-12-16:
[ChamferOnCylinder.zip](https://github.com/user-attachments/files/18154648/ChamferOnCylinder.zip)

tarman3, 2024-12-16:
Confirm.
But now it is possible to create a chamfer of 5 mm, and not 4.99 as before.

---

<details>
  <summary>FreeCAD 1.0.0.39109 (Git) makepkg, Arch Linux (KDE/plasma/wayland)</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods:lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.1; Silk 0.1.5; Curves 0.6.51

```
</details>

<details>
  <summary>FreeCAD 1.1.0dev.39560 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39560 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 1c35d08d87bb6dec7cb5adf51b2ed460ce43096d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.1; Silk 0.1.5; Curves 0.6.51

```
</details>

tarman3, 2024-12-16:
[image]

ElementW, 2026-08-25:
Still relevant as of faf5feae9d2d7b0d9bb81ee9a9fda266ed38ae2c and either OCC 7.9.3 or 8.1.0.dev1-3d097a0328

stefano2734, 2026-09-17:
Improvement to fillets will come with PRs in occt 8.0.2 and 8.1.0.