FreeCAD/FreeCAD #18559: BIM Views regressions

maxwxyz, 2024-12-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

BIM Views from https://github.com/FreeCAD/FreeCAD/pull/15836 does result in some errors on my system:

1. I the past I could double click on the workingplane in the BIM Views panel and activate it.
Now I get an error: [image]

2. Also, right clicking in the top view and creating a new prox WP results in an error:
[image]

3. The bottom part stays active, even when the file was closed and another (non-BIM file) is opened:
[image]

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.29.3
  * CurvedShapes 1.0.13
  * Curves 0.6.51
  * dodo 1.0.1
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.6.1
  * Ship 2024.11.26
  * Silk 0.1.5
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-16:
@yorikvanhavre I have added this issue for the comments in https://github.com/FreeCAD/FreeCAD/pull/15836

yorikvanhavre, 2024-12-17:
Thanks! Looking at it ASAP!

maxwxyz, 2025-01-06:
Also reported here: https://forum.freecad.org/viewtopic.php?p=800004
Double clicking a WP results in an error.

maxwxyz, 2025-03-01:
@yorikvanhavre this is still an issue in the latest weekly:

https://github.com/user-attachments/assets/e16a894e-241a-4ac2-a262-0e6d93d4b3df

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40444 (Git) Conda
Build type: Release
Branch: main
Hash: 089605bb4db98668661bf56db3f5f5bb4f7f0d87
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/108.888
Installed mods: 
  * Cables 0.1.3
  * CfdOF 1.29.8
  * CurvedShapes 1.0.13
  * Curves 0.6.58
  * dodo 1.0.1
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.1.2
  * parts_library
  * Rocket 4.1.1
  * sheetmetal 0.7.20
  * Silk 0.1.5

```

maxwxyz, 2025-03-30:
@yorikvanhavre FYI

furgo16, 2025-03-30:
@maxwxyz the fix could be similar to https://github.com/FreeCAD/FreeCAD/pull/19515, but I'd say it's a different issue. Happy to look into it. Could you please:

- Add the steps to reproduce in writing? It's difficult to see in the video what lead to this situation
- Add the stacktrace pasted in text from the report view. The notification area does not display the full traceback. Also text is searchable, so that those affected can find the relevant issue.

maxwxyz, 2025-03-31:
@furgo16 as mentioned in the issue: right click in the views window and "Add proxy"
This is the output:

```
20:35:41  Writing camera position
20:35:41  Writing objects shown/hidden state
20:35:41  IfcOpenShell was not found on this system. IFC support is disabled
20:35:41  Traceback (most recent call last):
20:35:41    File "C:\Users\max\Downloads\FreeCAD_weekly-builds-40971-conda-Windows-x86_64-py311\Mod\BIM\bimcommands\BimViews.py", line 359, in addProxy
20:35:41      self.addToSelection(obj)
20:35:41    File "C:\Users\max\Downloads\FreeCAD_weekly-builds-40971-conda-Windows-x86_64-py311\Mod\BIM\bimcommands\BimViews.py", line 368, in addToSelection
20:35:41      from nativeifc import ifc_tools
20:35:41    File "C:\Users\max\Downloads\FreeCAD_weekly-builds-40971-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_tools.py", line 52, in <module>
20:35:41      raise e
20:35:41    File "C:\Users\max\Downloads\FreeCAD_weekly-builds-40971-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_tools.py", line 37, in <module>
20:35:41      import ifcopenshell.geom
20:35:41    File "C:\Users\max\Downloads\FreeCAD_weekly-builds-40971-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\geom\__init__.py", line 55, in <module>
20:35:41      from . import occ_utils as utils
20:35:41    File "C:\Users\max\Downloads\FreeCAD_weekly-builds-40971-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\geom\occ_utils.py", line 27, in <module>
20:35:41      from typing_extensions import assert_never
20:35:41  ModuleNotFoundError: No module named 'typing_extensions'

```

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40971 (Git) Conda
Build date: 2025/03/24 18:02:34
Build type: Release
Branch: main
Hash: b33c2363226a5fc1e8222a02659118c95e1a5119
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/133.772/2
Installed mods: 
  * Cables 0.1.4
  * CfdOF 1.29.8
  * CurvedShapes 1.0.13
  * Curves 0.6.59
  * dodo 1.0.1
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Gridfinity 0.11.2
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.1.2
  * parts_library
  * Rocket 4.1.1
  * sheetmetal 0.7.21
  * Silk 0.1.5
  * Telemetry 1.0.0
  * Telemetry.backup1742214704.5089447 1.0.0 (Disabled)

```

furgo16, 2025-04-01:
This points to an issue in the `ifcopenshell.geom` module. It cannot find the `typing_extensions` package, so it fails to load. It's strange, as I thought `typing_extensions` was a dependency of `ifcopenshell`, which you have installed (ifcopenshell 0.8.1).

Could you try to manually install `typing_extensions` and see if the issue can still be reproduced? I'm not on Windows, so I cannot offer guidance on the installation. On Linux, this would be generally installed via `pip install typing_extensions`, if that helps.

```
20:35:41      import ifcopenshell.geom
20:35:41    File "C:\Users\max\Downloads\FreeCAD_weekly-builds-40971-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\geom\__init__.py", line 55, in <module>
20:35:41      from . import occ_utils as utils
20:35:41    File "C:\Users\max\Downloads\FreeCAD_weekly-builds-40971-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\geom\occ_utils.py", line 27, in <module>
20:35:41      from typing_extensions import assert_never
20:35:41  ModuleNotFoundError: No module named 'typing_extensions'
```

maxwxyz, 2025-04-02:
I did not install ifcopenshell manually and never used BIM WB before it was integrated in FreeCAD so if this is required it is not installed by FreeCAD.

After installing typing_extension manually, I get the same error, which says ther is no ifcopenshell installed at all ( IfcOpenShell was not found on this system. IFC support is disabled).

```
19:50:57  Writing camera position
19:50:57  Writing objects shown/hidden state
19:50:57  IfcOpenShell was not found on this system. IFC support is disabled
19:50:57  Traceback (most recent call last):
19:50:57    File "C:\Users\max\Downloads\FreeCAD_weekly-builds-40971-conda-Windows-x86_64-py311\Mod\BIM\bimcommands\BimViews.py", line 359, in addProxy
19:50:57      self.addToSelection(obj)
19:50:57    File "C:\Users\max\Downloads\FreeCAD_weekly-builds-40971-conda-Windows-x86_64-py311\Mod\BIM\bimcommands\BimViews.py", line 368, in addToSelection
19:50:57      from nativeifc import ifc_tools
19:50:57    File "C:\Users\max\Downloads\FreeCAD_weekly-builds-40971-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_tools.py", line 52, in <module>
19:50:57      raise e
19:50:57    File "C:\Users\max\Downloads\FreeCAD_weekly-builds-40971-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_tools.py", line 37, in <module>
19:50:57      import ifcopenshell.geom
19:50:57    File "C:\Users\max\Downloads\FreeCAD_weekly-builds-40971-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\geom\__init__.py", line 55, in <module>
19:50:57      from . import occ_utils as utils
19:50:57    File "C:\Users\max\Downloads\FreeCAD_weekly-builds-40971-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\geom\occ_utils.py", line 27, in <module>
19:50:57      from typing_extensions import assert_never
19:50:57  ModuleNotFoundError: No module named 'typing_extensions'

```

BTW another thing I've noticed in the views panel: The views panel is also shown on the start page and you can right click and add levels or proxy WP without even a document which results in this error:
```
19:49:18  Traceback (most recent call last):
19:49:18    File "C:\Users\max\Downloads\FreeCAD_weekly-builds-40971-conda-Windows-x86_64-py311\Mod\BIM\bimcommands\BimViews.py", line 345, in addLevel
19:49:18      FreeCAD.ActiveDocument.openTransaction("Create BuildingPart")
19:49:18      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
19:49:18  AttributeError: 'NoneType' object has no attribute 'openTransaction'

```

furgo16, 2025-04-02:
> After installing typing_extension manually, I get the same error, which says ther is no ifcopenshell installed at all ( IfcOpenShell was not found on this system. IFC support is disabled).

Can you try to type this on the Python console?

```
import typing_extensions
import ifcopenshell
```

It should produce no output. If it fails to load, please post the output. It might mean something went wrong with the manual installation of `typing_extensions`.

furgo16, 2025-04-03:
> FreeCAD does not recognize the manually installed typing_extension.

Ok, that's the issue, then. Unfortunately, I don't know enough about Python on Windows to be of much help here.

The next thing I would do on Linux would be a `pip list` to see if it's listed there and indeed installed, despite the successfully installed message.

> IMO it should not be necessary to manually fiddle around and install dependencies manually which are needed by FreeCAD.

Agreed. My first guess would be that whichever puts together the Windows builds (conda?) bundled IfcOpenshell (`0.8.1` by looking at your about info), but somehow did not install its dependencies. However, this theory is at odds with the fact that if you install `typing_extensions` FreeCAD does not seem to recognize it.

Perhaps someone more knowledgeable with the Windows builds can chime in.

furgo16, 2025-04-03:
Is Ifcopenshell also listed on pip list?

furgo16, 2025-04-03:
Then I think you have two Python versions on your system. The one shipped with FreeCAD won't see any packages you install with the pip you're executing, which is linked to the other Python you installed.

furgo16, 2025-04-07:
@maxwxyz this looks like the issue you're having: https://github.com/FreeCAD/FreeCAD-Bundle/issues/371

maxwxyz, 2025-04-07:
Correct, does this also result in the second issue described in the original post?

furgo16, 2025-04-07:
I'd say they're not related, but it'd be a guess.

furgo16, 2025-06-02:
@maxwxyz with https://github.com/FreeCAD/FreeCAD/pull/21641 merged, does this fix your issue with ifcopenshell and typing_extensions?

semhustej, 2025-08-22:
`2. Also, right clicking in the top view and creating a new prox WP results in an error:`

This is the typing_extensions issue, which should be solved: https://github.com/FreeCAD/FreeCAD/issues/22457

I am able to create WPproxy without an error.

`3. The bottom part stays active, even when the file was closed and another (non-BIM file) is opened:`

I am not able to reproduce with the latest weekly. A few seconds timeout was introduced ca. 8 months ago. After those few seconds, the BIM_Views panel is cleared. I am not able to find the issue or PR to post a link.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/12 22:08:35
Build type: Release
Branch: main
Hash: 4195f7fee28e65b289b9390ff7a0deab49e81040
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Revit/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/126.557/1.25
Installed mods: 
  * Reporting
  * Telemetry 1.0.2

```

closed 2025-08-22 by maxwxyz