FreeCAD/FreeCAD #18565: Path for Tool Bit Shape not honored for Job Templates

Connor9220, 2024-12-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When starting a new job, using a Job Template that has tools defined, FC uses defaults to using the /tmp/.mount....  instead of honoring the bit path specified in preferences.  This results in Bit Shapes not loading for custom shapes and/or using updated shapes for default shapes. (I updated some of the default shapes and stored them in person documents area).  

If you add the tool manually, it works correctly.

On additional note:  When adding tools via the tool dock or job template, the Tool controller is prefixed with TC:  When adding via Job dialog, the 'TC:' Prefix is omitted.

I toggle Allow Duplicate object labels in one document option to true, because I find it annoying when FC adds 0001 etc to the ends of my tools.  

Example:
TC: 1-1/4" Bowl & Tray Bit 0.500 loc 2.250 oal
TC: 1-1/4" Bowl & Tray Bit 0.500 loc 2.250 oal001

[image]
[image]

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * ThreadProfile.backup1725502069.7449253 1.90.0 (Disabled)
  * QuickMeasure 2022.10.28
  * dodo 1.0.1
  * btl.backup1725502066.664948 0.9.9 (Disabled)
  * btl.backup1720304573.9159045 0.9.9 (Disabled)
  * OpenTheme 2024.9.1
  * A2plus.backup1725502065.0763218 0.4.65 (Disabled)
  * btl 0.9.9 (Disabled)
  * OpenDark 2024.3.13
  * A2plus 0.4.68
  * btl.backup1717184435.3801382 0.9.9 (Disabled)
  * lattice2 1.0.0
  * ThreadProfile 1.92.0
  * fasteners.backup1725502068.289526 0.5.24 (Disabled)
  * fasteners 0.5.31
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-12-17:
Any CAM brethren willing to weigh-in on this?

freckletonj, 2025-07-25:
Can confirm this is still an issue on the linux AppImage release `1.0.1`