FreeCAD/FreeCAD #18566: AdditiveHelix Bugs

luiz-developer-7, 2024-12-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have found some bugs when making an additivehelix speccialy when the helix is on another plane/axis different from the Z.
I had tried make a t shape connection with "threads" on all sides i made the vertical ones, but the horizontal one keep failing.

The same thing happen on others files trying to make similar things, at some point i had this issue even on z axis.

I will attach the file where i found problems on the horizontral helix, the profile sketch is the last one for who wanna try.

[example.zip](https://github.com/user-attachments/files/18159608/example.zip)

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-17:
The problem not with horizontal helix.
AdditiveHelix001 create a lot of errors in model.

[image]

tarman3, 2024-12-17:
Rotate circle to hide this point.

[image]

luiz-developer-7, 2024-12-17:
hi, just wanna know how the additive helix one create the issues, and the circle thing, just to avoid it in the future. Thanks a lot

maxwxyz, 2024-12-17:
@FlachyJoe FYI, I guess similar to most helix issues with the refine option

FEA-eng, 2026-10-05:
[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260929 (Git shallow)
Build date: 2026/09/29 22:39:08
Build type: Release
Branch: grafted,grafted
Hash: 3160daf1e2b62b0472854da4848255c6b3ad554c
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```

moolight-seashell, 2026-10-06:
yep this is still broken