FreeCAD/FreeCAD #18569: STEP export creates files which don't show up in other applications

kliment, 2024-12-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello, I'm one of the KiCad librarians. 3D models made with recent versions of FreeCAD don't show up in KiCad anymore. The STEP files emitted are defective in some way. I've created a minimal example that consistently exports incorrectly. 
[step-export-defect.zip](https://github.com/user-attachments/files/18161917/step-export-defect.zip)
This contains the FreeCAD file as well as the corresponding exported STEP. This STEP does not show up in KiCad and several other applications, but shows up fine in FreeCAD and a handful of others. Analysis Situs shows it correctly but complains about it.

It appears that every surface is set to 100% transparent as the object still has volume. Adjusting the transparency in FreeCAD has no effect. This leads us to suspect it may be related to the fix for https://github.com/FreeCAD/FreeCAD/issues/14938

The minimal reproduceable case for me is fusion of two objects. Exporting a single object (padded rectangle sketch for example) generates non-broken STEPs.

This problem appears in FreeCad 1.0.0 and 1.0rc4, but *not* in 1.0rc2.

### Full version info

```shell
OS: KDE Flatpak runtime (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * kicadStepUpMod 11.4.4
  * Manipulator 1.5.7
  * fasteners 0.5.29
  * freecad.gears 1.0.0
  * sheetmetal 0.4.26
  * QuickMeasure 2022.10.28
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-17:
Confirm
<details>
  <summary>FreeCAD 1.1.0dev.39673 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39673 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.2; Silk 0.1.5; Curves 0.6.52

```
</details>

Syres916, 2024-12-17:
We're talking about a commit in the first three weeks of October from what I can see, I'll lock it down shortly.

kliment, 2024-12-17:
Sorry about that - a library contributor reported correct exports from 1.0rc2 on a specific part that we first saw this on, but apparently 1.0rc2 also produces bad output with some models, including the minimal test case I provided.

Syres916, 2024-12-18:
@davesrocketshop a Git Bisect shows commit https://github.com/FreeCAD/FreeCAD/commit/8b5a3b1124c06228266b5cc06dceb64db61718cd as responsible for this regression. A good Step file output using 
[PartDesignExampleFusion.FCStd.zip](https://github.com/user-attachments/files/18188311/PartDesignExampleFusion.FCStd.zip) would give the last few lines as:

```
#874 = ( GEOMETRIC_REPRESENTATION_CONTEXT(3) 
GLOBAL_UNCERTAINTY_ASSIGNED_CONTEXT((#878)) GLOBAL_UNIT_ASSIGNED_CONTEXT
((#875,#876,#877)) REPRESENTATION_CONTEXT('Context #1',
  '3D Context with UNIT and UNCERTAINTY') );
#875 = ( LENGTH_UNIT() NAMED_UNIT(*) SI_UNIT(.MILLI.,.METRE.) );
#876 = ( NAMED_UNIT(*) PLANE_ANGLE_UNIT() SI_UNIT($,.RADIAN.) );
#877 = ( NAMED_UNIT(*) SI_UNIT($,.STERADIAN.) SOLID_ANGLE_UNIT() );
#878 = UNCERTAINTY_MEASURE_WITH_UNIT(LENGTH_MEASURE(5.E-07),#875,
  'distance_accuracy_value','confusion accuracy');
#879 = PRODUCT_RELATED_PRODUCT_CATEGORY('part',$,(#7));
#880 = MECHANICAL_DESIGN_GEOMETRIC_PRESENTATION_REPRESENTATION('',(#881)
  ,#874);
#881 = STYLED_ITEM('color',(#882),#15);
#882 = PRESENTATION_STYLE_ASSIGNMENT((#883,#889));
#883 = SURFACE_STYLE_USAGE(.BOTH.,#884);
#884 = SURFACE_SIDE_STYLE('',(#885));
#885 = SURFACE_STYLE_FILL_AREA(#886);
#886 = FILL_AREA_STYLE('',(#887));
#887 = FILL_AREA_STYLE_COLOUR('',#888);
#888 = COLOUR_RGB('',0.823529011197,0.847059025451,0.882352996632);
#889 = CURVE_STYLE('',#890,POSITIVE_LENGTH_MEASURE(0.1),#891);
#890 = DRAUGHTING_PRE_DEFINED_CURVE_FONT('continuous');
#891 = COLOUR_RGB('',9.803921802644E-02,9.803921802644E-02,
  9.803921802644E-02);
ENDSEC;
END-ISO-10303-21;
```

and therefore the file has no entries containing `TRANSPARENCY` which with the commit above applied incorrectly makes all surfaces transparent.

oursland, 2025-01-14:
I am encountering issues with OCCT 7.8.1 exporting files with 1000x scaling.  OCCT 7.7.2 does not have this issue.  This could be related to @kliment's issue.

@tarman3 and @Syres916 are experiencing a different issue resulting from incorrect transparency.

eagabriel, 2025-01-14:
Same issue here I can't use steps exported from freecad in OrcaSlicer anymore, it freezes Orcaslicer.

dsa-t, 2025-01-19:
This issue is about transparency, @oursland @eagabriel your issues seem different.

hyarion, 2025-02-01:
We have a lot of users who have real issues because of this. How about adding a blocker label?

cpresser, 2025-02-09:
Hi,

Similar to kliment, I am also affected by this bug when trying to make 3D-Models for KiCad.
Is there a known workaround? Like tinkering with the .step file directly?

kliment, 2025-02-11:
I also cannot reproduce in latest nightly using either the legacy or non-legacy export. So presumably this has been fixed in some stage between 1.0 and current nightly.

kliment, 2025-02-11:
@Chocolateandmilkwin Can you share a file and corresponding output?

kliment, 2025-02-11:
I tested with 40176 as well, and the file attached in my initial report now exports correctly.

Chocolateandmilkwin, 2025-02-11:
@kliment I have attached a zip with a freecad project, and one step file exported by the feature and one step file exported by the body, where the issue can be seen on the step file exported by the body.

[Color Issue.zip](https://github.com/user-attachments/files/18750249/Color.Issue.zip)

To replicate with the file i will refer to my earlier message to avoid spamming images.
[Earlier Message](https://github.com/FreeCAD/FreeCAD/issues/18569#issuecomment-2633747419)

kliment, 2025-02-11:
I don't think the two issues should have been merged, as this appears to be a different problem.

Chocolateandmilkwin, 2025-02-11:
@kliment This is exactly what i said in my issue, maybe max was confused

Chocolateandmilkwin, 2025-02-11:
@maxwxyz It appears that the two issues are not related after all, could you reopen the one i created?

[STEP: Export missing color when exporting body or part](https://github.com/FreeCAD/FreeCAD/issues/19336#top)
#19336

hyarion, 2025-02-11:
So if that one is opened, then this should be closed, right?

kliment, 2025-02-11:
Yes, I believe this one has been fixed by recent changes (though I'm not sure which exact change). #19336 is unrelated and should remain open.

Syres916, 2025-02-23:
> I can't reproduce this issue on FreeCAD-1.1.0dev-40176, I'm using these settings: <img alt="Image" width="619" src="https://private-user-images.githubusercontent.com/349748/411971377-e45f2c90-6d9c-44cb-acf6-a953bd9a7155.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDAzMjcwNDEsIm5iZiI6MTc0MDMyNjc0MSwicGF0aCI6Ii8zNDk3NDgvNDExOTcxMzc3LWU0NWYyYzkwLTZkOWMtNDRjYi1hY2Y2LWE5NTNiZDlhNzE1NS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMjIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDIyM1QxNjA1NDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lOWNiNzBlN2E5MTYzN2E3MTA0ZmZkNjFmMDhhNzgzNmZjMTI0Zjc0NDIxZDY1ZjRmM2IzMGJmNjA4NDRiNTU3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.rRTWNN6LssUVvnOZzDiFMYRw6cIKerc7ggriMvKbWwg">
> 
> The exported step opens correctly in OpenCascade's CAD Assistant, but the provided step file in the zip opens as transparent.
> 
> [@kliment](https://github.com/kliment) can you check if this is still an issue, maybe it was fixed?

@hyarion this issue should never have been closed, it was never fixed, the Step output should not have any references to Transparent surfaces using a file that is solid and set to Transparent = 0. This is just a selection of the output from the file I attached back on December 18th but exported today using the build below:

```
#1063 = SURFACE_STYLE_TRANSPARENT(1.);
#1064 = OVER_RIDING_STYLED_ITEM('overriding color',(#1065),#824,#881);
#1065 = PRESENTATION_STYLE_ASSIGNMENT((#1066));
#1066 = SURFACE_STYLE_USAGE(.BOTH.,#1067);
#1067 = SURFACE_SIDE_STYLE('',(#1068,#1071));
#1068 = SURFACE_STYLE_FILL_AREA(#1069);
#1069 = FILL_AREA_STYLE('',(#1070));
#1070 = FILL_AREA_STYLE_COLOUR('',#899);
#1071 = SURFACE_STYLE_RENDERING_WITH_PROPERTIES(.NORMAL_SHADING.,#899,(
    #1072));
#1072 = SURFACE_STYLE_TRANSPARENT(1.);
#1073 = OVER_RIDING_STYLED_ITEM('overriding color',(#1074),#841,#881);
#1074 = PRESENTATION_STYLE_ASSIGNMENT((#1075));
#1075 = SURFACE_STYLE_USAGE(.BOTH.,#1076);
#1076 = SURFACE_SIDE_STYLE('',(#1077,#1080));
#1077 = SURFACE_STYLE_FILL_AREA(#1078);
#1078 = FILL_AREA_STYLE('',(#1079));
#1079 = FILL_AREA_STYLE_COLOUR('',#899);
#1080 = SURFACE_STYLE_RENDERING_WITH_PROPERTIES(.NORMAL_SHADING.,#899,(
    #1081));
#1081 = SURFACE_STYLE_TRANSPARENT(1.);
#1082 = OVER_RIDING_STYLED_ITEM('overriding color',(#1083),#853,#881);
#1083 = PRESENTATION_STYLE_ASSIGNMENT((#1084));
#1084 = SURFACE_STYLE_USAGE(.BOTH.,#1085);
#1085 = SURFACE_SIDE_STYLE('',(#1086,#1089));
#1086 = SURFACE_STYLE_FILL_AREA(#1087);
#1087 = FILL_AREA_STYLE('',(#1088));
#1088 = FILL_AREA_STYLE_COLOUR('',#899);
#1089 = SURFACE_STYLE_RENDERING_WITH_PROPERTIES(.NORMAL_SHADING.,#899,(
    #1090));
#1090 = SURFACE_STYLE_TRANSPARENT(1.);
#1091 = OVER_RIDING_STYLED_ITEM('overriding color',(#1092),#865,#881);
#1092 = PRESENTATION_STYLE_ASSIGNMENT((#1093));
#1093 = SURFACE_STYLE_USAGE(.BOTH.,#1094);
#1094 = SURFACE_SIDE_STYLE('',(#1095,#1098));
#1095 = SURFACE_STYLE_FILL_AREA(#1096);
#1096 = FILL_AREA_STYLE('',(#1097));
#1097 = FILL_AREA_STYLE_COLOUR('',#899);
#1098 = SURFACE_STYLE_RENDERING_WITH_PROPERTIES(.NORMAL_SHADING.,#899,(
    #1099));
#1099 = SURFACE_STYLE_TRANSPARENT(1.);
ENDSEC;
END-ISO-10303-21;
```

```
OS: Linux Mint 22.1 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.40329 (Git)
Build type: Release
Branch: main
Hash: f92fcbd089695c155c16bd0019f36544a78e2166
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Logical/physical DPI: 96/93.1131
Installed mods: 
  * FreeCAD-themes 2025.1.7
  * sheetmetal 0.7.11
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.29.8
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Curves 0.6.58

```

@maxwxyz could you please reopen this issue, we really require @davesrocketshop to give some constructive feedback regarding the regression introduced in June 2024 which is the root cause of these `SURFACE_STYLE_TRANSPARENT` entries

Syres916, 2025-02-23:
I'm happy to write a unit test for this if that's what's required.

hyarion, 2025-02-23:
@Syres916 but if the original issue can't be reproduced, then that issue must be fixed, right?

There seems to be multiple regressions so maybe what you've found is another issue?

A test would be wonderful!
It would be good if you could test it against 1.0 too so it can be patched for 1.0.1

Syres916, 2025-02-26:
@kliment @hyarion I'm getting to the crux of why I'll state this issue was partially fixed and completely fixed from the same root cause [commit](https://github.com/FreeCAD/FreeCAD/commit/8b5a3b1124c06228266b5cc06dceb64db61718cd) from June 2024, see Werner's comment https://github.com/FreeCAD/FreeCAD/pull/18206#issuecomment-2529369951 Note: I added my test file to this issue a day after @kliment opened it so it's not a stretch to say _Have both test files been fixed?_ and if the answer is _No_ then _Why?_ before closing the issue.

Basically I can create the detailed Step Unit Tests for **Export only**  and if @yorikvanhavre wants I can (hopefully) add an IFC Export Unit Test that includes a window in a Wall to ensure it's properly transparent only in the glass and nowhere else.

The export tests for transparency, color per face, color for complete object and multiple Part containers (assembly) will interrogate the text output of the Step/IFC files themselves (the current Step colour test I found which passes doesn't do this) as to not accidentally mask the problem using import code.

In order to fix all Step issues which includes `ExportOCAF.cpp` and `ImportOCAF.cpp` which _still_ aren't fully fixed in `main` yet (I'm a little puzzled on why but no doubt the reason will smack me when I try) for 1.0.1 based on Werner's feedback means the whole material/color needs aligning and I'm guessing that's been a large stumbling block as some of it was seen as a new feature in 1.1dev????

maxwxyz, 2025-03-07:
I still experience this issue. Even when I export STEP files from 1.1dev and import the same in 1.1dev they are fully transparent.
Not sure if it is this issue or https://github.com/FreeCAD/FreeCAD/issues/19336.
Reopening.

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.40504 (Git) Conda
Build type: Release
Branch: main
Hash: 855ad6c70904f911b2ce7114dd1a5bc24da98bd1
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, IfcOpenShell 0.0.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Logical/physical DPI: 96/133.772
Installed mods: 
  * Cables 0.1.3
  * CfdOF 1.29.8
  * CurvedShapes 1.0.13
  * Curves 0.6.58
  * dodo 1.0.1
  * fasteners 0.5.37
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.1.2
  * parts_library
  * Rocket 4.1.1
  * sheetmetal 0.7.21
  * Silk 0.1.5
  * Telemetry 1.0.0
```

Syres916, 2025-03-08:
> Not sure if it is this issue or [#19336](https://github.com/FreeCAD/FreeCAD/issues/19336). Reopening.

Really easy to prove, either open the step file you exported in a text editor and look for `SURFACE_STYLE_TRANSPARENT` there should be none if Transparency was set to zero. Alternatively import the attached known good step file from 0.21, just remove the .zip from the filename.

[PartDesignExampleFusion-Fusion021.step.zip](https://github.com/user-attachments/files/19143670/PartDesignExampleFusion-Fusion021.step.zip)

Syres916, 2025-03-09:
@hyarion after carrying out some crude debugging tests comparing the default supplied example file `PartDesignExample.FCStd` and the boolean test file `PartDesignExampleFusion.FCStd` I supplied at the top of this issue I have concluded that the Fusion file is being treat as a series of subshapes which means https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Import/App/ExportOCAF2.cpp#L333 is being triggered when it shouldn't, my understanding this color set should only be if the user has set faces differently to the solid color.

Therefore if I comment out this `aColorTool->SetColor(subLabel, color, colorType);` the Fusion object is exported correctly. Just to assist, here is the output for the standard example file for Face/Edge check:

```
07:14:10  ExportOCAF2::setupObject no dot : Face
07:14:10  ExportOCAF2::setupObject no dot : Edge
```

and here is the output for the Fusion file:

```
07:14:50  ExportOCAF2::setupObject no dot : Face
07:14:50  ExportOCAF2::setupObject no dot : Face10
07:14:50  ExportOCAF2::setupObject no dot : Face11
07:14:50  ExportOCAF2::setupObject no dot : Face12
07:14:50  ExportOCAF2::setupObject no dot : Face13
07:14:50  ExportOCAF2::setupObject no dot : Face14
07:14:50  ExportOCAF2::setupObject no dot : Face15
07:14:50  ExportOCAF2::setupObject no dot : Face16
07:14:50  ExportOCAF2::setupObject no dot : Face17
07:14:50  ExportOCAF2::setupObject no dot : Face18
07:14:50  ExportOCAF2::setupObject no dot : Face19
07:14:50  ExportOCAF2::setupObject no dot : Face20
07:14:50  ExportOCAF2::setupObject no dot : Face21
07:14:50  ExportOCAF2::setupObject no dot : Face22
07:14:50  ExportOCAF2::setupObject no dot : Face23
07:14:50  ExportOCAF2::setupObject no dot : Face24
07:14:50  ExportOCAF2::setupObject no dot : Face25
07:14:50  ExportOCAF2::setupObject no dot : Face26
07:14:50  ExportOCAF2::setupObject no dot : Face27
07:14:50  ExportOCAF2::setupObject no dot : Face28
07:14:50  ExportOCAF2::setupObject no dot : Face3
07:14:50  ExportOCAF2::setupObject no dot : Face7
07:14:50  ExportOCAF2::setupObject no dot : Face8
07:14:50  ExportOCAF2::setupObject no dot : Face9
07:14:50  ExportOCAF2::setupObject no dot : Edge
```

IMHO they should be the same. Still with that line commented out I tested exporting just a pure surface and that had the Transparency set to on so it feels like that I'll have to compile the commit from the day before June 2nd 2024 https://github.com/FreeCAD/FreeCAD/commit/958d83ed06ca32294227b4e1270cf41c838962e5 and see what Faces/Edges I get then, that Material commit is a monster for me to get my head around.

ambercap, 2025-04-11:
In 1.1.0dev-41214 on Windows, exporting any of my models to STEP format produces a file that looks like a wireframe on import. The same export in current official release 1.0 works correctly.

leoheck, 2025-04-15:
The title of this issue is a bit weird.

I saw similar behavior yesterday.

Workaround: When importing the step in FreeCad again, toggling the transparency on-off fixes the issue.
I am not sure if this can be achieved with other tools, but in FreeCad, it works.

mnesarco, 2025-04-18:
I have tried this PR with master and it works in my local build: https://github.com/FreeCAD/FreeCAD/pull/20861
More tests are needed.

closed 2025-04-21 by chennes