FreeCAD/FreeCAD #18570: Automatic display of constraint or sketch dimension selection

Roberto8303, 2024-12-17:
### Original PR

no blackport

### Description and Justification

in the first photo by selecting the height 37.57 it is not displayed in the constraints list. You have to scroll to find it.
It should scroll automatically to speed up operations when there are many constraints and quotas.
[image]
[image]

tarman3, 2024-12-17:
I would like to add that the same thing happens with selected geometry elements.

<details>
  <summary>FreeCAD 1.1.0dev.39673 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39673 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.2; Silk 0.1.5; Curves 0.6.52

```
</details>

luzpaz, 2024-12-17:
Thanks for the ticket Roberto. Just for your information, please use the correct bug reporting template 
[image]

closed 2025-04-17 by Roy-043