FreeCAD/FreeCAD #18571: Cannot export edge

fbucafusco, 2024-12-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a body that is mirrored against one of its faces. 

Over the mirrored result, I want to create a sketch and aligned it with the mirrored parent. 
I want my new sketch to be centered along one of the edges, belonging to the face that I ve used to mirror the main obj.

The export geometries works with all the edges but it it doesnt with the edge I mentioned. 

I recorded a small animated gif to show you the issue.

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Danish/Denmark (da_DK)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-17:
please attach file

fbucafusco, 2024-12-17:
[part2_bk.zip](https://github.com/user-attachments/files/18163896/part2_bk.zip)

tarman3, 2024-12-17:
Just hide mirror operation

https://github.com/user-attachments/assets/d9e16c4a-9b77-43ec-a8ab-aec7cc1bb8ef

fbucafusco, 2024-12-17:
Ah! Ok Thanks! 
And yeah, 1st I am learning but also it is that way because I just uploaded just a part, but my actual file has many all around. That is why it is offsetted from the origin

closed 2024-12-17 by fbucafusco