FreeCAD/FreeCAD #18573: Fillet on subtractet squared off cylinder creates round and square corners simultaneously

Roundwolf, 2024-12-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As I subrtracted a square off a cylinder on the middle axis and tried to add a fillet on the resulting edges, the corners of the fillet became weird as you can see in the picture. Find the file attached. 
[image]
[Fillet_Cylinder.zip](https://github.com/user-attachments/files/18165755/Fillet_Cylinder.zip)

I would prefer to have only edged corners. It is reproduceable. 
Is there a solution for that? 

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Assembly3 0.12.2
  * InventorLoader 1.4.0
  * CAM
  * Manipulator 1.5.7
  * Curves 0.6.45
  * sheetmetal 0.4.24
  * fasteners 0.5.18
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-17:
https://github.com/user-attachments/assets/e34b9860-4446-4e42-acb0-05dd5c68140b

tarman3, 2024-12-17:
or like this

https://github.com/user-attachments/assets/bd23dfc0-c69d-478d-ac7b-260f53ded90c

FEA-eng, 2025-09-18:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/16 20:38:23
Build type: Release
Branch: main
Hash: bac49e465908071e7400a547eb813bc5017ae7dc
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```