FreeCAD/FreeCAD #18574: freecad/utils.py installed into system wide location

pmjdebruijn, 2024-12-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The following file seems to be installed to a system wide location:

/usr/local/lib/python3.11/dist-packages/freecad/utils.py

Where I'm assuming it probably should have gone somewhere into:

/usr/lib/FreeCAD-Python3

### Full version info

```shell
1.0.0+36~g82d1210f81 (releases/FreeCAD-1-0 branch)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-12-18:
@adrianinsaval would you mind weighing-in ?

luzpaz, 2025-09-18:
@pmjdebruijn is this still the case ?