FreeCAD/FreeCAD #18575: [Import] Step files produced by FC 0.21.x and VariCAD (probably others too) are imported with Transparency @ 100 even though the file contains no reference to any transparency

Syres916, 2024-12-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

While investigating https://github.com/FreeCAD/FreeCAD/issues/18569 and comparing the internals of the Step files produced by FC 1.0, 1.1 and 0.21.x, I decided to use VariCAD to import and export a known good file only to realise that even that file is imported what looks like Wireframe but is in actual fact a Solid with Transparency set to 100%. Just remove the .zip off the end of the filename below to test.

[PartDesignExample-Fusion_VariCAD.step.zip](https://github.com/user-attachments/files/18167613/PartDesignExample-Fusion_VariCAD.step.zip)

### Full version info

```shell
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39673 (Git)
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.12.3, Qt 5.15.13, Coin 4.0.3, Vtk 9.1.0, OCC 7.8.2.dev
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * sheetmetal 0.6.2
  * Behave-Dark-Colors 0.1.1
  * 3dfindit-freecad-integration 1.2.0
```

### Subproject(s) affected?

None

### Anything else?

I'll have to do a separate Bisect on this issue, it may well be the same commit but I would have thought that this would have been picked up much earlier if that's the case.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-12-17:
The commit responsible is not the same as https://github.com/FreeCAD/FreeCAD/issues/18569 , the last good build I have is:

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda AppImage 2nd December 2024
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * sheetmetal 0.6.2
  * Behave-Dark-Colors 0.1.1
  * 3dfindit-freecad-integration 1.2.0
```

the first bad build I have is:

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39560 (Git) Conda AppImage 9th December 2024
Build type: Release
Branch: main
Hash: 1c35d08d87bb6dec7cb5adf51b2ed460ce43096d
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * sheetmetal 0.6.2
  * Behave-Dark-Colors 0.1.1
  * 3dfindit-freecad-integration 1.2.0
```

wwmayer, 2024-12-17:
Works fine with
```
OS: Ubuntu 22.04.5 LTS (XFCE/xubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39673 (Git)
Build type: debug
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/Classic theme/Qt default
```

It took several iterations to find and fix most of the alpha value vs. transparency issues of the Color class. This is the latest PR that has been merged recently: https://github.com/FreeCAD/FreeCAD/pull/18206

Syres916, 2024-12-17:
Even with clean config files I still get the issue so I added some crude debug info to https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/Part/Gui/ViewProviderExt.cpp#L358-L365 and changed to:

```
    else if (prop == &Transparency) {
        const App::Material& Mat = ShapeAppearance[0];
        long value = toPercent(Mat.transparency);
        Base::Console().Warning("value: %i\n", value);
        Base::Console().Warning("Transparency: %i\n", Transparency.getValue());
        if (value != Transparency.getValue()) {
            float trans = fromPercent(Transparency.getValue());
            ShapeAppearance.setTransparency(trans);
        }
    }
```

Expected output would be:

```
value: 0
Transparency: 0
```

Actual output:

```
value: 0
Transparency: 100
```

hence the solid being set to Transparency 100%.

Syres916, 2024-12-19:
See https://github.com/FreeCAD/FreeCAD/issues/18584#issuecomment-2554189456

Syres916, 2024-12-20:
See https://github.com/FreeCAD/FreeCAD/issues/18584#issuecomment-2557875162

wwmayer, 2024-12-27:
When using the option "Import invisible objects" then the imported STEP model is 100% transparent, too.

wwmayer, 2024-12-27:
https://github.com/FreeCAD/FreeCAD/pull/18767

sslupsky, 2025-01-08:
@wwmayer Not sure #18767 completely fixed the problem assuming build 39786 incorporates the fix.  I have attached a STEP file generated from Kicad that was created by drawing a rectangle on the Edge.Cuts layer and exporting to STEP.  When imported into FreeCAD, the object's transparency is set to 100.  If I disable the "Import Invisible Objects" setting during the STEP import the object is still transparent.

[test.step.zip](https://github.com/user-attachments/files/18350211/test.step.zip)

```
OS: macOS 14.6.1
Architecture: x86_64
Version: 1.1.0dev.39786 (Git) Conda
Build type: Release
Branch: main
Hash: 013eb6e6353ab6e5077ab4696653c9b493442638
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * Curves 0.6.52
```