FreeCAD/FreeCAD #18576: Assembly: the problem of rotation of coaxial moving parts

free777cat, 2024-12-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

the problem of rotation of coaxial moving parts

https://youtu.be/lE-K25MWI68

### Full version info

```shell
OS: Linux Mint 22.1 (MATE/mate/xcb)
Architecture: x86_64
Version: 1.1.0dev.39673 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 

  * sheetmetal 0.6.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

free777cat, 2024-12-17:
[Нова тека.zip](https://github.com/user-attachments/files/18167742/default.zip)

luzpaz, 2024-12-19:
I adjusted the ticket so it's more convenient for testers to comprehend the issue. Added a timecode to the Youtube link to succinctly show the bug.