FreeCAD/FreeCAD #18581: Pad feature not working for another isolated sketch

vironaxe, 2024-12-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The problem is that the pad feature is not working when I create another sketch that I want to select for pad. I have attached the video which shows that nothing happens when I try to apply this feature. While playing around, I assigned revolve feature to the sketch but then the first body disappears.

https://github.com/user-attachments/assets/c9cc0fcf-5a3b-4675-81ef-8962c4069ecb

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

None

### Anything else?

I have attached the FCStd file.

[tutorial_2.zip](https://github.com/user-attachments/files/18169148/tutorial_2.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-17:
Confirm. Before it was error message about multiple body.

Set Allow compound = True
[image]

luzpaz, 2024-12-18:
why was this closed?

vironaxe, 2024-12-18:
I closed it because the issue was solved by [tarman3](https://github.com/tarman3). As shown by him in the photo above, the _Allow Compound_ option should be set to _True_ which allowed the second sketch to be padded.

tarman3, 2024-12-18:
This solves the problem, but before FreeCAD gave an error and it was better than doing nothing.

closed 2024-12-18 by luzpaz