FreeCAD/FreeCAD #18582: Strange Symbol in report view, task view and property view

evansdaniel79, 2024-12-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have noticed that in Freecad 1.0 and 1.1 there seems to be a bug where there is the & symbol in the report view and other tabs. I haven't noticed any functional impart but it looks very strange. I am using open theme in the about section but it does it on the original theme too.
[image]

### Full version info

```shell
OS: KDE Flatpak runtime (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.39241 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at cdb9276)
Hash: cdb92768fe4f33d900664415c613f9eced5b8927
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Installed mods: 
  * OpenTheme 2024.9.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-17:
@snow-faerie could it be because or https://github.com/FreeCAD/FreeCAD/pull/15532 ?

Syres916, 2024-12-21:
This is a duplicate of https://github.com/FreeCAD/FreeCAD/issues/16640

closed 2024-12-21 by maxwxyz