FreeCAD/FreeCAD #18583: Cannot turn off the Commands panel from "View > Panels" menu

raindropsfromsky, 2024-12-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I cannot turn off the Commands panel from "View > Panels" menu.
When I turn it off, it disappears only for a couple of seconds, and then re-appears!

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39673 (Git) Conda
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Beltrami 1.1.0
  * CommandPanel
  * CurvedShapes 1.0.13
  * Curves 0.6.52
  * dodo 1.0.1
  * ExplodedAssembly
  * fasteners 0.5.33
  * frame 0.1.1
  * FrameForge 0.1.0
  * FreeCAD-Ribbon 1.4.16
  * freecad.gears 1.3.0
  * FreeGrid 2.1.1
  * OpenTheme 2024.9.1
  * SearchBar 1.2.1
  * sheetmetal 0.6.2
  * Silk 0.1.5
  * symbols_library
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-17:
Can not reproduce

<details>
  <summary>FreeCAD 1.1.0dev.39673 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39673 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.2; Silk 0.1.5; Curves 0.6.52

```
</details>

tarman3, 2024-12-17:
If it from addon [CommandPanel](https://github.com/triplus/CommandPanel) just remove it because last commit was 5 years ago.

luzpaz, 2024-12-18:
This is related to 3rd party CommandPanel[^1] external addon that has been unmaintained for several years, as tarman3 astutely points out. An aside, we should probably deprecate this in the addon repo[^2]. I'll open a ticket to do so.

[^1]: https://github.com/triplus/CommandPanel
[^2]: https://github.com/FreeCAD/FreeCAD-addons/blob/master/addonflags.json

closed 2024-12-18 by luzpaz