FreeCAD/FreeCAD #18584: Open a file created in FC 0.21.2 containing Part workbench boolean cut operation in Dev 1.1.0 opens like Wireframe not a solid

JohnPatrickMc, 2024-12-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Open FreeCAD 0.21.2, create a rectangle in sketcher, extrude it, create another rectangle or circle extrude it too, make cut  boolean operation between those, save it. Open it in FreeCAD 1.1.0 , it appears to be wireframe.
I have a bunch of files with Part WB boolean cuts and all of them open
[image]
 like that in Dev 1.1.0

### Full version info

```shell
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39673 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Argentina (es_AR)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * Manipulator 1.5.7
  * IconThemes
  * OpenTheme 2024.9.1
  * FreeCAD-themes 2024.7.24
  * sheetmetal 0.6.1
  * fasteners 0.5.29
  * lattice2 1.0.0
  * ProDarkThemePreferencePack 1.0.0
  * CurvedShapes 1.0.13
  * Render 2024.9.25
  * Behave-Dark-Colors 0.1.1
  * Curves 0.6.50
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-17:
please attach file

Syres916, 2024-12-17:
I believe it's the same faulty code as I've highlighted in https://github.com/FreeCAD/FreeCAD/issues/18575

JohnPatrickMc, 2024-12-17:
[Test opening 0.21 file 0.21 in Dev 1.1.0 .Zip](https://github.com/user-attachments/files/18171282/Test.opening.0.21.file.0.21.in.Dev.1.1.0.Zip)

JohnPatrickMc, 2024-12-17:
I think I closed this by mistake, I'm sorry

tarman3, 2024-12-17:
Confirm with 1.1.0dev.39673
Not confirm with 1.0.0.39109

<details>
  <summary>FreeCAD 1.0.0.39109 (Git) makepkg, Arch Linux (KDE/plasma/wayland)</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods:lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.1; Silk 0.1.5; Curves 0.6.51

```
</details>

<details>
  <summary>FreeCAD 1.1.0dev.39673 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39673 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.2; Silk 0.1.5; Curves 0.6.52

```
</details>

Syres916, 2024-12-19:
@wwmayer in order to correct this issue and #18575 I had to make changes as per https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Fix_Import_Step_021_Transparency
If this inverts the transparency on your setup do we need to check for certain dependency library versions?

Syres916, 2024-12-20:
I'm throwing the towel in on this one, I can get the models to load correctly but not with the CI tests pass and vice versa.

If I use a 0.21 heavy Part based model and test for Transparency the test passes but because it's a heavy model you can see on screen it loads with Transparency @ 0 (hence pass) then a split second later it changes to 100. Every object in the tree does the same as I see the Transparency Property code repeatedly triggered.

If anyone wants me to create a branch with the test model and the transparency test that I wrote, let me know.

maxwxyz, 2024-12-21:
@kadet1090 maybe you can take a look?

wwmayer, 2024-12-27:
https://github.com/FreeCAD/FreeCAD/pull/18761

JohnPatrickMc, 2025-05-06:
Hi, thank you for the hard work you are doing, I'd like to know if fixing this issue is still on the scope. Thnk you

chennes, 2025-10-23:
As we discussed in the most recent developer meeting, it seems that the best path forward to this is still the solution presented by @wwmayer in https://github.com/FreeCAD/FreeCAD/pull/18761 -- though at the time of the original discussion there was not agreement that this was the appropriate path and it seemed we would have time to develop an alternative strategy, that did not materialize.