FreeCAD/FreeCAD #18585: Make the constraint and tool cursor icon colors configurable

nspaeth, 2024-12-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The icons in this screenshot are red, which is somewhat difficult to see as someone with mild red-green colorblindness, especially in a dark theme. The color is currently not configurable.
[image]

It would also be nice if the constraint icons could have similar configuration. 
[image]

### Full version info

```shell
OS: KDE Flatpak runtime (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Installed mods: 
  * woodworking 0.21.2.33771.0.23.1
  * sheetmetal 0.6.2
  * FreeCAD-themes 2024.7.24
  * Behave-Dark-Colors 0.1.1
  * ProDarkThemePreferencePack 1.0.0
  * fasteners 0.5.33
  * ArchTextures
  * Gridfinity 0.6.0
  * OpenTheme 2024.9.1
  * parts_library
  * PieMenu 1.10.2
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hyarion, 2024-12-17:
Do we need colors here? Can’t we just use black and white icons for cursors like many other programs?

kadet1090, 2024-12-18:
We use red color for constraints so using the same one for tool cursor seems appropriate. I personally don't like the use of red color here as I think that it should be reserved for error states but it is a bigger discussion.

As for runtime color mapping - bitmap factory is capable of recoloring svgs at load time so it is possible to implement without much problem

maxwxyz, 2024-12-18:
@FreeCAD/design-working-group FYI
IMO if they chance, the icons in the toolbar should be the same, as they are now the same too.

nspaeth, 2024-12-18:
> @FreeCAD/design-working-group FYI IMO if they chance, the icons in the toolbar should be the same, as they are now the same too.

As it happens, those can be difficult as well, but since they are bigger and tend to be in the same spot I have less trouble with them and didn't bother bringing it up.

maxwxyz, 2024-12-18:
I mean there is no need to have red taskbar icons if the dimensions and constraints end up being all blue or whatever. In this case I'd also be in favor of neutral icons for the taskbar and instead make the constraints icon the same color on the sketch as constrained geometry (to not introduce another color). @obelisk79 FYI

MisterMakerNL, 2024-12-18:
White in dark mode and black in light mode, would be the logical solution here.
But both will not work in classic, so classic needs to be red. Which is annoying.

Roberto8303, 2024-12-18:
[image]
[image]

constraints can already be changed in color

hyarion, 2024-12-18:
@maxwxyz the title doesn't match with the description anymore.
It previously it stated that both constraint cursor icon color and tool cursor icon color should be configurable. Now I read the title as the tool icons (not cursor icons) colors should be configureable.