FreeCAD/FreeCAD #18588: Helix Error

IngoBrose, 2024-12-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I mirrored a substractive helix around a cylindre on a base plane on xz. there are parts there is no substraction of helix
[infinity_002a.pdf](https://github.com/user-attachments/files/18171526/infinity_002a.pdf)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39673 (Git) Conda
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * A2plus 0.4.68
  * BOLTSFC
  * CommandPanel
  * fasteners 0.5.33
  * fcgear
  * freecad.gears 1.3.0
  * Gridfinity 0.6.0
  * kicadStepUpMod 11.4.4
  * Lithophane
  * MeshRemodel 1.10.35
  * offline-documentation 1.0.0-alpha
  * pcb 6.2023.1
  * Reinforcement
  * sheetmetal 0.6.1
  * workfeature
```

### Subproject(s) affected?

None

### Anything else?

[InfinityAchse_002a.zip](https://github.com/user-attachments/files/18171565/InfinityAchse_002a.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-17:
Please attach FCStd file

tarman3, 2024-12-17:
Please specify the problem more precisely

tarman3, 2024-12-17:
It's a bad idea to make things so complicated.
Divide the sketch into two parts for each helix.

[image]

tarman3, 2024-12-17:
Also maybe need to find another path to create this model.

tarman3, 2024-12-18:
Separated sketch and move profile little bit outside. Now looks good.
[InfinityAchse_002a_.zip](https://github.com/user-attachments/files/18175544/InfinityAchse_002a_.zip)

[image]

https://github.com/user-attachments/assets/f691cb95-5a4f-4e54-bd4e-6c0d9b6742d8

tarman3, 2024-12-18:
- Separated sketch
- Replaced ellipse curve by simple curves
- Moved profile little bit outside
- Replaced mirror by helix feature with left hand direction

Now looks good

[image]

https://github.com/user-attachments/assets/6f8d46ce-7959-477b-94ff-39be780176a4

IngoBrose, 2025-01-03:
It fails also with a simple cirlce as afigure to make the infinite helix also in the actual developer version 1.1.0 build 39748

FEA-eng, 2025-09-27:
It looks like this after recomputing in the newest weekly:

[image]

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```

FlachyJoe, 2025-09-27:
It works on my side
```
OS: Debian GNU/Linux forky/sid (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.43729 +1 (Git)
Build date: 2025/09/27 10:05:37
Build type: Unknown
Branch: PointOnSegment
Hash: 0876219bb64bb0e1abd8822828389d8cb4658a0c
Python 3.13.7, Qt 6.8.2, Coin 4.0.3, Vtk 9.3.0, boost 1_83, Eigen3 3.4.0, PySide 6.8.2.1
shiboken 6.8.2.1, SMESH 7.7.1.0, xerces-c 3.2.4, IfcOpenShell 0.8.3.post1, OCC 8.0.0.rc1-783c3440b2
Locale: English/United States (en_US) [ OS: French/France (fr_FR) ]
Navigation Style/Orbit Style/Rotation Mode: CAD/Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 85/86.4394/1
Installed mods: 
  * Render 2024.12.15
  * freecad.gears 1.3.0
  * lattice2 1.0.0
  * CfdOF 1.32.2
  * Lithophane
  * Curves 0.6.70
  * fasteners 0.5.42
  * Assembly4 0.60.2
  * Telemetry 1.0.2
  * Road 2025.2.22
  * parts_library
  * FCPDWorkbench 0.4.0
  * sheetmetal 0.7.55
```

Can you please check if it's a tesselation problem or a geometric one? On the first case, if you Part_Section the result you should show the external wire without hole or island.

[image]

FEA-eng, 2025-09-29:
> Can you please check if it's a tesselation problem or a geometric one? On the first case, if you Part_Section the result you should show the external wire without hole or island.

I made a cross-section here:

[image]

This is the result:

[image]

FlachyJoe, 2025-09-30:
> that's not just a visual issue.

to be investigate 👀

closed 2025-12-14 by github-actions[bot]

FEA-eng, 2025-12-14:
> > that's not just a visual issue.
> 
> to be investigate 👀

@maxwxyz Can you reopen it?