FreeCAD/FreeCAD #18593: Add-on manager: too many update buttons

furgo16, 2024-12-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]

There are four updates-related buttons on the Add-on UI. I feel this brings a confusing experience to users, and generally similar add-on/plugin managers from other software seem to have less such buttons.

1. <kbd>Refresh local cache</kbd> => there is most probably a valid reason for this, but to the user, it seems it's doing the same as button no. 3
2. <kbd>No updates available</kbd> => could the filter not be used for this?
3. <kbd>Check for updates</kbd>
4. <kbd>Check for update</kbd>

I've personally never figured out what the difference between <kbd>Refresh local cache</kbd> and <kbd>Check for updates</kbd> is. This is just an illustrative example, I can probably find out by doing some more research.

Could these perhaps be reduced to two?
1. <kbd>Check for updates</kbd> => bottom left, applies to all add-ons. It would fetch the full add-ons list and let the user know if there are any updates. It could also be called Refresh add-ons list, but perhaps "Check for updates" is better for consistency with the button below to check the update of an individual add-on.
5. <kbd>Check for update</kbd> => top right, applies to the currently selected add-on. Going further, could this button also be removed by automatically checking for an update when the user selects the individual package?

Regardless of the outcome of this issue, it would be great if each button had a tooltip that tells the user what it will do.

Thanks!

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39651 (Git) Snap 1227
Build type: Release
Branch: main
Hash: e698c73a3a1ec36bf6b735b2cd681d43b95ee0c1
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.33
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2025-01-14:
The Addon Manager could certainly use some @FreeCAD/design-working-group input, but they have a lot on their plate right now. If anyone is interested in mocking up a revised UI please feel free, I'm happy work on the implementation end.

pierreporte, 2025-01-14:
Button 4 could be removed. There is no point of checking updates of a single add-on when a full check is possible and fast.

Action triggered by buttons 1 and 3 could be automatic. Button 2 could automatically display how many updates are available.