FreeCAD/FreeCAD #18595: Feature icons all show generic part cube after restarting FreeCAD

maxwxyz, 2024-12-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating a new file in 1.0 with features from the Part WB, most of the icons in the treeview are replaced by a generic cube when restarting FreeCAD again.

Example (Check Slice or Compound filter icons):
[image]

When saving the file, closing FreeCAD and opening FreeCAD again, those icons are gone:
[image]

So it is really hard to quickly glance what object is which feature.
This also happens when files are opened across different computer and OS, verified on macOS too.

@FreeCAD/design-working-group FYI

Here is the file:

[part-features.zip](https://github.com/user-attachments/files/18180900/part-features.zip)

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.29.4
  * CurvedShapes 1.0.13
  * Curves 0.6.52
  * dodo 1.0.1
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * FreeCAD_SketchArch
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.6.2
  * Silk 0.1.5
```

### Subproject(s) affected?

Part

### Anything else?

Happening also in weekly:

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39673 (Git) Conda
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.29.4
  * CurvedShapes 1.0.13
  * Curves 0.6.52
  * dodo 1.0.1
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * FreeCAD_SketchArch
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.6.2
  * Silk 0.1.5
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-19:
@wwmayer do you know if this is a recent regression or if this was always the behavior in Part?

wwmayer, 2024-12-19:
To me it looks like a problem that has existed for many years.
If I create a compound filter it shows the correct icon. If a save the document, close it and reload the project it still shows the correct icon. If a close FreeCAD and restart it then it also shows the generic icon.

I have tested it with v0.21, v1.0 and v1.1 and they all show this behaviour.

maxwxyz, 2024-12-19:
And it only affects some features. Unfortunately when you rename the features, you have no idea if that feature is a compound filter or a boolean operation or something else as all look the same after opening the project again.

wwmayer, 2024-12-19:
The problem is the following:
1. The getIcon() of the corresponding view providers return the icon name of the associated command. E.g. for the compound filter this is "Part_CompoundFilter"
2. If you load the Part workbench in order to invoke the command the icon is registered under this name "Part_CompoundFilter"
3. This explains why reloading the previously saved file shows the correct icon
4. If you restart FreeCAD and Part is not the default workbench then the icon to "Part_CompoundFilter" won't be registered because the associated command is not created yet. In this case the generic icon is shown as fallback

Solution:
Change getIcon() to not return the name of the registered icon but return the path name inside the resource file. For the compound filter this will be: ":/icons/booleans/Part_CompoundFilter.svg"

maxwxyz, 2024-12-19:
I see, so just recomputing the file does not work, even when Part WB is loaded but you are in PartDesign WB. When I switch to Part WB and then recompute the document, it works. Weird workaround though. 

Any idea why this only effects some icons? E.g. Fusion or Cube works but not Slice or CompoundFilter?

wwmayer, 2024-12-19:
This is a proper solution and not a workaround.

PR #18618 (2024-12-20): Part: Fix getIcon() of several view providers to return full path nam…
…e of the icon file

Fixes #18595
closed 2024-12-20 by chennes