FreeCAD/FreeCAD #18601: Crash after loading an assembly project twice

wwmayer, 2024-12-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Download the project file from https://github.com/FreeCAD/FreeCAD/issues/17902
2. Open the file in FreeCAD
3. Close the document
4. Reopen the file
5. Crash

### Full version info

```shell
OS: Ubuntu 22.04.5 LTS (XFCE/xubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39679 (Git)
Build type: debug
Branch: main
Hash: 8e0bf192df2e22f29904a80f5e91252eea193a27
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: unset/Classic theme/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-12-19:
Confirmed

<details><summary>Stacktrace on Version: 1.1.0dev.39673 (Git) Conda AppImage</summary>

```
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3d1d0) [0x7f057084c1d0]
#1  0x7f05729c8688 in App::PropertyLinkList::~PropertyLinkList() from /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libFreeCADApp.so+0x68
#2  0x7f05729c87fa in App::PropertyLinkList::~PropertyLinkList() from /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libFreeCADApp.so+0xa
#3  0x7f057299d53b in App::Property::hasSetValue() from /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libFreeCADApp.so+0x9b
#4  0x7f0572a10ef4 in App::PropertyString::setValue(char const*) from /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libFreeCADApp.so+0x134
#5  0x7f057285e0d6 in App::Document::Document(char const*) from /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libFreeCADApp.so+0x19c6
#6  0x7f0572a5eff3 in App::Application::newDocument(char const*, char const*, bool, bool) from /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libFreeCADApp.so+0x463
#7  0x7f0572a627ca in App::Application::openDocumentPrivate(char const*, char const*, char const*, bool, bool, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&&) from /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libFreeCADApp.so+0x2ba
#8  0x7f0572a65321 in App::Application::openDocuments(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >*, bool) from /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libFreeCADApp.so+0x7a1
#9  0x7f0572a671d8 in App::Application::openDocument(char const*, bool) from /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libFreeCADApp.so+0xc8
#10  0x7f0572a72b46 in App::Application::sOpenDocument(_object*, _object*, _object*) from /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libFreeCADApp.so+0x106
#11  /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libpython3.11.so.1.0(+0x20603d) [0x7f057300603d]
#12  /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x7f0572fe5b5c]
#13  /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7f0572ff1532]
#14  /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libpython3.11.so.1.0(+0x2abd1d) [0x7f05730abd1d]
#15  /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7f05730ab794]
#16  /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libpython3.11.so.1.0(+0x2cabf3) [0x7f05730cabf3]
#17  /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libpython3.11.so.1.0(+0x2c669a) [0x7f05730c669a]
#18  /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x7f05730ba9c9]
#19  0x7f05723466bc in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libFreeCADBase.so+0x6c
#20  0x7f05739bd784 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libFreeCADGui.so+0x94
#21  0x7f05739bd913 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libFreeCADGui.so+0xe3
#22  0x7f05738ae79c in Gui::Application::open(char const*, char const*) from /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libFreeCADGui.so+0x24c
#23  0x7f057397c77a in Gui::ModuleIO::openFile(QString const&) from /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libFreeCADGui.so+0xda
#24  0x7f05739afaa9 in Gui::RecentFilesAction::activateFile(int) from /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libFreeCADGui.so+0xd9
#25  0x7f05739c717f in Gui::Command::_invoke(int, bool) from /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libFreeCADGui.so+0x23f
#26  0x7f05739c756f in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libFreeCADGui.so+0x12f
#27  /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7f0570ceb6f2]
#28  0x7f0571b6db7f in QActionGroup::triggered(QAction*) from /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libQt5Widgets.so.5+0x2f
#29  /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libQt5Core.so.5(+0x2eb964) [0x7f0570ceb964]
#30  0x7f0571b6a462 in QAction::triggered(bool) from /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libQt5Widgets.so.5+0x32
#31  0x7f0571b6ce2b in QAction::activate(QAction::ActionEvent) from /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libQt5Widgets.so.5+0xab
#32  /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libQt5Widgets.so.5(+0x2e12e2) [0x7f0571ce12e2]
#33  /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libQt5Widgets.so.5(+0x2e8c34) [0x7f0571ce8c34]
#34  0x7f0571baec08 in QWidget::event(QEvent*) from /tmp/.mount_FreeCAKMnOBO/usr/bin/../lib/libQt5Widgets.so.5+0x178
[... cut ...]
#57  /usr/lib/libc.so.6(__libc_start_main+0x8c) [0x7f0570834ecc]
#58  /tmp/.mount_FreeCAKMnOBO/usr/bin/freecad(+0x7c2a) [0x5646ca6d0c2a]
```

</details>

```
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39673 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

wwmayer, 2024-12-19:
With this older build it doesn't crash:
Version: 1.1.0dev.39229 (Git)

wwmayer, 2024-12-20:
When calling `expView.removeProperty("Moves")` the property Moves will be removed from the object and internally added to the global list of PropertyCleaner for later destruction.

The destruction is done inside the destructor of PropertyCleaner but because there is no tmp. instance of PropertyCleaner the destruction of the properties will be done anytime later.

When closing the document and creating a new document then the object of the removed property will be deleted and now the collected properties will be destroyed, too.

In this case the destructor of PropertyLinkList tries to access the object it belonged to but the point is dangling so that this leads to the segfault.

The fix is to clear the pointer inside the destructor of  PropertyCleaner before destroying the property.

An alternative fix could be:
```
    static void add(Property* prop)
    {
        if (_PropCleanerCounter == 0) {
            delete prop;
        }
        else {
            _RemovedProps.push_back(prop);
        }
    }
```

chennes, 2024-12-23:
Does this affect 1.0?

adrianinsaval, 2025-02-18:
backported #18041 and the fix from #18644 to 1.0.1

PR #18644 (2024-12-23): App: Fix possible crash when postponing the destruction of a removed …
…property

Fixes #18601
commit 4ba6e8e1 (2024-12-23): App: Fix possible crash when postponing the destruction of a removed property

Fixes #18601
commit c75fe83a (2025-02-18): App: Fix possible crash when postponing the destruction of a removed property

Fixes #18601
closed 2025-02-18 by adrianinsaval