FreeCAD/FreeCAD #18602: Units are set US customary (in lb) and 4 decimal places but display in ft and fractions of inch

Lin1Jay1, 2024-12-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I started File/New and started a sketch.  Rectangle 19.00" x 12.875   Dimensions displayed in ft-in 

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
```

### Subproject(s) affected?

None

### Anything else?

[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-19:
Is it the same as 
- https://github.com/FreeCAD/FreeCAD/issues/9963 or
- https://github.com/FreeCAD/FreeCAD/issues/12690

WandererFan, 2024-12-19:
US Customary is terrible.  

Use Imperial Decimal (inches only) or Building US (feet & inches) or Imperial for Civil Engineering (feet).

Lin1Jay1, 2024-12-19:
I’m new to FreeCAD so not exactly sure where to respond.  But this seems easiest.  If I’m wrong let me know.

Not the same as #9963 <https://github.com/FreeCAD/FreeCAD/issues/9963> 

Sounds pretty much the same as #12690 <https://github.com/FreeCAD/FreeCAD/issues/12690>    Fractions might be a separate issue.

Jay H

 

*	#12690 <https://github.com/FreeCAD/FreeCAD/issues/12690>

Lin1Jay1, 2024-12-19:
Good to know.  Thanks

I tried every option that was in the ball park and all gave same result.   As part of that I tried Building US and that gave an option of increments so I put is 1/32”  so that may be where fractions developed.

Jay H

WandererFan, 2025-08-31:
Check the document level unit schema in File > Document Information.

closed 2025-10-31 by luzpaz