FreeCAD/FreeCAD #18610: UI is too complecated and unintuiteve

hemangjoshi37a, 2024-12-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I am having hard time learning freecad because of it's hard UI.

If any developer can integrate this UI implementation it will be a blessing : https://github.com/HakanSeven12/Modern-UI

### Full version info

```shell
nothing else
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-12-19:
Please open a forum thread to discuss more elaborate and constructive feedback. ModernUI is not maintained anymore. It has been superceded by OpenTheme. Download that through the addon manager.

closed 2024-12-19 by luzpaz

hemangjoshi37a, 2024-12-19:
why not make https://github.com/geolta/FreeCAD-Ribbon this as default UI , it looks so modern and intuitive .

luzpaz, 2024-12-19:
> Please open a forum thread to discuss more elaborate and constructive feedback. 

But why not listen to what I stated above?  
The issue tracker is for triaging bugs, not for UI/UX preferences and tastes. That's what the forum is for. UI/UX is a very complex and controversial topic.