FreeCAD/FreeCAD #18616: Pocket Shape tool path produces jagged profile.  Doesn't do final profile to clean-up jagged edges.

Lin1Jay1, 2024-12-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have limited experience with FreeCAD CAM.  I was expecting the tool path shown but I also expected a final profile cleanup up to be part of the tool path.  As a work around I created separate jobs to clean-up the jagged edges.  It seems like a problem to me.   Is this a problem or just the way FC works?  It seems like a problem to me.  I have used Carveco Maker and it does a final profile to cleanup all the jagged edges.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default

I wasted a day yesterday trying to do toolpaths with 1.0.0 and finally gave up and went back to to RC2
```

### Subproject(s) affected?

None

### Anything else?

[image]
[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-19:
Select patter **ZigZagOffset**

Lin1Jay1, 2024-12-20:
I am going to have to abandon my attempt to use FreeCAD for CAM toolpaths.

I’m getting  and I have no idea how to fix this problem.  My latest download of RC04 is no completely non-functional is being able to edit tool bit dimensions.  Previously I wasted a day trying to create tool paths using 1.0.0.  When I went back to RC02 no problem creating TP.

 

Jay H

tarman3, 2024-12-20:
Try [last weekly build](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds).
Please describe the problem in detail.
Attach file.

phaseloop, 2025-03-17:
@Lin1Jay1 I agree that back then CAM was unusable  If you want to give it another try - please download the last "daily build".

Other than that - I suppose bug can be closed.

Lin1Jay1, 2025-07-13:
You can close.  I’ve given up on FreeCAD for CAM.  My solution is to generate design in SWX, create dwg, create dxf, import dxf into Carveco Maker.  Maker is US$17.50 / month ($180 / yr).  Process is a bit messy having to use 2 softwares but it allows me to be very productive.

Maker is EASY and always works.  So it’s worth it.

I’ll give FC another chance sometime in the future.

Jay H

 

Keep open or close ?

FEA-eng, 2025-09-20:
@tarman3 Should it be closed?

tarman3, 2025-09-20:
Yes

FEA-eng, 2025-09-22:
@luzpaz or @maxwxyz Can you close this?