FreeCAD/FreeCAD #18619: Offsed a shape from text doesn't work

duub, 2024-12-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm trying to offset a text but couldn't find a way that works.

1st option:

In the Draft WB create a "Shape from text" (Draft_ShapeString), use Offset (Draft_Offset).

```
20:10:12  Either Part.Wire or Part.Edges should be provided, returning None
20:10:12  Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 1.0\Mod\Draft\draftutils\todo.py", line 165, in doTasks
    Gui.doCommand(string)
  File "<string>", line 1, in <module>
  File "C:\Program Files\FreeCAD 1.0\Mod\Draft\draftfunctions\offset.py", line 133, in offset
    if DraftGeomUtils.hasCurves(newwire) and copy:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\FreeCAD 1.0\Mod\Draft\draftgeoutils\general.py", line 140, in hasCurves
    for e in shape.Edges:
             ^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'Edges'

20:10:12  Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 1.0\Mod\Draft\draftutils\todo.py", line 165, in doTasks
    Gui.doCommand(string)
  File "<string>", line 1, in <module>
  File "C:\Program Files\FreeCAD 1.0\Mod\Draft\draftfunctions\offset.py", line 133, in offset
    if DraftGeomUtils.hasCurves(newwire) and copy:
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\FreeCAD 1.0\Mod\Draft\draftgeoutils\general.py", line 140, in hasCurves
    for e in shape.Edges:
             ^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'Edges'

20:10:12  ToDo.doTasks, Unexpected error:
<class 'AttributeError'>
in ['offst = Draft.offset(FreeCAD.ActiveDocument.ShapeString, FreeCAD.Vector(1.8121923206663162, 1.6899461626940635, 0.7961117235880977), copy=True, occ=False)']
```

Option 2

In the Draft WB create a "Shape from text" (Draft_ShapeString), then convert to sketch with Draft to Sketch (Draft_Draft2Sketch), later in the PartWB use 2D offset and says "Wire is not closed". Checked the sketch and is completely closed. And shows a partial offset.

[textOffset.zip](https://github.com/user-attachments/files/18202266/textOffset.zip)

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: Spanish/Spain (es_ES) ]
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Curves 0.6.51
  * fasteners 0.5.29
  * Manipulator 1.5.7
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-19:
Please attach FreeCAD file and font

tarman3, 2024-12-19:
The ShapeString object represents bsplne.
Offset doesn't work with bspline.

Perhaps there is a way to convert bsplne to arcs. This would help work around the problem.

tarman3, 2024-12-19:
You can create geometry with offset by InkScape and import svg to FreeCAD

[image]

duub, 2024-12-19:
don't get the result i want with inkscape, also I tried to import the geometry and when I try to boolean cut the inside shape to the outer shape i get an error too.

tarman3, 2024-12-20:
show something

duub, 2024-12-20:
This is an example of  what I meant with the boolean cut of the importeg geometry and the svg

[textOffset-importedSVG.zip](https://github.com/user-attachments/files/18207409/textOffset-importedSVG.zip)

duub, 2024-12-20:
I get this error: 08:26:38  Cut: A fatal error occurred when running boolean operation
is it because the path are bsplines too?? are bsplines something to avoid?

tarman3, 2024-12-20:
fix the geometry

https://github.com/user-attachments/assets/e1b86282-999c-47fa-9364-49e884b4b9da

duub, 2024-12-20:
Thanks for everything! This is gold tip! <3

closed 2024-12-20 by duub