FreeCAD/FreeCAD #18622: Possible license problems

coldtobi, 2024-12-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

(Found during preparation of the Debian FreeCAD 1.0 package.)

- Y14.5Font license modification grant:  
src/Mod/TechDraw/Gui/Resources/fonts/Y14.5Font.license states the font license and it seems not to allow modifications... 

- Kuka datasheets (src/Mod/Robot/Lib/Kuka/*.pdf seems to be proprietary datasheets by the KuKa company, and very likely "all rights reserved." 

### Full version info

```shell
Y14.5Font -> found while performing code review, font introduced with commit 5800d9155bb
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

oursland, 2024-12-19:
The original license of the Y14.5Font is:

> "This font is free to use for all personal and commercial applications. Copying and redistribution of the font is encouraged.  Credit to me as the creator of the font is appreciated but not necessary."

I do not see where it states that modifications are prohibited.  Credit is appreciated by the author, and is acknowledged by FreeCAD.

coldtobi, 2024-12-19:
@oursland  Copyright grants must be explicit. modification is not included in "use" as well...

coldtobi, 2024-12-20:
Regarding src/Mod/Robot/Lib/Kuka, there are several wrl files that do not have a license statement, but say:
`downloaded from http://www.kuka-robotics.com`

Possibly problematic as well.

jspiros, 2024-12-20:
I [sent this to the Debian bug](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1090833#10), but after I submitted it this issue was added to the bug's metadata, so I figured I'd copy my message here as well if it helps. I assume if we hear back from the author, that'll cover any ambiguity, but if not:

> I don't know if these are sufficient to resolve this issue, but if it's 
> helpful for whoever needs to make this call, the TTF files themselves 
> include the following "Copyright" metadata:
> 
> Y14.5-2018.ttf:
> 
> > Created with FontForge 2.0 (http://fontforge.sf.net/)
> > 
> > Copyright 2012-2013 Peter Kanold - Public domain for all personal and 
> > commercial use.
> 
> Y14.5-FreeCAD.ttf:
> 
> > Created with FontForge 2.0 (http://fontforge.sf.net/)
> > 
> > Copyright 2012-2020 Peter Kanold - Public domain for all personal and 
> > commercial use.
> > 
> > 
> > Modified 2023 WandererFan (wandererfan@gmail.com)
> 
> And the latest version of Y14.5-2018.ttf which I just downloaded from 
> Peter Kanold's website:
> 
> > Public Domain for all personal and commercial use.
> > Created with FontForge 2.0 (http://fontforge.sf.net/)
> > Copyright 2012-2023 Peter Kanold

coldtobi, 2024-12-21:
> I don't know if that's as unambiguous as it needs to be (especially as a declaration of copyright generally contradicts a public domain grant), but it's perhaps slightly better than the "license" file, as it at least suggests a public domain grant was intended.

This is information is in direct conflict with the information given in the source and what the author writes on their homepage, https://peterkanold.com/fonts:

    This font is free to use for all personal and commercial applications. Copying
    and redistribution of the font is encouraged.  Credit to me as the creator of
    the font is appreciated but not necessary.

Thanks @WandererFan  for contacting the Author we should hopefully know soon. 
(I'm probably late for this remark, but "Public Domain" has its problems too, as it is not generally a legal concept worldwide, mostly a US thing; It would be great if the author uses CC0 instead, or some font-related copyleft license like the SIL Open Font License.)

jspiros, 2024-12-21:
> This is information is in direct conflict with the information given in the source and what the author writes on their homepage, https://peterkanold.com/fonts:
>
> ```
> This font is free to use for all [...]

I wouldn't say that's in _direct conflict_, as I could see that text on the author's website as being their casual/informal analysis of the freedoms inherent, having already intended to dedicate it to the public domain. As far as I know, FontForge doesn't pre-populate that field with a public domain declaration, and that metadata is present in the current version of the font offered by the author's website, having been further updated with a new copyright year (yes, again, something of a contradiction) and with the public domain declaration's position rearranged to be the first encountered text, so it does seem intentional.

But, to be clear, I primarily meant "better" than the license file in the sense of doing a better job of suggesting the depth of the author's intent, which hopefully predicts a satisfying conclusion for the interests of FreeCAD and Debian when and however this is eventually concluded. No doubt, it certainly makes the most sense to get an explicit and unambiguous DFSG-compatible license from the author directly where possible, and indeed thanks to @WandererFan for seeking that.

> (I'm probably late for this remark, but "Public Domain" has its problems too, as it is not generally a legal concept worldwide, mostly a US thing; It would be great if the author uses CC0 instead, or some font-related copyleft license like the SIL Open Font License.)

Yes, and even CC0, like most licenses, isn't thorougly tested[^1] and can't be a complete guarantee. But it does at least provide a common way to express that intent explicitly, and hopefully well enough for any court that ends up in a position to assess it. I agree it would be a great option.

[^1]: There's perhaps some irony here. Since copyright enforcement tends to fall to the rightsholder, if the rightsholder fundamentally has a positive attitude towards free use and _intends_ to devote their work to the public domain (where that concept is recognized), that would make it very unlikely that the work would ever be the subject of any legal action in the first place. So, CC0 may be the license, or license-like instrument, least likely to ever _be_ tested in any jurisdiction. :smiley:

coldtobi, 2024-12-22:
Unfortunatly I've found addtional problems for Debian:

[src/Mod/TechDraw/Templates/locale/es-AR](https://github.com/FreeCAD/FreeCAD/tree/main/src/Mod/TechDraw/Templates/locale/es-AR)/*svg are CC-BY-NC-SA, the "NC" makes it problematic and [DFSG](https://wiki.debian.org/DebianFreeSoftwareGuidelines) incompatible.

coldtobi, 2024-12-22:
Another licensing issue, actually only probably clarification needed (@jriegel) 

The file [src/Tools/dir2qrc.py src/Tools/generate.py](https://github.com/FreeCAD/FreeCAD/blob/main/src/Tools/generate.py is marked as "Licence: GPL") , which is bit vague (which version of the GPL, is "or later" allowed?)

As there is GPL2+ and a few files GPL3+ (located under src/Mod/BIM) this can create a license incompatibility.
As the BIM workspace is the outlier, maybe the author could be asked to re-license to LGPL2.1+ ?

chennes, 2024-12-24:
> As there are a few files GPL3+ (located under src/Mod/BIM) this can create a license incompatibility. As the BIM workspace is the outlier, maybe the author could be asked to re-license to LGPL2.1+ ? (@yorkik)

@yorikvanhavre is the person to ask about this

yorikvanhavre, 2024-12-24:
> As there are a few files GPL3+ (located under src/Mod/BIM) this can create a license incompatibility. As the BIM workspace is the outlier, maybe the author could be asked to re-license to LGPL2.1+ ? (@yorkik)

I cannot remember what tragic mistake I did there :sweat_smile: 
The above PR fixes the license of those files to LGPL2+

WandererFan, 2024-12-24:
re: src/Mod/TechDraw/Templates/locale/es-AR)/*svg

I don't have an email for the author, but we can try pinging @andesfreedesign.

WandererFan, 2024-12-25:
replay from Mr. Kanold granting us rights to modify the font:
[Re_ Y14.5 License Issue - Debian_FreeCAD.pdf](https://github.com/user-attachments/files/18246274/Re_.Y14.5.License.Issue.-.Debian_FreeCAD.pdf)

coldtobi, 2024-12-26:
thanks @WandererFan! This is good news for FreeCAD
For Debian this is unfortunately not enough, as this is a license specific to FreeCAD, and thus still violating the DFSG (it restricts modification by people not associated with FreeCAD. 

(I need to think about how to solve this for Debian, but possibly a dedicated freecad-nonfree-assests package that hold distirbuteable, but non DFSG compliant stuff)

JohnOCFII, 2024-12-26:
> For Debian this is unfortunately not enough, as this is a license specific to FreeCAD, and thus still violating the DFSG (it restricts modification by people not associated with FreeCAD.

Doesn't this phrase "Yes, I grant you and FreeCAD and **any other companies you are associated with** full
licence to use and modify the Y14.5 font in any way you desire." include the Debian folks?

coldtobi, 2024-12-26:
To be DFSG compliant, anyone, associated or not, must enjoy the same rights.

yorikvanhavre, 2025-01-06:
For kuka files, I think they can safely be removed.

As for the fonts, are they absolutely needed @WandererFan ? What happens if they are not there? Could they be moved to an addon?

maxwxyz, 2025-06-19:
adding Blocker label so this is not lost

chennes, 2025-06-19:
That sounds like it would be an excellent addition: to have places in FreeCAD that load fonts have some ability to find them from within an Addon. I can think of a couple of ways of accomplishing this. Anyone looking for a small project?

jspiros, 2025-06-20:
Just to be clear, has anyone tried following up with Mr. Kanold to explain the nuances of what licensing is required, beyond just FreeCAD and including Debian? I'd be happy to do so, if that would be helpful.

WandererFan, 2025-06-21:
@jspiros:  here is the text of the last correspondence from Peter K (2024-12-24).  His website is here: https://peterkanold.com/fonts

I have an email for Peter if you want to contact him.  PM me on the forum or email wandererfan@freecad.org.

> Chris 
> Yes, I grant you and FreeCAD and any other companies you are associated with full licence to use and modify the Y14.5 font in any way you desire.
> I am very happy you have found it useful.
> 
> Peter Kanold
> Waterloo Ontario
> 
> On Fri, Dec 20, 2024 at 9:18 AM Wanderer Fan <[wandererfan@freecad.org](mailto:wandererfan@freecad.org)> wrote:
> 
>     Hi Peter.  Hope all is well with you.
> 
>     We've received a complaint from Debian about our use of your Y14.5 font
>     in FreeCAD.   We ship your original font as well as a slightly modified
>     version with the software.
> 
>     Apparently, because your license text of "This font is free to use for
>     all personal and commercial applications. Copying and redistribution of
>     the font is encouraged.  Credit to me as the creator of the font is
>     appreciated but not necessary." does not explicitly grant the right to
>     modify the font, we can not include our modified version in any Linux
>     distributions based on Debian (ubuntu, linux mint, etc).
> 
>     Our changes make the glyphs fit the em square a bit better.  This makes
>     aligning text easier for us.  I've attached a screen grabs of the the
>     original and our version.
> 
>     We would appreciate it greatly if you could grant us a license (lgpl or
>     creative commons? even a simple reply to this email should work) to
>     modify your font.
> 
>     I've found many copies of the font on various font download sites, but
>     the only "master copy" I can find is on your https://peterkanold.com/
>     site.  I you are still actively maintaining the font, I'd be happy to
>     send you a copy of our version.
> 
>     Thanks for your attention.

coldtobi, 2025-07-08:
Thanks @WandererFan.

> Yes, I grant you and FreeCAD and any other companies you are associated with full licence to use and modify the Y14.5 font in any way you desire.
> I am very happy you have found it useful.

Unfortunately, while Peter’s email is generous, the grant is personal. For Debian inclusion (especially in main), we need a license that clearly permits use, modification, and redistribution by _anyone_, not just FreeCAD or associated parties.

(For the time being I'll package the font for non-free)