FreeCAD/FreeCAD #18623: Support for holes in imported SVG

Beep6581, 2024-12-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Here is a very simple shape: a square, and within it a smaller square. This is part of a large QR code generated by Inkscape.

[svg_test.svg](https://github.com/user-attachments/assets/bc41e86a-bb56-4d34-acf0-4be98b244652)

It should look like this:
[image]
Notice the checkboard pattern visible between the squares. That is not a surface, it is a hole.

The problem is that when I import this into FreeCAD I get three squares, instead of two squares with a gap between them:
[image]

The reason for this is that the shape is actually made from a single path:
```
d="M 0,0 V 12.444202 H 12.444202 V 0 Z m 1.777669,1.7776693 h 8.888863 V 10.666532 H 1.777669 Z m 1.7776687,1.7776689 v 5.3335255 h 5.333525 V 3.5553382 Z"
```

You can make the image in Inkscape appear the same as in FreeCAD  if you select the shape and invoke Path > Break Apart.

According to this https://github.com/FreeCAD/FreeCAD/issues/15692#issuecomment-2260611610 FreeCAD supports only the SVG Tiny specification, and that is because it uses Qt for SVG import which only supported SVG Tiny. However, the linked issue pertains to TechDraw SVG export not import, and is about clipgroup, while the SVG in this issue uses nothing fancy, so I decided to open this dedicated issue.

Please add support for preserving "holes" in imported SVG.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39673 (Git)
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.12.7, Qt 6.8.1, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * OpenDark 2023.12.17
  * A2plus 0.4.68
  * fasteners 0.5.29
  * Woodworking 0.21.2.33771
  * CurvedShapes 1.0.12
  * Curves 0.6.51
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Beep6581, 2024-12-21:
[image]

Here is a full sample QR code.

Beep6581, 2024-12-21:
I found that selecting the imported parts and performing Part > Boolean XOR leads to accurate results. Even though that solves this problem, it is yet another arcane procedure one must learn in order to get results which match what is intuitively expected, and what other software do by default. I will leave this issue open.

aprospero, 2025-04-12:
See #20749 .