FreeCAD/FreeCAD #18625: Errors in BIM Example and a weird turning grid

Mannshoch, 2024-12-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If I open the BIM Example I get a lot off errors. Also there is a Grid shown, I do not know where it comes from. It flows around against how I turn the model.

[image]

```
07:34:21  Migrating Start Workbench to Start command... done.
07:34:38  pyException: Traceback (most recent call last):
  File "/snap/freecad/1230/usr/Mod/BIM/ArchSite.py", line 1008, in onChanged
    self.addDisplaymodeTerrainSwitches(vobj)
  File "/snap/freecad/1230/usr/Mod/BIM/ArchSite.py", line 973, in addDisplaymodeTerrainSwitches
    if main_switch.getNumChildren() == 4:   # Check if all display modes are available.
  File "/snap/freecad/1230/usr/lib/python3/dist-packages/pivy/coin.py", line 3991, in __getattr__
    raise e
  File "/snap/freecad/1230/usr/lib/python3/dist-packages/pivy/coin.py", line 3983, in __getattr__
    return SoBase.__getattribute__(self, name)
<class 'AttributeError'>: 'SoCoordinate3' object has no attribute 'getNumChildren'
07:34:38  pyException: Traceback (most recent call last):
  File "/snap/freecad/1230/usr/Mod/BIM/ArchSite.py", line 948, in updateData
    self.onChanged(obj.ViewObject,"SolarDiagramPosition")
  File "/snap/freecad/1230/usr/Mod/BIM/ArchSite.py", line 1008, in onChanged
    self.addDisplaymodeTerrainSwitches(vobj)
  File "/snap/freecad/1230/usr/Mod/BIM/ArchSite.py", line 973, in addDisplaymodeTerrainSwitches
    if main_switch.getNumChildren() == 4:   # Check if all display modes are available.
  File "/snap/freecad/1230/usr/lib/python3/dist-packages/pivy/coin.py", line 3991, in __getattr__
    raise e
  File "/snap/freecad/1230/usr/lib/python3/dist-packages/pivy/coin.py", line 3983, in __getattr__
    return SoBase.__getattribute__(self, name)
<class 'AttributeError'>: 'SoCoordinate3' object has no attribute 'getNumChildren'
07:34:38  pyException: Traceback (most recent call last):
  File "/snap/freecad/1230/usr/Mod/BIM/ArchSite.py", line 946, in updateData
    self.onChanged(obj.ViewObject,"SolarDiagram")
  File "/snap/freecad/1230/usr/Mod/BIM/ArchSite.py", line 1008, in onChanged
    self.addDisplaymodeTerrainSwitches(vobj)
  File "/snap/freecad/1230/usr/Mod/BIM/ArchSite.py", line 973, in addDisplaymodeTerrainSwitches
    if main_switch.getNumChildren() == 4:   # Check if all display modes are available.
  File "/snap/freecad/1230/usr/lib/python3/dist-packages/pivy/coin.py", line 3991, in __getattr__
    raise e
  File "/snap/freecad/1230/usr/lib/python3/dist-packages/pivy/coin.py", line 3983, in __getattr__
    return SoBase.__getattribute__(self, name)
<class 'AttributeError'>: 'SoCoordinate3' object has no attribute 'getNumChildren'
07:34:38  pyException: Traceback (most recent call last):
  File "/snap/freecad/1230/usr/Mod/BIM/ArchSite.py", line 946, in updateData
    self.onChanged(obj.ViewObject,"SolarDiagram")
  File "/snap/freecad/1230/usr/Mod/BIM/ArchSite.py", line 1008, in onChanged
    self.addDisplaymodeTerrainSwitches(vobj)
  File "/snap/freecad/1230/usr/Mod/BIM/ArchSite.py", line 973, in addDisplaymodeTerrainSwitches
    if main_switch.getNumChildren() == 4:   # Check if all display modes are available.
  File "/snap/freecad/1230/usr/lib/python3/dist-packages/pivy/coin.py", line 3991, in __getattr__
    raise e
  File "/snap/freecad/1230/usr/lib/python3/dist-packages/pivy/coin.py", line 3983, in __getattr__
    return SoBase.__getattribute__(self, name)
<class 'AttributeError'>: 'SoCoordinate3' object has no attribute 'getNumChildren'
07:34:39  pyException: Traceback (most recent call last):
  File "/snap/freecad/1230/usr/Mod/BIM/ArchSite.py", line 1008, in onChanged
    self.addDisplaymodeTerrainSwitches(vobj)
  File "/snap/freecad/1230/usr/Mod/BIM/ArchSite.py", line 973, in addDisplaymodeTerrainSwitches
    if main_switch.getNumChildren() == 4:   # Check if all display modes are available.
  File "/snap/freecad/1230/usr/lib/python3/dist-packages/pivy/coin.py", line 3991, in __getattr__
    raise e
  File "/snap/freecad/1230/usr/lib/python3/dist-packages/pivy/coin.py", line 3983, in __getattr__
    return SoBase.__getattribute__(self, name)
<class 'AttributeError'>: 'SoCoordinate3' object has no attribute 'getNumChildren'
07:34:39  pyException: Traceback (most recent call last):
  File "/snap/freecad/1230/usr/Mod/BIM/ArchSite.py", line 946, in updateData
[... 169 lines cut ...]
<class 'AttributeError'>: 'SoCoordinate3' object has no attribute 'getNumChildren'

```

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.39681 (Git) Snap 1230
Build type: Release
Branch: main
Hash: 0609fa2335b0a4a3277875a986618c226a3edbe1
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: German/Switzerland (de_CH)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-12-20:
This will be fixed with https://github.com/FreeCAD/FreeCAD/pull/18578

closed 2024-12-20 by yorikvanhavre

Mannshoch, 2025-01-02:
@yorikvanhavre the error message is disappeared but the grid - as seen in the gif -is still move weird.

yorikvanhavre, 2025-01-06:
I don't see that grid here... Are you using xorg or wayland? Might be a graphics driver issue...

Mannshoch, 2025-01-06:
Tested in Ubuntu 24.10 in x11 and wayland. Both have the same problem. 

It works fine on windows.

yorikvanhavre, 2025-01-07:
Are you by chance running ubuntu in a virtual machine?

Mannshoch, 2025-01-07:
Yes

yorikvanhavre, 2025-01-07:
That might be the cause of the problem then... OpenGL apps are notoriously having difficulties in virtual machines.

The problem is, I don't really know what to do here or even if there is a solution. This "grid" object seems to mimic the axes in the model, though... Does it go away if you remove the axes?

Mannshoch, 2025-01-07:
Yes, it only happens on the content of the Axes directory. And it realy seems that somehere the direction of Z is miscalculated. The move of the axes is completely mirrored.

yorikvanhavre, 2025-01-08:
Could you test if:

1) the grid disappear if you hide or delete the axes?
2) changing their Draw Style to "solid" changes anything?

Not sure we'll be able to do much to solve the issue, but this could give some clue.

Mannshoch, 2025-01-09:
> Could you test if:
> 
>     1. the grid disappear if you hide or delete the axes?
Yes it does, If I hid "BIMExample/Site/Axes" the axes hid. Alsoaxes get highlighted if I select one. Same goes for "BIMExample/2D plan view/Axes" If I activate it it is correctly (also mirrored)  placed beside the other grid.

>     2. changing their Draw Style to "solid" changes anything?
Yes, depending on the Draw Style the Axes get hidden or appear again.

[image]