FreeCAD/FreeCAD #18626: Assembly: Solve failed: vector

oursland, 2024-12-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On my assembly with 147 joints, I get the error:

    00:50:56  84.8003 <Assembly> AssemblyObject.cpp(167): Solve failed: vector

Followed by:

    00:51:18  107.832 <App> Document.cpp(3211): project_v0_1#Joint still touched after recompute
    00:51:18  107.832 <App> Document.cpp(3211): project_v0_1#Joint001 still touched after recompute
    00:51:18  107.832 <App> Document.cpp(3211): project_v0_1#Joint002 still touched after recompute
    00:51:18  107.832 <App> Document.cpp(3211): project_v0_1#Joint003 still touched after recompute
    ...

I have added `cpptrace` to my build and generated a backtrace from the exception and found what appears to be an indexing issue in frame `#6` at `MbD::GESpMatFullPv::backSubstituteIntoDU()` found in `src/3rdParty/OndselSolver/OndselSolver/GESpMat.cpp` resulting in `std::vector::at()` throwing an exception:

```
Cpptrace internal error: Unable to read object file /usr/lib/libc++abi.dylib
Cpptrace internal error: Unable to read object file /usr/lib/system/libunwind.dylib
Cpptrace internal error: Unable to read object file /usr/lib/libc++abi.dylib
Cpptrace internal error: Unable to read object file /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation
Cpptrace internal error: Unable to read object file /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation
Cpptrace internal error: Unable to read object file /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation
Cpptrace internal error: Unable to read object file /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation
Cpptrace internal error: Unable to read object file /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation
Cpptrace internal error: Unable to read object file /System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/HIToolbox
Cpptrace internal error: Unable to read object file /System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/HIToolbox
Cpptrace internal error: Unable to read object file /System/Library/Frameworks/Carbon.framework/Versions/A/Frameworks/HIToolbox.framework/Versions/A/HIToolbox
Cpptrace internal error: Unable to read object file /System/Library/Frameworks/AppKit.framework/Versions/C/AppKit
Cpptrace internal error: Unable to read object file /System/Library/Frameworks/AppKit.framework/Versions/C/AppKit
Cpptrace internal error: Unable to read object file /System/Library/Frameworks/AppKit.framework/Versions/C/AppKit
84.8004 <Assembly> AssemblyObject.cpp(168): Stack trace (most recent call first):
#0  0x000000019769975f at /usr/lib/libc++abi.dylib
#1  0x00000001a5272d77 at /usr/lib/system/libunwind.dylib
#2  0x0000000197698c7b at /usr/lib/libc++abi.dylib
#3  0x00000003537ab3cb in std::__1::__throw_out_of_range[abi:ue170006](char const*) at /Users/jso/code/FreeCAD/FreeCAD/.pixi/envs/default/bin/../include/c++/v1/stdexcept:264:5
#4  0x00000003537f856f in std::__1::vector<unsigned long, std::__1::allocator<unsigned long>>::__throw_out_of_range[abi:ue170006]() const at /Users/jso/code/FreeCAD/FreeCAD/.pixi/envs/default/bin/../include/c++/v1/vector:968:7
#5  (inlined)          in std::__1::vector<unsigned long, std::__1::allocator<unsigned long>>::at(unsigned long) at /Users/jso/code/FreeCAD/FreeCAD/.pixi/envs/default/bin/../include/c++/v1/vector:1557:15
#6  0x000000035389853f in MbD::GESpMatFullPv::backSubstituteIntoDU() at /Users/jso/code/FreeCAD/FreeCAD/src/3rdParty/OndselSolver/OndselSolver/GESpMatFullPv.cpp:0:0
#7  0x0000000353897497 in MbD::GESpMat::basicSolvewithsaveOriginal(std::__1::shared_ptr<MbD::SparseMatrix<double>>, std::__1::shared_ptr<MbD::FullColumn<double>>, bool) at /Users/jso/code/FreeCAD/FreeCAD/src/3rdParty/OndselSolver/OndselSolver/GESpMat.cpp:29:8
#8  0x00000003538af443 in MbD::MatrixSolver::timedSolvewithsaveOriginal(std::__1::shared_ptr<MbD::SparseMatrix<double>>, std::__1::shared_ptr<MbD::FullColumn<double>>, bool) at /Users/jso/code/FreeCAD/FreeCAD/src/3rdParty/OndselSolver/OndselSolver/MatrixSolver.cpp:47:8
#9  0x000000035389727b in MbD::GESpMat::solvewithsaveOriginal(std::__1::shared_ptr<MbD::SparseMatrix<double>>, std::__1::shared_ptr<MbD::FullColumn<double>>, bool) at /Users/jso/code/FreeCAD/FreeCAD/src/3rdParty/OndselSolver/OndselSolver/GESpMat.cpp:17:8
#10 0x00000003538d919f in MbD::PosICNewtonRaphson::lookForRedundantConstraints() at /Users/jso/code/FreeCAD/FreeCAD/src/3rdParty/OndselSolver/OndselSolver/PosICNewtonRaphson.cpp:136:20
#11 0x00000003538d8edb in MbD::PosICNewtonRaphson::handleSingularMatrix() at /Users/jso/code/FreeCAD/FreeCAD/src/3rdParty/OndselSolver/OndselSolver/PosICNewtonRaphson.cpp:121:11
#12 0x00000003539102c7 in MbD::VectorNewtonRaphson::solveEquations() at /Users/jso/code/FreeCAD/FreeCAD/src/3rdParty/OndselSolver/OndselSolver/VectorNewtonRaphson.cpp:57:9
#13 0x00000003538d8df7 in MbD::PosICNewtonRaphson::handleSingularMatrix() at /Users/jso/code/FreeCAD/FreeCAD/src/3rdParty/OndselSolver/OndselSolver/PosICNewtonRaphson.cpp:114:13
#14 0x00000003539102c7 in MbD::VectorNewtonRaphson::solveEquations() at /Users/jso/code/FreeCAD/FreeCAD/src/3rdParty/OndselSolver/OndselSolver/VectorNewtonRaphson.cpp:57:9
#15 0x00000003538b760b in MbD::NewtonRaphson::iterate() at /Users/jso/code/FreeCAD/FreeCAD/src/3rdParty/OndselSolver/OndselSolver/NewtonRaphson.cpp:75:9
#16 0x00000003538d844b in MbD::PosICNewtonRaphson::run() at /Users/jso/code/FreeCAD/FreeCAD/src/3rdParty/OndselSolver/OndselSolver/PosICNewtonRaphson.cpp:31:4
#17 0x0000000353904db3 in MbD::SystemSolver::runAllIC() at /Users/jso/code/FreeCAD/FreeCAD/src/3rdParty/OndselSolver/OndselSolver/SystemSolver.cpp:69:4
#18 0x00000003538ff443 in MbD::System::runKINEMATIC(std::__1::shared_ptr<MbD::System>) at /Users/jso/code/FreeCAD/FreeCAD/src/3rdParty/OndselSolver/OndselSolver/System.cpp:86:16
#19 0x00000003537c305f in MbD::ASMTAssembly::runKINEMATIC() at /Users/jso/code/FreeCAD/FreeCAD/src/3rdParty/OndselSolver/OndselSolver/ASMTAssembly.cpp:1447:20
#20 0x0000000350c04c17 in Assembly::AssemblyObject::solve(bool, bool) at /Users/jso/code/FreeCAD/FreeCAD/src/Mod/Assembly/App/AssemblyObject.cpp:164:22
#21 0x0000000350c04987 in Assembly::AssemblyObject::execute() at /Users/jso/code/FreeCAD/FreeCAD/src/Mod/Assembly/App/AssemblyObject.cpp:133:9
#22 0x0000000103b324bb in App::DocumentObject::recompute() at /Users/jso/code/FreeCAD/FreeCAD/src/App/DocumentObject.cpp:167:22
#23 0x0000000103ab3d87 in App::Document::_recomputeFeature(App::DocumentObject*) at /Users/jso/code/FreeCAD/FreeCAD/src/App/Document.cpp:3469:32
#24 0x0000000103ab241f in App::Document::recompute(std::__1::vector<App::DocumentObject*, std::__1::allocator<App::DocumentObject*>> const&, bool, bool*, int) at /Users/jso/code/FreeCAD/FreeCAD/src/App/Document.cpp:3177:31
#25 0x0000000103ba15bf in App::DocumentPy::recompute(_object*) at /Users/jso/code/FreeCAD/FreeCAD/src/App/DocumentPyImp.cpp:690:31
#26 0x0000000103b9a3a7 in App::DocumentPy::staticCallback_recompute(_object*, _object*) at /Users/jso/code/FreeCAD/FreeCAD/build/src/App/DocumentPy.cpp:2025:57
#27 0x00000001049f1307 at /Users/jso/code/FreeCAD/FreeCAD/.pixi/envs/default/lib/libpython3.12.dylib
#28 0x00000001049951cb at /Users/jso/code/FreeCAD/FreeCAD/.pixi/envs/default/lib/libpython3.12.dylib
#29 0x0000000104aafe07 at /Users/jso/code/FreeCAD/FreeCAD/.pixi/envs/default/lib/libpython3.12.dylib
#30 0x0000000104aa1bdb at /Users/jso/code/FreeCAD/FreeCAD/.pixi/envs/default/lib/libpython3.12.dylib
#31 0x0000000104b1919b at /Users/jso/code/FreeCAD/FreeCAD/.pixi/envs/default/lib/libpython3.12.dylib
#32 0x0000000104b1cdd7 at /Users/jso/code/FreeCAD/FreeCAD/.pixi/envs/default/lib/libpython3.12.dylib
#33 0x00000001032df59b in Base::InterpreterSingleton::runString(char const*) at /Users/jso/code/FreeCAD/FreeCAD/src/Base/Interpreter.cpp:249:15
[... 45 lines cut ...]
Python 3.12.0, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/FreeCAD Dark/
Installed mods: 
  * OpticsWorkbench 1.0.26
  * OpenTheme 2024.9.1
  * CfdOF 1.29.4
  * A2plus 0.4.68
  * kicadStepUpMod 11.4.4
  * fasteners 0.5.33
  * AirPlaneDesign 0.4.1
  * A2plus.backup1731435765.889473 0.4.65 (Disabled)
  * CurvedShapes 1.0.13
  * freecad.cross 0.4.0
  * Curves 0.6.52
  * Assembly4 0.50.15
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-28:
Any chance you could share an example file?

oursland, 2024-12-28:
Unfortunately, the file that I have is protected IP.

drwho495, 2025-03-01:
im having this issue

drwho495, 2025-03-01:
So I assume the issue is that the assembly is over-constrained, and it would be good if Freecad said as much and showed the affected joints by changing their icons.

maxwxyz, 2025-03-01:
@drwho495 could you add an example file and steps how top reproduce?

drwho495, 2025-03-01:
[solve-failed-ex.zip](https://github.com/user-attachments/files/19040937/solve-failed-ex.zip)

Here is an example, open the file, then change the sketch in Body001 called CHANGE_ME, and change the dimension that says 30 to 35

hyarion, 2025-03-19:
@aiksiongkoh is this something you could look into?

aiksiongkoh, 2025-11-16:
I am on travel. Will look Nov 23. Thanks

oursland, 2025-11-17:
@mnadler 

Thanks for investigating this and resolving the issue!

The current location for OndselSolver is: https://github.com/FreeCAD/OndselSolver

Feel free to submit a PR!

mnadler, 2025-11-17:
Thanks for pointing me in the right direction @oursland !

[PR](https://github.com/FreeCAD/OndselSolver/pull/12) submitted

closed 2026-02-11 by hyarion

fuzzybear3, 2026-02-12:
It seems that the same logical error lives in  `src/3rdParty/OndselSolver/OndselSolver/GESpMatFullPv.cpp:118-119`
// BUG: colOrder has size n, so valid indices are 0..n-1
```
  auto jn = colOrder->at(n);          // accesses index n — OUT OF BOUNDS
  answerX->at(jn) = rightHandSideB->at(m) / matrixA->at(m)->at(jn);  // at(m) also OOB
```
  Should be:
```  
auto jn = colOrder->at(n - 1);
  answerX->at(jn) = rightHandSideB->at(m - 1) / matrixA->at(m - 1)->at(jn);

```

PaddleStroke, 2026-02-12:
@aiksiongkoh can you please check what @fuzzybear3 said above?

@fuzzybear3 do you want to make a PR to the solver?

aiksiongkoh, 2026-02-13:
If I have to debug more, please do the following.
In Assembly Workbench
ClickMenuItem/Assembly/Export ASMT file/
Attach *.asmt to issue here.
If dragging is causing the problem,
In FreeCAD,
ClickMenuItem/Edit/Preferences/
In dialog, Select/Assembly/
Check/Log dragging steps/
Attach runPreDrag.asmt, runDragStep.asmt and dragging.log to issue here

fuzzybear3, 2026-02-13:
Hello,

As I said, I have not seen any problems with `src/3rdParty/OndselSolver/OndselSolver/GESpMatFullPv.cpp:118-119` today, so it could be fine, but it does seem to follow the same pattern as the previous bug from this PR and I hope that you can take a look at it.

My main issue now is the thrown complaint about the function not yet being implemented.
`OndselSolver/GESpMatFullPv.cpp:143 -  throw SimulationStoppingError("To be implemented.");`

Here is the is the terminal output:
```
❯ env -u WAYLAND_DISPLAY ./build/release/bin/FreeCAD
FreeCAD 1.2.0, Libs: 1.2.0devR45323 (Git)
(C) 2001-2026 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

NOTICE: Addon "/home/stevenguido/.local/share/FreeCAD/v1-2/Mod/Ondsel-Lens" disabled by presence of ADDON_DISABLED stopfile
NOTICE: Addon "/home/stevenguido/.local/share/FreeCAD/v1-2/Mod/3DfindIT" disabled by presence of ADDON_DISABLED stopfile
Failed to create wl_display (No such file or directory)
4e-08 <Assembly> AssemblyObject.cpp(177): Solve failed: To be implemented.
Unhandled std exception in ViewProvider::eventCallback: To be implemented.
(Event type: SoMouseButtonEvent, object type: AssemblyGui::ViewProviderAssembly)
Unhandled std exception in ViewProvider::eventCallback: To be implemented.
(Event type: SoMouseButtonEvent, object type: AssemblyGui::ViewProviderAssembly)
Unhandled std exception in ViewProvider::eventCallback: To be implemented.
(Event type: SoMouseButtonEvent, object type: AssemblyGui::ViewProviderAssembly)
100.143 <Assembly> AssemblyObject.cpp(177): Solve failed: To be implemented.
105.261 <Assembly> AssemblyObject.cpp(177): Solve failed: To be implemented.
106.115 <Assembly> AssemblyObject.cpp(177): Solve failed: To be implemented.
106.573 <Assembly> AssemblyObject.cpp(177): Solve failed: To be implemented.
107.05 <Assembly> AssemblyObject.cpp(177): Solve failed: To be implemented.
107.488 <Assembly> AssemblyObject.cpp(177): Solve failed: To be implemented.
107.942 <Assembly> AssemblyObject.cpp(177): Solve failed: To be implemented.
111.694 <Assembly> AssemblyObject.cpp(177): Solve failed: To be implemented.
```

Here is the file. I have reverted all of my changes before exporting this file.
[XMM62400.zip](https://github.com/user-attachments/files/25279629/XMM62400.zip)

fuzzybear3, 2026-02-16:
As for the solver message error. I have made a PR for that here https://github.com/FreeCAD/OndselSolver/pull/14

Sorry for making a lot of noise on a closed issue.