FreeCAD/FreeCAD #18627: Crash after changing light source settings

thomasgi1, 2024-12-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

How to reproduce in MacOS:

- start FC
- go to Preferences / Display / Light Sources
- drag the light direction arrow (changing the x, y, z values only does not trigger the crash)
- click OK
- FC crashes with following crash dump:
```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [2578]
Path:                  /Applications/FreeCAD.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-12-20 10:15:43.3149 +0100
OS Version:            macOS 12.7.6 (21H1320)
Report Version:        12
Anonymous UUID:        00D0AAE4-4FA2-1AA2-1DC3-429829F4143B

Sleep/Wake UUID:       C72F7EC0-F6D2-46EA-8A2D-3AA79856787A

Time Awake Since Boot: 1700000 seconds
Time Since Wake:       1418 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Codes:       0x0000000000000001, 0x0000000000000010
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [2578]

VM Region Info: 0x10 is not in any region.  Bytes before following region: 140737486356464
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      VM_ALLOCATE              7fffffe18000-7fffffe19000 [    4K] r-x/r-x SM=ALI  

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libCoin.80.0.3.dylib          	       0x111dd75a0 SoCallbackAction::~SoCallbackAction() + 144
1   libCoin.80.0.3.dylib          	       0x111dd770e SoCallbackAction::~SoCallbackAction() + 14
2   libCoin.80.0.3.dylib          	       0x111e47fb2 SoDragger::~SoDragger() + 194
3   libCoin.80.0.3.dylib          	       0x111e6dc6e SoRotateSphericalDragger::~SoRotateSphericalDragger() + 14
4   libCoin.80.0.3.dylib          	       0x111f9a8ac SoBase::destroy() + 172
5   libCoin.80.0.3.dylib          	       0x111f8e21a SoBaseList::truncate(int) + 74
6   libCoin.80.0.3.dylib          	       0x111f9ee80 SoChildList::~SoChildList() + 16
7   libCoin.80.0.3.dylib          	       0x111ff2425 SoGroup::~SoGroup() + 37
8   libCoin.80.0.3.dylib          	       0x11200da4e SoSeparator::~SoSeparator() + 14
9   libCoin.80.0.3.dylib          	       0x111f9a8ac SoBase::destroy() + 172
10  libCoin.80.0.3.dylib          	       0x111f8e21a SoBaseList::truncate(int) + 74
11  libCoin.80.0.3.dylib          	       0x111f9ee80 SoChildList::~SoChildList() + 16
12  libCoin.80.0.3.dylib          	       0x111fcbe98 SoBaseKit::~SoBaseKit() + 40
13  libCoin.80.0.3.dylib          	       0x111e51d3e SoDirectionalLightDragger::~SoDirectionalLightDragger() + 14
14  libCoin.80.0.3.dylib          	       0x111f9a8ac SoBase::destroy() + 172
[... 306 lines cut ...]
Architecture: x86_64
Version: 1.1.0dev.39673 (Git) Conda
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * MeshRemodel 1.10.35
  * DynamicData 2.70.0
  * OpenTheme 2024.9.1
  * Defeaturing 1.2.2
  * fasteners 0.5.33
  * CurvedShapes 1.0.13
  * Curves 0.6.52
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-20:
Can not reproduce in ArchLinux

<details>
  <summary>FreeCAD 1.0.0.39109 (Git) makepkg, Arch Linux (KDE/plasma/wayland)</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods:lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.1; Silk 0.1.5; Curves 0.6.51

```
</details>

<details>
  <summary>FreeCAD 1.1.0dev.39673 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39673 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.2; Silk 0.1.5; Curves 0.6.52

```
</details>

semhustej, 2024-12-20:
Can not reproduce on Windows.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39646 (Git) Conda
Build type: Release
Branch: main
Hash: b25b83c1a58bf0ae95cbd502d0ccae27e8127cad
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

```

JohnOCFII, 2024-12-20:
I **CAN** reproduce on macOS.  Problem occurs whether or not you click `Apply` before `OK` or not.

```
OS: macOS 15.2
Architecture: arm64
Version: 1.1.0dev.39673 (Git) Conda
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * FeedsAndSpeeds 0.6.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.33
  * Curves 0.6.52
```

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [38007]
Path:                  /Applications/Other Apps/FreeCAD-39673.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-12-20 09:14:52.5005 -0600
OS Version:            macOS 15.2 (24C101)
Report Version:        12
Anonymous UUID:        E1A9B3E8-17C0-3635-633F-31D076EC9805

Sleep/Wake UUID:       D5EC04D7-1A78-4CE1-849A-F6327FBACDC5

Time Awake Since Boot: 310000 seconds
Time Since Wake:       2154 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Codes:       0x0000000000000001, 0x0000000000000010

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [38007]

VM Region Info: 0x10 is not in any region.  Bytes before following region: 4331454448
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      1022cc000-1022ec000    [  128K] r-x/r-x SM=COW  /Applications/Other Apps/FreeCAD-39673.app/Contents/Resources/bin/freecad

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libCoin.80.0.3.dylib          	       0x10768dee4 SoCallbackAction::~SoCallbackAction() + 124
1   libCoin.80.0.3.dylib          	       0x10768e014 SoCallbackAction::~SoCallbackAction() + 12
2   libCoin.80.0.3.dylib          	       0x1076ec7b4 SoDragger::~SoDragger() + 192
3   libCoin.80.0.3.dylib          	       0x10770ef88 SoRotateSphericalDragger::~SoRotateSphericalDragger() + 12
4   libCoin.80.0.3.dylib          	       0x10783151c SoBase::destroy() + 212
5   libCoin.80.0.3.dylib          	       0x1078252a4 SoBaseList::truncate(int) + 80
6   libCoin.80.0.3.dylib          	       0x107835a14 SoChildList::~SoChildList() + 24
7   libCoin.80.0.3.dylib          	       0x107882a24 SoGroup::~SoGroup() + 40
8   libCoin.80.0.3.dylib          	       0x10789d480 SoSeparator::~SoSeparator() + 12
9   libCoin.80.0.3.dylib          	       0x10783151c SoBase::destroy() + 212
10  libCoin.80.0.3.dylib          	       0x1078252a4 SoBaseList::truncate(int) + 80
11  libCoin.80.0.3.dylib          	       0x107835a14 SoChildList::~SoChildList() + 24
12  libCoin.80.0.3.dylib          	       0x10785ea04 SoBaseKit::~SoBaseKit() + 40
13  libCoin.80.0.3.dylib          	       0x1076f59f8 SoDirectionalLightDragger::~SoDirectionalLightDragger() + 12
14  libCoin.80.0.3.dylib          	       0x10783151c SoBase::destroy() + 212
15  libCoin.80.0.3.dylib          	       0x107825404 SoBaseList::remove(int) + 44
16  libCoin.80.0.3.dylib          	       0x107882f90 SoGroup::removeChild(int) + 72
17  libFreeCADGui.dylib           	       0x105010f08 Gui::coinRemoveAllChildren(SoGroup*) + 552
18  libFreeCADGui.dylib           	       0x104fd7d78 Gui::View3DInventorViewer::~View3DInventorViewer() + 232
19  libFreeCADGui.dylib           	       0x104fd868c Gui::View3DInventorViewer::~View3DInventorViewer() + 12
20  libQt5Core.5.15.15.dylib      	       0x1047f7ba4 QObjectPrivate::deleteChildren() + 108
21  libQt5Core.5.15.15.dylib      	       0x1047f7ba4 QObjectPrivate::deleteChildren() + 108
22  libFreeCADGui.dylib           	       0x104c66ed8 Gui::Command::_invoke(int, bool) + 516
23  libFreeCADGui.dylib           	       0x104c66c98 Gui::Command::invoke(int, Gui::Command::TriggerSource) + 368
24  libQt5Core.5.15.15.dylib      	       0x1047fbfec 0x1046b4000 + 1343468
25  libQt5Core.5.15.15.dylib      	       0x1047df12c QCoreApplication::notifyInternal2(QObject*, QEvent*) + 176
26  CoreFoundation                	       0x18b4bc828 __CFRunLoopDoSource0 + 176
27  CoreFoundation                	       0x18b4bc58c __CFRunLoopDoSources0 + 244
28  CoreFoundation                	       0x18b4bb128 __CFRunLoopRun + 840
29  CoreFoundation                	       0x18b4ba724 CFRunLoopRunSpecific + 588
30  HIToolbox                     	       0x196a12530 RunCurrentEventLoopInMode + 292
31  HIToolbox                     	       0x196a1817c ReceiveNextEventCommon + 216
32  HIToolbox                     	       0x196a18508 _BlockUntilNextEventMatchingListInModeWithFilter + 76
33  AppKit                        	       0x18f025034 _DPSNextEvent + 660
34  AppKit                        	       0x18f9892d4 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 688
35  AppKit                        	       0x18f018060 -[NSApplication run] + 480
36  libqcocoa.dylib               	       0x1083c73c0 0x108398000 + 193472
37  freecad                       	       0x1022d2e6c main + 3228
38  dyld                          	       0x18b054274 start + 2840
```
[image]

luzpaz, 2025-01-14:
@hyarion any chance you could take a look at this macOS crash bug?

> As an aside, that large gray box jammed between the controls looks really awkward on that page...

@JohnOCFII what would remedy the situation here ?

maxwxyz, 2025-08-19:
Closing this as fixed as the light preferences have been reworked and it is not possible to reproduce the issue anymore

closed 2025-08-19 by maxwxyz