FreeCAD/FreeCAD #18630: UI improvement: Freecad super key and search bar to save space and improve ease of use

AdrianCM95, 2024-12-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This is an improvement request over the current user interface:

- to have more working space.
- to use the Freecad icon as a super key to hide the normal text menus under it and also display a search bar to quickly find commands.

In order to call this function, a shortcut like  Superkey+F, Shift+F, (F,F)..., or whatever combination could be used.
I think this can help to speed up the workflow even without knowing all the shortcuts for the rest of the functions.

This is the current layout:
[image]

This is the proposed layout:
[image]

### Full version info

```shell
OS: TUXEDO OS 3 (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Spain (es_ES)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-20:
https://github.com/Grubuntu/PieMenu

maxwxyz, 2024-12-20:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/8290

luzpaz, 2024-12-23:
Is it a duplicate though ? #8290 doesn't propose altering the top of the UI to save space, @maxwxyz

AdrianCM95, 2024-12-23:
I would say it is not a full duplicate. The search bar is a duplicate, true. However, the button to hide the menu and removing the window top bar is different.

maxwxyz, 2024-12-23:
Could you rephrase the original issue and title please

AdrianCM95, 2024-12-23:
@maxwxyz I rephrased the contract a little bit, is it better now?

yorikvanhavre, 2024-12-23:
This is more a duplicate of https://github.com/FreeCAD/FreeCAD/issues/12742

luzpaz, 2024-12-23:
Right, can we merge this ticket in to 12742 ?

yorikvanhavre, 2024-12-24:
Closing this one then, in favour of #12742

closed 2024-12-24 by yorikvanhavre

luzpaz, 2024-12-24:
@AdrianCM95 would you mind posting your content from this ticket into #12742 ?