FreeCAD/FreeCAD #18637: BIM: Arch_Remove tool not working for subtractions

maxwxyz, 2024-12-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a BIM wall and subtracted a part from it, by selecting the object to subtract and the wall. When I click the "-" (Arch_Remove) the subtracted part becomes a child of the wall.

I cannot select that children and hit the "-" to remove it from its parent (I want to remove the subtraction temporarily).
I cannot drag it manually outside the parent object.

I have to double click the wall, select the children in the subtraction folder and click the minus in the task dialog. Only then it is removed.

Expected behavior: Drag and drop or just the selected child is removed when clicking on Arch_Remove.

@yorikvanhavre FYI

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39673 (Git) Conda
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.29.4
  * CurvedShapes 1.0.13
  * Curves 0.6.52
  * dodo 1.0.1
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * FreeCAD_SketchArch
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.6.2
  * Silk 0.1.5
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

semhustej, 2024-12-20:
- I am able to remove the child from the wall using the Arch_Remove tool
- I am not able to remove the child from the wall by dragging it in the project tree. I get this error:
`17:35:09  141.624 Tree.cpp(2161): [ComboView] 'Unnamed#Structure' cannot be dragged out of 'Unnamed#Wall'`

https://github.com/user-attachments/assets/63bdbfd2-fd36-4d72-b035-9f8707df2d40

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39673 (Git) Conda
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

```

maxwxyz, 2024-12-20:
I see, it works for walls too but not for other elements like Structures or Slabs, can you confirm this?

maxwxyz, 2024-12-20:
Does it work when inside a Site>Building>Level hierarchy?

maxwxyz, 2024-12-21:
This also seems to be resolved in today's weekly

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * CfdOF 1.29.4
  * CurvedShapes 1.0.13
  * Curves 0.6.52
  * dodo 1.0.1
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * FreeCAD_SketchArch
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.6.2
  * Silk 0.1.5

```

closed 2024-12-21 by maxwxyz