FreeCAD/FreeCAD #18639: BIM: Cannot use other WB tools if selection elements are within a BIM site/group

maxwxyz, 2024-12-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I have a BIM project and create geometry inside it (e.g. a wall) and select it in the 3D view, always the parent at root level "Default site" is selected. I suspect this being the underlying issue of the much bigger issue:
I cannot use the elements of the geometry in the BIM project with other commands from other workbenches like Part or Sketcher.

Examples:
1. Create a new face from a selection of edges: `'Part.Shape' object has no attribute Edge xx`
2. Reference an edge from a wall inside Sketcher as external geometry: `Failed to add`

Here is the demo file with just a wall:
[BIM_element_bug.zip](https://github.com/user-attachments/files/18213778/BIM_element_bug.zip)

Screencast:

https://github.com/user-attachments/assets/a0b5a59e-7952-46e9-a451-3293444d9fb0

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39673 (Git) Conda
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * CfdOF 1.29.4
  * CurvedShapes 1.0.13
  * Curves 0.6.52
  * dodo 1.0.1
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * FreeCAD_SketchArch
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.6.2
  * Silk 0.1.5
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-20:
@semhustej @marcuspollio @yorikvanhavre is this a known limitation or am I using this the wrong way? Can you confirm this?

maxwxyz, 2024-12-20:
Also, when trying to attach a Sketch to the face of a wall (selected in the 3D view), I get:
[image]

maxwxyz, 2024-12-20:
It is also not possible to create e.g. a Facebinder in Draft from a 3D selection of a face of a wall if it is nested inside a Site object. The binder is empty then.

maxwxyz, 2024-12-20:
What might be related: When I try to hide an object I have selected in the 3D view by pressing the spacebar, the entire site is hidden, as each selection just selects the site object at root level.
It has no difference on what I select here:
[image]

maxwxyz, 2024-12-20:
Also, when I select an object to substract from a wall which is already inside the BIM side (selected in 3D view) and then select the wall from which it should be selected and click on `Arch_Remove`, the entire BIM site is added in the substraction group.

I'm really not sure if I am doing it all wrong or if BIM simply does not support selection in the 3D view?

semhustej, 2024-12-20:
There seems to be something wrong with Site/Building/Level containers in the current weekly. All these problems might be related to that.

Steps to reproduce (I am able to reproduce each time, also in Safe Mode):

- Open new project
- start Arch_Building (or Arch_Site, Arch_level)
- error shows up

<details><summary>Error messsage</summary>

```
20:46:27  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39673-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_observer.py", line 128, in slotRemoveDynamicProperty
    from nativeifc import ifc_psets
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39673-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39673-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_psets.py", line 29, in <module>
    from nativeifc import ifc_tools
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39673-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39673-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_tools.py", line 34, in <module>
    from ifcopenshell import geom
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39673-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39673-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\geom\__init__.py", line 55, in <module>
    from . import occ_utils as utils
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39673-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_imp<class 'ModuleNotFoundError'>: No module named 'typing_extensions'
20:46:27  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39673-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_observer.py", line 128, in slotRemoveDynamicProperty
    from nativeifc import ifc_psets
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39673-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39673-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_psets.py", line 29, in <module>
    from nativeifc import ifc_tools
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39673-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39673-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_tools.py", line 34, in <module>
    from ifcopenshell import geom
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39673-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39673-conda-Windows-x86_64-py311\bin\Lib\site-packages\ifcopenshell\geom\__init__.py", line 55, in <module>
    from . import occ_utils as utils
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39673-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_imp<class 'ModuleNotFoundError'>: No module named 'typing_extensions'
20:46:27  pyException: Traceback (most recent call last):
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39673-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_observer.py", line 128, in slotRemoveDynamicProperty
    from nativeifc import ifc_psets
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39673-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39673-conda-Windows-x86_64-py311\Mod\BIM\nativeifc\ifc_psets.py", line 29, in <module>
    from nativeifc import ifc_tools
[... cut ...]
  File "C:\Users\semhu\Documents\tyo\BF\FreeCAD\FC_daily\FreeCAD_weekly-builds-39673-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_imp<class 'ModuleNotFoundError'>: No module named 'typing_extensions'

```

</details>

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39673 (Git) Conda
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

```

Syres916, 2024-12-20:
Surely this is a packaging issue, if the user has installed IFCOpenShell 0.8.0 properly, the total components installed is as per selected in the screenshot:

[image]

following your steps no errors are raised, I'm sure I've read in the recent past that Conda hasn't caught up and therefore the IFCOpenShell package is older than 0.8.0.

maxwxyz, 2024-12-21:
I'm not using IFC mode though, I see only some IFC tags in the objects.

furgo16, 2024-12-21:
> There seems to be something wrong with Site/Building/Level containers in the current weekly. All these problems might be related to that.

I cannot reproduce this one in the daily:

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39709 (Git) Snap 1231
Build type: Release
Branch: main
Hash: cad2f33b4e02b92659de3fb27f686f91c94b447f
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.30
  * ezydxf
  * Assembly4 0.50.15
```

```python
>>> import ifcopenshell
>>> ifcopenshell.version
'0.8.0'
```

furgo16, 2024-12-21:
@maxwxyz I could not reproduce the issue on the latest daily on Linux (see version on my last comment). Following the steps in the screencast I could create both the face via the Part workbench and the sketch via the Sketcher workbench with no errors or warnings.

maxwxyz, 2024-12-21:
I can confirm the issue is solved with the latest weekly.
icfopenshell.version shows 0.0.0 
[image]

```
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * CfdOF 1.29.4
  * CurvedShapes 1.0.13
  * Curves 0.6.52
  * dodo 1.0.1
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * FreeCAD_SketchArch
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.6.2
  * Silk 0.1.5
```

closed 2024-12-21 by maxwxyz