FreeCAD/FreeCAD #18645: git source, addon manager not loading

clytle374, 2024-12-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD 1.1.0, Libs: 1.1.0devR39681 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

During initialization the error "No module named 'PySide2'" occurred in /home/cory/freecad-test/Mod/AddonManager/InitGui.py
Please look into the log file for further information
Unknown command 'Std_AddonMgr'
Thanks
[Top](https://forum.freecad.org/viewtopic.php?style=3&p=798681#top)

### Full version info

```shell
At least the last 10 git commits.
```

### Subproject(s) affected?

Addon Manager

### Anything else?

https://forum.freecad.org/viewtopic.php?style=3&p=798681#p798681

https://github.com/FreeCAD/FreeCAD/pull/18634

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2025-02-02:
@clytle374 is this still going on ?   
Could you please add your full [About](https://wiki.freecad.org) info to the OP?  
`At least the last 10 git commits.` means nothing retroactively.

clytle374, 2025-02-02:
> [@clytle374](https://github.com/clytle374) is this still going on ? Could you please add your full [About](https://wiki.freecad.org) info to the OP? `At least the last 10 git commits.` means nothing retroactively.

Still getting the same error on startup and addon manager is still missing.  

`OS: Gentoo Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.40041 (Git)
Build type: Release
Branch: main
Hash: 64a4f6e3669279e5dbe9cb59d8e0123eaeb70833
Python 3.12.8, Qt 6.8.1, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical/physical DPI: 96/128.093
Installed mods: 
  * fasteners 0.5.31
  * kicadStepUpMod 11.4.1
`

PR #19388 (2025-02-04): Addon Manager: Use FreeCAD QUiLoader wrapper when available
Replaces #18634 so may fix #18645
closed 2025-02-04 by chennes