FreeCAD/FreeCAD #18647: "GetToolBarItems" does not return newly added datum  button in structure toolbar

APEbbers, 2024-12-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When listing the toolbar items for a workbench, the datum button is not listed for the structure toolbar.
[image]
[image]

I'm working on the FreeCAD Ribbon, I use this line of code, to retrieve all the toolbars and commands for a workbench.

### Full version info

```shell
OS: Windows 11 build 22635
Architecture: x86_64
Version: 1.1.0dev.39681 (Git)
Build type: Release
Branch: main
Hash: 0609fa2335b0a4a3277875a986618c226a3edbe1
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.0
Locale: English/Netherlands (en_NL)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * A2plus 0.4.68
  * Assembly3 0.12.2 (Disabled)
  * Assembly4 0.50.15
  * BillOfMaterials 1.0.2 (Disabled)
  * boltsfc 2022.11.5
  * CfdOF 1.29.4
  * Curves 0.6.52
  * DynamicData 2.70.0
  * ExtremeProDark 2.9.0
  * fasteners 0.5.33
  * FEMbyGEN 2.3.0 (Disabled)
  * FEM_FrontISTR 0.1.0
  * frame 0.1.1
  * FreeCAD-Beginner-Assistant 1.0.0
  * FreeCAD-Ribbon 1.5.0
  * freecad.gears 1.2.0
  * IconThemes (Disabled)
  * Launcher
  * NikraDAP 2.0.0
  * OpenTheme 2024.9.1
  * PieMenu 1.10.2
  * Plot 2024.11.26
  * ProDarkThemePreferencePack 1.0.0
  * Render 2024.12.15
  * SearchBar 1.2.1
  * sheetmetal 0.6.2
  * StemfieWB 0.2.0
  * ThreadProfile 1.92.0
  * WorkFeature-WB
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-21:
@PaddleStroke FYI

PaddleStroke, 2024-12-21:
The datum command is actually a Part workbench tool. It is being added to the structure toolbar which is a core toolbar.
I think this is why this happen.
I can't remember who introduced this behavior.
I think it may be the same for the measure tool in the view toolbar no?

APEbbers, 2024-12-21:
The measurement tool is indeed not showing as well. I noticed that on linux, the datum command is not showing at all. So it is maybe a work in progress?

Syres916, 2024-12-22:
@APEbbers seems logical when you look at the code, the first line in `Workbench::getToolbarItems()` is `setupToolBars()` which has the standard setup https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/Workbench.cpp#L771-L830 and this does not include the Datum and Measurement tools.

APEbbers, 2024-12-22:
Yes that's true. However you would expect that all buttons of a toolbar should be returned, Maybe it these buttons can be added to the standard setup?

Syres916, 2024-12-22:
> Maybe it these buttons can be added to the standard setup?

You have to remember not all users build FreeCAD with all CMake options set to True so BUiLD_MEASURE and BUILD_PART will have to set as conditions. If I get chance I'll see if I can get it to work later this evening.

Syres916, 2024-12-22:
This seems to work: https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Fix_Structure_View_toolbars
@APEbbers are you able to compile and test?

This is top of the output I get:

`{'File': ['Std_New', 'Std_Open', 'Std_Save'], 'Edit': ['Std_Undo', 'Std_Redo', 'Separator', 'Std_Refresh'], 'Clipboard': ['Std_Cut', 'Std_Copy', 'Std_Paste'], 'Workbench': ['Std_Workbench'], 'Macro': ['Std_DlgMacroRecord', 'Std_DlgMacroExecute', 'Std_DlgMacroExecuteDirect'], 'View': ['Std_ViewFitAll', 'Std_ViewFitSelection', 'Std_ViewGroup', 'Std_AlignToSelection', 'Separator', 'Std_DrawStyle', 'Part_SelectFilter', 'Std_TreeViewActions', 'Std_Measure'], 'Individual views': ['Std_ViewIsometric', 'Std_ViewFront', 'Std_ViewTop', 'Std_ViewRight', 'Std_ViewRear', 'Std_ViewBottom', 'Std_ViewLeft'], 'Structure': ['Std_Part', 'Part_Datums', 'Std_Group', 'Std_LinkActions', 'Std_VarSet'], 'Help': ['Std_WhatsThis'], `

APEbbers, 2024-12-22:
Hi,

Sure no problem. I'll compile FreeCAd and come back to you

APEbbers, 2024-12-29:
Hi,

Sorry it took a while to get the build working. But I can confirm that it is working from my side as well.
Thanks!

Kind regards

Syres916, 2024-12-29:
OK, I checked the CI tests and I removed a little too much so I need to put back some empty functions and will submit a PR later this evening.

Syres916, 2024-12-31:
@APEbbers as the PR is not an acceptable solution, I'll have to revisit it in mid January.

APEbbers, 2024-12-31:
Ok. no problem. I have a workaround present for the time being