FreeCAD/FreeCAD #18649: Settings: tree view options should not be scattered among different setting pages

furgo16, 2024-12-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]

See attached screenshot. Having related tree view settings in multiple different pages makes it unnecessarily difficult to find those settings. When compared to those in the UI page, there is nothing advanced in the tree view settings that live within the Advanced page.

A better user experience would be to move the advanced tree view settings to the UI page.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39709 (Git) Snap 1231
Build type: Release
Branch: main
Hash: cad2f33b4e02b92659de3fb27f686f91c94b447f
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.30
  * ezydxf
  * Assembly4 0.50.15
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-12-22:
I found out this related issue: https://github.com/FreeCAD/FreeCAD/issues/14412. There was some rework to improve the UI preference pages. If I understand it correctly, there should be another review post-1.0.

In any case, as a small improvement, I would argue that at least the font size should stay together with the rest of UI settings.

Tiago-Almeida007, 2025-05-08:
Has this issue been solved? if not i could work on it

commit 0d48351c (2025-09-27): Gui: Organize tree view preferences (#21867)

* feat #18649: Organize tree options in one single preference page
-Removed tree view related code from all DlgSettingsAdvanced files
-Added Font size and Item Background Padding to DlgSettingsUI

* fix #18649: Organize tree options in one single preference page
-Added new line at the end of file DlgSettingsAdvanced.cpp
-Added new line at the end of file DlgSettingsAdvanced.py
-Added new line at the end of file DlgSettingsUI.ui
-Delete trailing space on file DlgSettingsAdvanced.py
-Delete extra line on file DlgSettingsAdvanced.h

* Gui: Remove over [... cut]
closed 2025-09-27 by chennes