FreeCAD/FreeCAD #18650: Show ifcopenshell version in About info

furgo16, 2024-12-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I've seen a few issues related to the version of ifcopenshell, which is becoming more and more a crucial part of BIM (even if it's AFAIK an optional dependency, yet shipped by default in some FreeCAD packages). Inevitably, at some point there is a question of which ifcopenshell is being used. While easy to find out, it's not obvious, and generally there is a back and forth in the comments asking to provide the version.

This back and forth (and issue close time) could be reduced if the ifcopenshell version were shown in the About info. At the minimum, in the textual version, but arguably it could also be useful to show it additionally on the About dialog.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.39709 (Git) Snap 1231
Build type: Release
Branch: main
Hash: cad2f33b4e02b92659de3fb27f686f91c94b447f
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * fasteners 0.5.30
  * ezydxf
  * Assembly4 0.50.15
```

### Subproject(s) affected?

BIM

### Anything else?

To provide the ifcopenshell version:

1. Open Python panel (View > Panels > Python console)
2. Type the Python code below
   ```python
   >>> import ifcopenshell
   >>> ifcopenshell.version
   '0.8.0'
   ```

This issue could have benefitted from displaying the ifcopenshell version: https://github.com/FreeCAD/FreeCAD/issues/18639

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

semhustej, 2024-12-21:
Just FYI, there is currently an IfcOpenShell version reporting bug on Windows builds: https://github.com/FreeCAD/FreeCAD/issues/18141

I agree that showing IfcOpenShell version info would be useful.

yorikvanhavre, 2024-12-23:
Good idea indeed! I'll have a look at how to do that

furgo16, 2025-01-25:
Related PRs adding information to About info, for reference:
- https://github.com/FreeCAD/FreeCAD/pull/16281
- https://github.com/FreeCAD/FreeCAD/pull/16492

closed 2025-02-05 by yorikvanhavre

semhustej, 2025-02-18:
I think IfcOpenShell should also be mentioned in the 'Libraries' section.

Proposed info:

**IfcOpenShell**
Version: -.-.-
Web site: https://ifcopenshell.org/
License: LGPL 3.0: https://github.com/IfcOpenShell/IfcOpenShell?tab=LGPL-3.0-1-ov-file

[image]

[image]

luzpaz, 2025-05-08:
soft bump

furgo16, 2025-05-08:
I think it would be best to open a new issue and realise this one. The linked PR fixed the current issue already. The follow up comment makes sense, but it is a new feature that I feel should be filed and fixed separately.

Reusing an existing issue for multiple different purposes leads to confusion in my opinion.

If no one disagrees, I'll close this issue a week from now on.

luzpaz, 2025-05-27:
> ifcopenshell is LGPL3, not LGPL2.1+ so including it would probably not be a good idea.
My understanding is that FreeCAD would have to be distributed as LGPL3.0/GPL3.0 if we include it.

> We could improve the error message and create an addon that makes it easy to install latest ifcopenshell (would just be an empty package that tells FC to install it through pip)

https://github.com/FreeCAD/FreeCAD/pull/21204#issuecomment-2912455386

semhustej, 2025-05-28:
> BTW, [@furgo16](https://github.com/furgo16) and [@semhustej](https://github.com/semhustej) I just added both of you to the "Developers" team of FreeCAD here on GitHub. With that, you can now officially review, approve or request changes to other people's PRs. I encourage you (no obligation or course ;) ) to use this and help reviewing PRs. When PRs are reviewed and approved, they get through the merge queue much faster.

Thanks, Yorik. I don't feel like I have the knowledge to do such work, but maybe in the future :)