FreeCAD/FreeCAD #18654: BIM: BIMExample crashes FreeCAD on open

maxwxyz, 2024-12-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This is new in today's weekly, last weekly was good.

Steps to reproduce:
1. Open FreeCAD
2. Open the BIMExample.FCStd
3. Crash without warning

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * CfdOF 1.29.4
  * CurvedShapes 1.0.13
  * Curves 0.6.52
  * dodo 1.0.1
  * fasteners 0.5.33
  * freecad.gears 1.3.0
  * FreeCAD_SketchArch
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.26
  * Rocket 4.0.1
  * sheetmetal 0.6.2
  * Silk 0.1.5
```

### Subproject(s) affected?

BIM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-12-21:
Same Hash, self-compiled build opens BIMExample error free.

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39708 (Git)
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.12.3, Qt 5.15.13, Coin 4.0.3, Vtk 9.1.0, OCC 7.8.2.dev
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Defeaturing 1.2.2
  * sheetmetal 0.6.2
  * Behave-Dark-Colors 0.1.1
  * 3dfindit-freecad-integration 1.2.0
```

Syres916, 2024-12-21:
No issues using the AppImage either:

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

maxwxyz, 2024-12-21:
It works in safe mode but I am not sure which preference or addon crashes it.

semhustej, 2024-12-21:
I am not able to reproduce the issue.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

```

maxwxyz, 2024-12-21:
What is weird, all other BIM files work. Disabling SketchArch makes the file usable. I have updated to the latest SketchArch but it still crashes FreeCAD, and loading BIM projects takes significantly longer.
@paullee0 FYI

Shall I close this issue here? IMO FreeCAD should not crash without an error/warning, even by a third party addon.

Syres916, 2024-12-21:
OK, back to my self-compiled build with the latest SketchArch installed, yep I agree with the massive impact on loading the file, I still don't get a crash but I do get some familiar output:

```
19:14:14  pyException: Traceback (most recent call last):
  File "/home/john/.local/share/FreeCAD/Mod/FreeCAD_SketchArch/ArchCurtainWall.py", line 691, in onChanged
    self.colorize(vobj.Object)
  File "/home/john/.local/share/FreeCAD/Mod/FreeCAD_SketchArch/ArchCurtainWall.py", line 704, in colorize
    panelcolor = ArchCommands.getDefaultColor("WindowGlass")
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/john/freecad-daily-build-782-403/Mod/BIM/ArchCommands.py", line 73, in getDefaultColor
    c = params.get_param_arch("WindowGlassColor")
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/john/freecad-daily-build-782-403/Mod/Draft/draftutils/params.py", line 647, in get_param_arch
    return get_param(entry, path="Mod/Arch", ret_default=ret_default)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'RecursionError'>: maximum recursion depth exceeded
19:14:14  pyException: Traceback (most recent call last):
  File "/home/john/.local/share/FreeCAD/Mod/FreeCAD_SketchArch/ArchCurtainWall.py", line 691, in onChanged
    self.colorize(vobj.Object)
  File "/home/john/.local/share/FreeCAD/Mod/FreeCAD_SketchArch/ArchCurtainWall.py", line 704, in colorize
    panelcolor = ArchCommands.getDefaultColor("WindowGlass")
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/john/freecad-daily-build-782-403/Mod/BIM/ArchCommands.py", line 73, in getDefaultColor
    c = params.get_param_arch("WindowGlassColor")
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/john/freecad-daily-build-782-403/Mod/Draft/draftutils/params.py", line 647, in get_param_arch
    return get_param(entry, path="Mod/Arch", ret_default=ret_default)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'RecursionError'>: maximum recursion depth exceeded
19:14:14  pyException: Traceback (most recent call last):
  File "/home/john/.local/share/FreeCAD/Mod/FreeCAD_SketchArch/ArchCurtainWall.py", line 691, in onChanged
    self.colorize(vobj.Object)
  File "/home/john/.local/share/FreeCAD/Mod/FreeCAD_SketchArch/ArchCurtainWall.py", line 704, in colorize
    panelcolor = ArchCommands.getDefaultColor("WindowGlass")
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/john/freecad-daily-build-782-403/Mod/BIM/ArchCommands.py", line 73, in getDefaultColor
    c = params.get_param_arch("WindowGlassColor")
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/john/freecad-daily-build-782-403/Mod/Draft/draftutils/params.py", line 647, in get_param_arch
    return get_param(entry, path="Mod/Arch", ret_default=ret_default)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'RecursionError'>: maximum recursion depth exceeded
19:14:14  pyException: Traceback (most recent call last):
  File "/home/john/.local/share/FreeCAD/Mod/FreeCAD_SketchArch/ArchCurtainWall.py", line 691, in onChanged
    self.colorize(vobj.Object)
  File "/home/john/.local/share/FreeCAD/Mod/FreeCAD_SketchArch/ArchCurtainWall.py", line 704, in colorize
    panelcolor = ArchCommands.getDefaultColor("WindowGlass")
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/john/freecad-daily-build-782-403/Mod/BIM/ArchCommands.py", line 73, in getDefaultColor
    c = params.get_param_arch("WindowGlassColor")
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/john/freecad-daily-build-782-403/Mod/Draft/draftutils/params.py", line 647, in get_param_arch
    return get_param(entry, path="Mod/Arch", ret_default=ret_default)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'RecursionError'>: maximum recursion depth exceeded
19:14:14  pyException: Traceback (most recent call last):
  File "/home/john/.local/share/FreeCAD/Mod/FreeCAD_SketchArch/ArchCurtainWall.py", line 691, in onChanged
    self.colorize(vobj.Object)
  File "/home/john/.local/share/FreeCAD/Mod/FreeCAD_SketchArch/ArchCurtainWall.py", line 704, in colorize
    panelcolor = ArchCommands.getDefaultColor("WindowGlass")
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/john/freecad-daily-build-782-403/Mod/BIM/ArchCommands.py", line 73, in getDefaultColor
    c = params.get_param_arch("WindowGlassColor")
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/john/freecad-daily-build-782-403/Mod/Draft/draftutils/params.py", line 647, in get_param_arch
    return get_param(entry, path="Mod/Arch", ret_default=ret_default)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'RecursionError'>: maximum recursion depth exceeded

```

So chances are that @paullee0 will need to bring in the fixes (whenever I can find them) into the SketchArch addon or possibly look into where in the code it's getting triggered and remove a block altogether.

Syres916, 2024-12-21:
@paullee0 have a look at https://github.com/FreeCAD/FreeCAD/commit/0607c555d6c56d1b617dc0d3a52431bef562c7dc#diff-b4849bbc0941a0f60f25d13c40d4bfeca94171f55bd78a6196362343c3fdcf6f basically it appears as though your code is in conflict with the `ArchCurtainWall` in core particularly around the `panelcolor` but possibly the `basecolor` also.

Syres916, 2024-12-21:
Yep, renaming `ArchCurtanWall.py` to `ArchCurtainWall_SA.py` enables the file to load error free and a lot quicker.

paullee0, 2024-12-22:
OK, I have 'rebased' ArchCurtainWall.py after your findings above.

Anyone can try testing again if it helps?  Thanks.

paullee0, 2024-12-22:
@maxwxyz  Out of curiosity, you test all the add-ons to ensure they are good with FreeCAD (SketchArch is not on FC add-on list yet though)?  Or are you testing some specific aspect of these add-on?

It is usually good on my side, so good to have others to verify :)

maxwxyz, 2024-12-22:
@paullee0 no, I just wanted to check out how to work with BIM over the holidays.
I can confirm it now works.

closed 2024-12-22 by maxwxyz

paullee0, 2024-12-22:
Likewise, commits done in ArchWall.py and ArchStructure.py in SketchArch repository already merged in in FreeCAD/FreeCAD, the latter is latest currently.

Deleted ArchWall.py and ArchStructure.py in SketchArch repository now, so users always use the latest in FreeCAD/FreeCAD.