FreeCAD/FreeCAD #18656: Regression - Groove not Performed Correctly on One Side of a Profile

point-droid, 2024-12-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Please see the attached file. This was created in FC 0.21.1. Both grooves are as expected. Now open the file in FC1.0 release. The groove on the positive Y direction is not performed. This is not related to the recalculation from an older version of FC as the same issue occurs if the sketches are drawn from scratch in FC 1.0

[Grooves_0.21.zip](https://github.com/user-attachments/files/18219454/Grooves_0.21.zip)

### Full version info

```shell
OS: openSUSE Leap 15.5 (KDE//usr/share/xsessions/plasma5/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * CurvedShapes 1.0.13
  * Curves 0.6.52
  * freecad.gears 1.3.0
```

### Subproject(s) affected?

None

### Anything else?

A discussion about this can be found [here](https://forum.freecad.org/viewtopic.php?t=93052). The last post states that this problem was introduced between 38728 and 39109

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-21:
Trying with 1.0.0 and dev but do not saw any issue

https://github.com/user-attachments/assets/963e2999-7264-4ee7-9dbd-33ef1b257c05

<details>
  <summary>FreeCAD 1.0.0.39109 (Git) makepkg, Arch Linux (KDE/plasma/wayland)</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods:lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.1; Silk 0.1.5; Curves 0.6.51

```
</details>

<details>
  <summary>FreeCAD 1.1.0dev.39673 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39673 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 1289997dfd604cb24f060f905f04ec3728d964e4
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.2; Silk 0.1.5; Curves 0.6.52

```
</details>

point-droid, 2024-12-21:
Strange - I and at least 2 others on different systems have seen this

https://github.com/user-attachments/assets/280e8178-8af5-42d9-b8d3-d556733a3999

tarman3, 2024-12-22:
Maybe you attached different file

kadet1090, 2025-12-15:
@pieterhijma can you check it with different OCC versions? It either would be a bug with tnp mitigation or OCC.

pieterhijma, 2025-12-17:
I can confirm on FreeCAD `main` as well with OCC 7.9.3:

```
OS: Arch Linux (tty)
Architecture: x86_64
Version: 1.2.0dev.44683 (Git)
Build date: 2025/12/17 10:01:02
Build type: Unknown
Branch: main
Hash: e147e5a260a9621a9b2f49d3111e784d033ec3a6
Python 3.13.7, Qt 6.10.1, Coin 4.0.5, Vtk 9.5.2, boost 1_89, Eigen3 5.0.1, PySide 6.10.1
shiboken 6.10.1, SMESH 7.7.1.0, xerces-c 3.3.0, OCC 7.9.3
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1.33333
Installed mods: 
  * OpenTheme 2024.5.16
  * fasteners 0.5.34
  * WebTools 1.0.0
  * osh-autodoc-workbench 0.2.3
  * DynamicData 2.70.0
  * Curves 0.6.61
  * freecad.gears 1.3.0
  * sheetmetal 0.7.24
  * Ondsel-Lens 2025.11.21.01
  * BCFPlugin 1.0.0
```

Now with different OCC versions.

pieterhijma, 2026-03-28:
@kadet1090, I'm working on this one. This is another instance of Part Design where the preview is correct, but the resulting geometry not.