FreeCAD/FreeCAD #18657: Chamfer Does not Change units

Walt8, 2024-12-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using the chamfer command in FreeCAD 1.0, it always comes up with a size in "thou" when i am in "US Customary". When I change it to inches ("), it still uses "thou". When I continue I get an error "BRep:_Comand Not Done". 

### Full version info

```shell
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * Assembly4 0.50.15
  * fasteners 0.5.33
  * QuickMeasure 2022.10.28
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-21:
Can you share an example file and steps how to reproduce? Did you try other imperial unit systems in FreeCAD preferences?

FEA-eng, 2026-08-27:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 26.3.0dev.20260825 (Git shallow)
Build date: 2026/08/25 17:50:34
Build type: Release
Branch: (HEAD detached at 964fce2ed)
Hash: 964fce2edc9a7e1905c2fd3fcd6e7be90afc8375
Python 3.11.14, Qt 6.8.3, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, Clipper2 , IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL) [ OS: Polish/Poland (pl_PL) ]
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```