FreeCAD/FreeCAD #18658: Self intersecting Part::Sweep

rostskadat, 2024-12-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello FreeCAD Team!

It seems that in some circumstances the Part::Sweep does not behave as expected and creates a self intersecting sweep.
The following python code shows the problem. It creates a section at the start, a section at the end, and a straight line as the spine of the sweep.

The strange thing is that depending on the size of the section the sweep will be correct or not, demonstrating that the order of the edges in each section is not the problem (as far as I can tell).

```python

import Draft
import Part
import FreeCAD as App

# Just used to colour edges to make ordering easier to see
def hex2rgb(hexcode):
    return (int(hexcode[0:2], 16), int(hexcode[2:4], 16), int(hexcode[4:6], 16))

EDGES_COLORS = [hex2rgb("5bc0eb"), hex2rgb("fde74c"), hex2rgb("9bc53d"), hex2rgb("e55934"), hex2rgb("fa7921")]
Z = 2000
L = 1000

def create_sweep(dx:int):
    # the first section of the sweep
    s1p1 = App.Vector(-dx, dx, 0)
    s1p2 = App.Vector(dx, -dx, 0)
    s1p3 = App.Vector(dx, -dx, Z)
    s1p4 = App.Vector(-dx, dx, Z)

    section_start = Draft.makeRectangle([s1p1, s1p2, s1p3, s1p4])
    section_start.ViewObject.LineColorArray = EDGES_COLORS
    section_start.MakeFace = False

    # the second section of the sweep
    s2p1 = App.Vector(L+dx,  dx, 0)
    s2p2 = App.Vector(L-dx, -dx, 0)
    s2p3 = App.Vector(L-dx, -dx, Z)
    s2p4 = App.Vector(L+dx,  dx, Z)

    section_end = Draft.makeRectangle([s2p1, s2p2, s2p3, s2p4])
    section_end.ViewObject.LineColorArray = EDGES_COLORS
    section_start.MakeFace = False

    spine = Draft.makeLine(App.Vector(0, 0, 0), App.Vector(L, 0, 0))

    # Let's create the sweep...
    sweep = App.ActiveDocument.addObject('Part::Sweep')
    sweep.Sections = [section_start, section_end]
    sweep.Spine = spine
    sweep.Solid = True
    sweep.Frenet = False

    section_start.Visibility = False
    section_end.Visibility = False
    spine.Visibility = False

    App.ActiveDocument.recompute()

# create a good sweep
create_sweep(dx=100)

# create a self-intersecting sweep
create_sweep(dx=200)

```

When creating the `dx=200` sweep it self intersect as shown in the image below
[image]

I would have expected the sweep to give the same result as the  `dx=100` sweep as shown below 
[image]

Could you point me toward the problem, as I am not sure I understand what I'm doing wrong.

Thanks a lot for your help

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.0.0.38641 +468 (Git) Snap 1202
Build type: Release
Branch: tag: 1.0.0
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * FreeCAD-geodata2 0.1.0
  * ExplodedAssembly
  * Help 1.0.3
  * A2plus 0.4.67
  * FreeCAD-SH3D 0.0.1
  * ArchTextures
  * Render 2024.9.25
  * lattice2 1.0.0
  * Curves 0.6.37
  * Assembly4 0.50.15
  * fasteners 0.5.29
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-12-22:
@mwganson any chance you can weigh-in here ?

mwganson, 2024-12-23:
I don't know why it's doing it.  If you're interested in a workaround, in the Gui select Rectangle003 (or Rectangle002, but not both) and in the Part menu -> reverse shapes.  Then edit the Sweep001 Sections property to replace the original rectangle with the reversed one.

rostskadat, 2025-01-10:
@mwganson  thanks for your insight. Its seems however that the workaround you are proposing is not *always* working. I can confirm that in the case of the Sweep created from the script above the workaround does work. However I have other example where it fails. Allow me to attach the an example.

I'm at a loss as of why this happens or in which condition it does work.
Do you happen to know of a way to create a Sweep as close as possible to the OCC backend?
Maybe a C++ piece of code... to maybe illustrate the issue with OCC...

[TwistedSweep-failed-workaround.zip](https://github.com/user-attachments/files/18379809/TwistedSweep-failed-workaround.zip)

mwganson, 2025-01-10:
PartDesign additive pipe also fails here.  I tried on OCCT 7.6.3 and OCCT 7.8.1, so probably is not an OCCT regression.  Did you try on 0.21?  I no longer have that version installed.  This could perhaps be TNP mitigation code related.

If you just want to get the project done try creating 2 ruled surfaces from the top and bottom edges and then loft the ruled surfaces to one another.  Alternatively, create surfaces from appropriate edges in Surface workbench, and then use the parametric solid tool in Curves workbench to make the solid.

rostskadat, 2025-01-11:
@mwganson I'm trying to get FreeCAD 0.21 up and running without success so far (the AppImage fails with some OpenGL problem). I'll keep you posted about my success regarding that. 
However I doubt that might be related to the TNP problem as even when using the `BRepOffsetAPI` as closely as possible (using the Part.BRepOffsetAPI.MakePipeShell) it also fails for `dx=200` while working for `dx=100`. 
Allow me to share the code that illustrate that:

```python
import Draft
import Part
from draftutils.messages import _err, _log, _msg, _wrn
import FreeCAD as App

doc = App.newDocument("TwistedSweep-18658-BRepOffestAPI")

def hex2rgb(hexcode):
    return (int(hexcode[0:2], 16), int(hexcode[2:4], 16), int(hexcode[4:6], 16))

def _pv(vect):
    return f"({round(getattr(vect, 'X', getattr(vect,'x')))},{round(getattr(vect, 'Y', getattr(vect,'y')))})"

def _ps(section):
    v = section.Shape.Vertexes
    return f"[{_pv(v[0].Point)}, {_pv(v[1].Point)}, {_pv(v[2].Point)}, {_pv(v[3].Point)}]"

EDGES_COLORS = [hex2rgb("5bc0eb"), hex2rgb("fde74c"), hex2rgb("9bc53d"), hex2rgb("e55934"), hex2rgb("fa7921")]
Z = 2000
L = 1000

def create_shell(dx:int):
    s1p1 = App.Vector(-dx, dx, 0)
    s1p2 = App.Vector(dx, -dx, 0)
    s1p3 = App.Vector(dx, -dx, Z)
    s1p4 = App.Vector(-dx, dx, Z)
    section_start = Draft.makeRectangle([s1p1, s1p2, s1p3, s1p4], face=False)
    section_start.ViewObject.LineColorArray = EDGES_COLORS
    section_start.recompute()
    s2p1 = App.Vector(L+dx,  dx, 0)
    s2p2 = App.Vector(L-dx, -dx, 0)
    s2p3 = App.Vector(L-dx, -dx, Z)
    s2p4 = App.Vector(L+dx,  dx, Z)
    section_end = Draft.makeRectangle([s2p1, s2p2, s2p3, s2p4], face=False)
    section_end.ViewObject.LineColorArray = EDGES_COLORS
    section_end.recompute()
    spine = Draft.makeWire([App.Vector(0, 0, 0), App.Vector(L, 0, 0)])
    spine.recompute()
    shell = Part.BRepOffsetAPI.MakePipeShell(spine.Shape)
    shell.setFrenetMode(False)
    shell.setTransitionMode(1)
    shell.add(section_start.Shape)
    shell.add(section_end.Shape)
    if not shell.isReady():
        print(f"WARNING: shell is not ready!")

    shell.build()
    shell.makeSolid()
    shape = shell.shape()
    feature = App.ActiveDocument.addObject('Part::Feature')
    feature.Shape = shape
    feature.Label = f"Shell-{dx}"

# Shell is OK
create_shell(dx=100)

# Shell is Twisted
create_shell(dx=200)

```

As far as I can tell there is nothing really between the Py interface and the OCC library (cf. `FreeCAD/src/Mod/Part/App/BRepOffsetAPI_MakePipeShellPyImp.cpp`)

As for the reason I'm using the sweep, it is simple: I'm trying to import walls that can be curved (whose spine is an arc) but also whose start and end sections might have different height. The `Part::Sweep` felt like the easiest and most complete tool to do the job (and it would if not for that strange behavior). Since the whole thing is for the BIM workbench, I might actually be limited when it comes to dependencies toward external workbenches which are not in the stock distribution...

I'll continue investigating. Do you have any suggestion which attribute of the `shell.Shape` might be relevant or indicative of the problem?

rostskadat, 2025-01-11:
@mwganson  BTW I'm using a docker container to isolate by build:

https://github.com/rostskadat/FreeCAD-Docker/

However I'm not sure if that will be OK to build FreeCAD-0-21.

rostskadat, 2025-01-11:
@mwganson  I can confirm that the same problem occur in 0.21.2 (tip of releases/FreeCAD-0-21), both while using the `Part::Sweep` and the `Part.BRepOffsetAPI.MakePipeShell` infrastructure. 

Could we safely assume it is therefore a problem with the underlying library?

mwganson, 2025-01-12:
Good detective work.  I think it's an upstream OCCT bug.  It also fails using Part.Wire.makePipeShell()

`sweep = Line.Shape.makePipeShell([Rectangle002.Shape, Rectangle003.Shape], True, False, 0) #Line is the path, solid = True, Frenet = False, Transition mode = 0`

I tried in the Gui to rotate one of the rectangles about the z axis at its own center of mass, but even using the rotated rectangle it still crisscrossed during the pipe creation.

If you can create all the edges, then you can make the faces and the solid without needing any additional dependencies (Curves WB or Surface WB).    Make the wires, faces from the wires, a shell from the faces, and finally a solid from the shell.

rostskadat, 2025-01-12:
> Make the wires, faces from the wires, a shell from the faces, and finally a solid from the shell.

Thanks for your suggestion. However, that only solved my problem for straight walls, as I also have curved walls (i.e. like a silo). I'll implement this workaround when the sweep fails and see how it looks.

codeotaku, 2026-08-18:
Came across this issue when investigating something else.  However, I can't seem to reproduce the problem currently (running the python script as reported)? 

[image]

[image]

Perhaps this was resolved at some point?