FreeCAD/FreeCAD #18660: B-Spline 'Access violation'

DeeKay789, 2024-12-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

>> This particular 'Access violation' also disables FC 'Copy to Clipboard' (data is not passed to the Windows clipboard). Nevertheless I do have saved versions listed here along with step-by-step recreation and video. There's no macro as FC has fully crashed so nothing can be saved (afaik)
<<

Problem: When concluding a B-spline with the two sketcher 'continuation' flags unset, FC crashes with an 'access violation' error.

Below are the steps I took to replicate the crash (reliably and consistently). 

**  The steps I use below does not crash FC when started in Safe Mode **

New document
Save as (whatever)
Add Body
Create sketch on XZ plane
Create a B-Spline using control points constrained to the +'ve X axis
I used 4 control points.
End the B-spline operation near the Y axis. Right click or press escape.
This will crash FreeCAD and render the file unrecoverable.
FreeCAD will need to be restarted as when any file is subsequently loaded the error reappears.

The following text seems to appear all at once in the report window.
>>
[X] Unhandled Base:: Exception caught in GUIApplication::notify
The error message is: Access violation (xx times)
[X] Unhandled exceception in ViewProvider::eventCallback: Access violation
(Event type: SoKeyboardEvent, object type: SketcherGui::ViewProviderSketch)
[X] Extension is not a python addable version: 'Gui::ViewProviderGeoFeatureGroupExtension'
[X] Cannot create object 'Origin001': (App::Point' is not a document object type)
<<

The screen is blank (all data disappears).
If a save is tried a small dialogue box reconfirms the Access violation.
The 'About FreeCAD' fails to copy any data to the clipboard. It does not work after a restart FC , or, a restart of the machine.
As I said, In Safe Mode there is no crash. the 'About FC' 'Copy to Clipboard' does still work in Safe Mode' and gives the following:

OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion

So only a change to the FC supplied theme. No modules/macros loaded as far as I can tell.

I've also tested all the above on my last nightly build which was 39673. It too ends in 'Access violation'.

I posted a similar set of steps on the FC forum on 20/12/24 but no one replied (30 views as of this writing)

The attached video shows the B-Spline creation and conclusion ending correctly. I then toggle the 'Sketcher>Display' continuation flags and repeat the steps which end in the Access violation crash.

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Subproject(s) affected?

Sketcher

### Anything else?

https://github.com/user-attachments/assets/38ab2762-5642-4e33-b388-b97da14298f7

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-12-21:
Confirmed, here's the output on Linux:

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x7f7be2445320]
#1  0x7f7be5ee9aba in Gui::ToolHandler::unsetCursor() from /home/username/freecad-101-build/lib/libFreeCADGui.so+0x2a
#2  0x7f7ba1eec32c in SketcherGui::DrawSketchDefaultHandler<SketcherGui::DrawSketchHandlerBSpline, SketcherGui::StateMachines::TwoSeekEnd, 2, SketcherGui::ConstructionMethods::BSplineConstructionMethod>::finish() from /home/username/freecad-101-build/Mod/Sketcher/SketcherGui.so+0x1c
#3  0x7f7ba205ce9c in SketcherGui::ViewProviderSketch::keyPressed(bool, int) from /home/username/freecad-101-build/Mod/Sketcher/SketcherGui.so+0x9c
#4  0x7f7be638b9ac in Gui::ViewProvider::eventCallback(void*, SoEventCallback*) from /home/username/freecad-101-build/lib/libFreeCADGui.so+0x4dc
#5  0x7f7be1e805c7 in SoEventCallback::handleEvent(SoHandleEventAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0xb7
#6  0x7f7be1c7e818 in SoAction::traverse(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x68
#7  0x7f7be1e38628 in SoChildList::traverse(SoAction*, int, int) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x188
#8  0x7f7be1e8a8a1 in SoGroup::doAction(SoAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x51
#9  0x7f7be1ea9dbc in SoSeparator::doAction(SoAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x2c
#10  0x7f7be1c7e818 in SoAction::traverse(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x68
#11  0x7f7be1e38628 in SoChildList::traverse(SoAction*, int, int) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x188
#12  0x7f7be1e8a8a1 in SoGroup::doAction(SoAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x51
#13  0x7f7be1ea9dbc in SoSeparator::doAction(SoAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x2c
#14  0x7f7be1c7e818 in SoAction::traverse(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x68
#15  0x7f7be1c8c15b in SoHandleEventAction::beginTraversal(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0xcb
#16  0x7f7be1c7ea0c in SoAction::apply(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0xcc
#17  0x7f7be1e56028 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x78
#18  0x7f7be62fe902 in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /home/username/freecad-101-build/lib/libFreeCADGui.so+0x22
#19  0x7f7be630d118 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) from /home/username/freecad-101-build/lib/libFreeCADGui.so+0x58
#20  0x7f7be62f5305 in SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) from /home/username/freecad-101-build/lib/libFreeCADGui.so+0x175
#21  0x7f7be2ed7e82 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x92
#22  0x7f7be3b6bd35 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x75
#23  0x7f7be3b740e9 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xcc9
#24  0x7f7be5eabcc8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/username/freecad-101-build/lib/libFreeCADGui.so+0xc8
#25  0x7f7be2ed8118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x128
#26  0x7f7be2ed7e82 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x92
#27  0x7f7be3b6bd35 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x75
#28  0x7f7be3b740e9 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xcc9
#29  0x7f7be5eabcc8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/username/freecad-101-build/lib/libFreeCADGui.so+0xc8
#30  0x7f7be2ed8118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x128
#31  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1ce00a) [0x7f7be3bce00a]
#32  0x7f7be3b6bd45 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x85
#33  0x7f7be5eabcc8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/username/freecad-101-build/lib/libFreeCADGui.so+0xc8
#34  0x7f7be2ed8118 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x128
#35  0x7f7be3341533 in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0xf3
#36  0x7f7be3317bfc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0xac
#37  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x75d06) [0x7f7bdcbfcd06]
#38  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x5d5b5) [0x7f7be11145b5]
#39  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xbc717) [0x7f7be1173717]
#40  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7f7be1113a53]
#41  0x7f7be2f35279 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x69
#42  0x7f7be2ed6a7b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x13b
#43  0x7f7be2edf3e8 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x98
[... cut ...]
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.0.0.39146 (Git)
Build type: Release
Branch: releases/FreeCAD-1-0
Hash: 4f9e833512cc8d867a349cb8b32137e5b895aa65
Python 3.12.3, Qt 5.15.13, Coin 4.0.3, Vtk 9.1.0, OCC 7.8.2.dev
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Defeaturing 1.2.2
  * sheetmetal 0.6.2
  * Behave-Dark-Colors 0.1.1
  * 3dfindit-freecad-integration 1.2.0
```

Syres916, 2024-12-21:
This is a regression around the middle of this year, last working build I have is:

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37302 (Git) AppImage
Build type: Release
Branch: main
Hash: 0e24e121eb3e5708c380596e0a9fd583befac977
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Defeaturing 1.2.2
  * sheetmetal 0.6.2
  * Behave-Dark-Colors 0.1.1
  * 3dfindit-freecad-integration 1.2.0
```

the first failing build I have is:

```
OS: Linux Mint 22 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git) AppImage
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Defeaturing 1.2.2
  * sheetmetal 0.6.2
  * Behave-Dark-Colors 0.1.1
  * 3dfindit-freecad-integration 1.2.0
```

luzpaz, 2025-08-22:
@DeeKay789 is this still happening on latest 1.1.0dev ? https://github.com/FreeCAD/FreeCAD/releases/tag/weekly-2025.08.20

DeeKay789, 2025-08-24:
On holiday atm but will test on my return. Thx

DeeKay789, 2025-08-29:
@luzpaz  Yup. Confirmed fixed using the dev version you quoted (1.1.0dev)

Thx :-)

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/08/20 00:00:55
Build type: Release
Branch: main
Hash: 5e4fe3a3c3af46be530969d323368fee8bfcba3b
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Navigation Style/Orbit Style/Rotation Mode: TinkerCAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/169.333/2
Installed mods: 
  * CurvedShapes 1.0.13
  * Curves 0.6.70
  * lattice2 1.0.0

closed 2025-08-29 by luzpaz