FreeCAD/FreeCAD #18664: Link objects have no getGlobalPlacementOf  method.

edwilliams16, 2024-12-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Link objects do not possess the getGlobalPlacementOf() method.  Since  getGlobalPlacement() is deprecated in favor of this method because it failed to properly support links, this seem to be an omission.
See [https://forum.freecad.org/viewtopic.php?p=798494#p798494](url) for a forum discussion.

### Full version info

```shell
OS: macOS 14.4.1
Architecture: x86_64
Version: 1.1.0dev.39503 (Git) Conda
Build type: Release
Branch: main
Hash: da45328f6f9b35cbbdc7167d7faadb6a07634bdb
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * MeshRemodel 1.10.22
  * DynamicData 2.62.0
  * OpenTheme 2024.9.1
  * fcgear 1.0.0
  * workfeature
  * Pyramids-and-Polyhedrons
  * QuickMeasure 2022.10.28
  * Manipulator 1.5.7
  * fasteners 0.5.28
  * lattice2 1.0.0
  * CurvedShapes 1.0.9
  * Alternate_OpenSCAD 1.0.0
  * sheetmetal 0.4.25
  * toSketch 1.0.1
  * Help 1.0.3
  * Curves 0.6.48
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2024-12-22:
The forum link above leads to: https://github.com/FreeCAD/FreeCAD/issues/url

Related:
 No globalPlacement() equivalent for LinkElement #13978 
 getGlobalPlacement does not work for linked objects. #11977

edwilliams16, 2024-12-22:
https://forum.freecad.org/viewtopic.php?p=798494#p798494

Apparently I'm not supposed to put url tags around the url.