FreeCAD/FreeCAD #18666: Freecad limits the length of the file and folder paths to 260 characters

PabloBeranger, 2024-12-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Hello,
I use Freecad 0.21.2 and 1.0.
Freecad limits the length of the file and folder paths to 260 characters while Windows 10 no longer has this limit.
I have a function that duplicates the .FCSTD files.
This function launched in Freecad generates the error "Filenotfounderror" while the same function launched in VSCODE does not generate any error.
Why freecad has this limitation and how to delete it.

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: French/France (fr_FR)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * 3DfindIT 1.2.0
  * 3D_Printing_Tools (Disabled)
  * A2plus 0.4.68
  * addFC 0.7.5
  * AirPlaneDesign 0.4.1 (Disabled)
  * Alternate_OpenSCAD 1.0.0
  * AnimationFreeCAD 1.0.0
  * ArchTextures
  * Assembly3 0.12.2
  * Assembly4 0.50.15
  * assembly_handbook
  * BCFPlugin 1.0.0
  * BillOfMaterials 1.0.2
  * BIM 2021.12.0
  * BIM021 2021.12.0
  * BIMBots
  * btl 0.9.9 (Disabled)
  * CadbaseLibrary 1.0.3 (Disabled)
  * cadquery_module 2.0.0 (Disabled)
  * Cfd (Disabled)
  * CfdOF 1.29.3 (Disabled)
  * CommandPanel (Disabled)
  * CurvedShapes 1.0.13 (Disabled)
  * Curves 0.6.51 (Disabled)
  * Defeaturing 1.2.2 (Disabled)
  * Design456 0.0.1 (Disabled)
  * DesignSPHysics 0.7.0(15-09-2023)
  * dodo 1.0.1 (Disabled)
  * dxf-library (Disabled)
  * DynamicData 2.70.0
  * EM 2.1.1 (Disabled)
  * Estimate 0.1.2
  * ExplodedAssembly
  * fasteners 0.5.33
  * Fcmcua 1.0.0
  * FeedsAndSpeeds 0.6.0 (Disabled)
  * FEMbyGEN 2.3.0
  * FEM_FrontISTR 0.1.0
  * frame 0.1.1
  * freecad.gears 1.3.0
  * GDML 2.0.1Beta (Disabled)
  * Glass (Disabled)
  * Help 1.0.3
  * ImportNURBS 1.1.0 (Disabled)
  * InventorLoader 1.5.1 (Disabled)
  * kerkythea (Disabled)
  * kicadStepUpMod 11.4.4 (Disabled)
  * lattice2 1.0.0 (Disabled)
  * Launcher (Disabled)
  * LCInterlocking
  * Lithophane (Disabled)
  * Manipulator 1.5.7
  * MeshRemodel 1.10.35
  * MnesarcoUtils 0.2.14
  * Movie 2024.2.15 (Disabled)
  * NavigationIndicator (Disabled)
  * Nodes 0.1.36
  * Nodes.backup1723381727.939789 0.1.36
  * nurbs (Disabled)
  * OpticsWorkbench 1.0.26 (Disabled)
  * OSAFE 2022.5.29 (Disabled)
  * ose-piping (Disabled)
  * ose-workbench-core
  * pcb 6.2023.1 (Disabled)
  * persistenttoolbars (Disabled)
  * PieMenu 1.2.7 (Disabled)
  * POV-Ray-Rendering
  * PVPlant
  * Pyramids-and-Polyhedrons
  * QuickMeasure 2022.10.28
  * Reinforcement (Disabled)
  * Render 2024.11.22
  * Reporting
  * retr3d (Disabled)
  * SearchBar 1.0.2
  * sheetmetal 0.6.1
  * Ship 2024.11.26
  * Silk 0.1.5 (Disabled)
  * slic3r-tools (Disabled)
  * SteelColumn (Disabled)
  * symbols_library
  * TitleBlock 0.5.2.2
  * toSketch 1.0.1
  * Trails 2022.1.0 (Disabled)
  * WebTools 1.0.0
  * woodworking 0.21.2.33771.0.23.1
  * workfeature
  * workfeature-macro
  * yaml-workspace 0.1.2
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-22:
Please attach file

PabloBeranger, 2024-12-23:
Attached is the zip file of my test project tree and the python script.
In addition I noticed that the longest path accepted by FreeCAD to save a document is 223 characters.
[ACA_FC_Test_Path_Length.zip](https://github.com/user-attachments/files/18231331/ACA_FC_Test_Path_Length.zip)
[Structure_V01_Materiaux_AAAAAA__Couleur_Magenta.zip](https://github.com/user-attachments/files/18231332/Structure_V01_Materiaux_AAAAAA__Couleur_Magenta.zip)

xtemp09, 2024-12-23:
@PabloBeranger, this is a built-in limitation.

https://doc.qt.io/qt-6/windows-issues.html#maximum-path-length

PabloBeranger, 2024-12-24:
I thought the apps use the Windows 10 standard.
https://learn.microsoft.com/en-us/windows/win32/fileio/maximum-file-path-limitation?tabs=registry