FreeCAD/FreeCAD #18667: Core datums - attachment to partial cylinders

mno13, 2024-12-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

While trying to attach datum origin  to partial cylinder, datum origin is not in cylinder's axis.

While trying to attach datum line to partial cylinder, datum line is normal to face, not in cylinder's axis.

Line:
[image]

Axis:
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * 3DfindIT 1.2.0
  * Curves 0.6.51
  * Defeaturing 1.2.2
  * dodo 1.0.1
  * fasteners 0.5.29
  * FrameForge 0.1.0
  * freecad.gears 1.3.0
  * MeshRemodel 1.10.32
  * Nodes 0.1.36
  * PieMenu 1.9.2
  * sheetmetal 0.5.3
  * Silk 0.1.5
  * ThreadProfile 1.92.0
  * yaml-workspace 0.1.2
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-22:
Please attach file with simple example and write steps to reproduce

mno13, 2024-12-22:
[image]

steps to reproduce:
- select any partial cylindrical face (half-bore for example) 
- select create coordinate system / add datum line
- effect: datum elements are not in the cylinder axis (afaik should be, when there is single, cylindrical surface as reference).

mno13, 2024-12-22:
[FC_PARTIAL_CYLINDERS_FAILED_DATUMS.zip](https://github.com/user-attachments/files/18221547/FC_PARTIAL_CYLINDERS_FAILED_DATUMS.zip)