FreeCAD/FreeCAD #18671: [TNP] TNP issues with external workbenches (maybe not only there)

shaise, 2024-12-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As far as I understood, The TNP mitigation fix on FreeCAD was supposed to be transparent to external workbenches, as long as the use supported link properties such as PropertyLinkSub.
However I get quite some TNP issues with the SheetMetal workbench (also reported by other uses)
I know that TNP Mitigation does not solve all possible scenarios, But I made the same test on FreeCAD Link and it seems to work there. So it seems there are still some missing code to migrate.
Unfortunately @bgbsww is no longer with us, his work was great. But maybe its a small thing and someone else can help.

Here is a small video depicting this issue:
https://youtu.be/bSYNlRJ5kqQ

Also, attached the files used in Main and in Link version    

[test-tnp2.zip](https://github.com/user-attachments/files/18221535/test-tnp2.zip)

Thanks!

### Full version info

```shell
OS: Windows 11 build 22631
Architecture: x86_64
Version: 1.1.0dev.39708 (Git) Conda
Build type: Release
Branch: main
Hash: 173ff4da1ef4f075d56c0dc617d1bf00d5f49d4b
Python 3.11.11, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * fasteners 0.5.33
  * freecad.gears 1.3.0 (Disabled)
  * kicadStepUpMod 11.4.1
  * MakerWorkbench 1.0.1 (Disabled)
  * Manipulator 1.5.7 (Disabled)
  * OpenTheme 2024.9.1
  * PieMenu 1.9.5 (Disabled)
  * Render 2024.9.25
  * SelectorToolbar (Disabled)
  * sheetmetal 0.6.11
  * sheetmetal-todo.txt
  * Ship 2024.11.26 (Disabled)
  * STEMFIE 0.2.4 (Disabled)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-12-22:
@CalligaroV fyi

CalligaroV, 2024-12-29:
@maxwxyz thanks for the ping!

@shaise I can confirm the issue: main FreeCAD shows the behavior you described while LS3 doesn't as in your video.

However I noticed something interesting: if I try to reproduce this issue, using main FreeCAD, with your file `test-tnp2-link.FCStd` then main behaves like LS3.

Tested:
```ruby
OS: Linux Mint 22 (X-Cinnamon/cinnamon/xcb)
Architecture: x86_64
Version: 1.1.0dev.39642 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: c80bb958f377e4d43c6a4075da173a79d153db68
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Fusion
Installed mods: 
  * sheetmetal 0.6.13
  * Beltrami 1.1.0
  * kicadStepUpMod 11.4.4
  * OpenTheme 2024.9.1
  * fasteners 0.5.34
  * Assembly4 0.50.15
  * Render 2024.12.15
  * Curves 0.6.53
  * freecad.gears 1.3.0
``` 
 
The behavior is the following:
[image]

This makes me think that this issue may be caused by differences, between main and LS3, in the code that saves the models.

For curiosity, which versions of FreeCAD have you used to design your models?
I guess
 * v1.0.0 or newer for `test-tnp2.FCStd`
 * LS3 for `test-tnp2-link.FCStd`

Is my guess correct?

shaise, 2024-12-29:
@CalligaroV ,
Yes! I Forgot to mention that but indeed loading the file saved on LS3 works on Main, which is even stranger...

CalligaroV, 2025-01-03:
@shaise thank you for confirming my hypothesis and for the additional info!

> Another thing I noticed is that the first SheetMetal feature after any PD feature is TNP correct, but the second one after that is not.

I can confirm that too:
 * using main, if I query for the mapped name of the element used for `Bend` (`Extend [Edge29]`) I get `'Edge26;:H,E'`
 * using LS3, if I perform the same query I get `'g1;SKT;:H11bd,E;:G;XTR;:H11bd:7,F;:U;XTR;:H11bd:7,E;:H-11bf,E;:G;SMW;:H11bf:7,F;:U3;SMW;:H11bf:8,E;:G;XTR;:H11bf:7,F;:U;XTR;:H11bf:7,E;:H,E'`

Therefor yes, looks like some code is missing for `PropertyLinkSub` or `PropertyLinkBase` and derived classes.

I'll analyze more in detail in the next days/weeks but, for my experience so far about finding the missing code and import what's needed to fix the issue, ATM I can't say how long it will take

shaise, 2025-01-03:
@CalligaroV ,
Thanks a lot! and take your time.
Also, Can you please tell me how can I query the mapped element in python? 
shai

CalligaroV, 2025-01-05:
You're welcome @shaise and thanks for your patience!

Regarding
> Can you please tell me how can I query the mapped element in python?

Usually I execute the following code in the Python Console:

```python
>>> App.ActiveDocument.UseHasher = False
>>> App.ActiveDocument.recompute()
>>> App.ActiveDocument.Feature.Shape.ElementReverseMap['indexedName']
``` 
where `Feature` is the feature you want to analyze and `indexedName` is the element related to the mapped name you want to get.
For example, in my previous https://github.com/FreeCAD/FreeCAD/issues/18671#issuecomment-2569235055, the last command is

```python
>>> App.ActiveDocument.Extend.Shape.ElementReverseMap['Edge29']
``` 

The following code
```python
>>> App.ActiveDocument.UseHasher = False
>>> App.ActiveDocument.recompute()
```
is used to get the complete mapped name of an element, otherwise you would get shorter, hashed, names

You can also query the entire element map of a `Feature` simply removing `['indexedName']` after `ElementReverseMap` (it returns a Python Dict)

Please let me know if you need other info or if you have issues querying the mapped names.

In the meantime I'll continue with the analysis of the missing code.

TIA!

shaise, 2025-01-05:
@CalligaroV , Thanks! this is extremely helpful.

belinski-roman, 2025-07-19:
We are looking forward to it

shaise, 2025-07-20:
Thanks! I had a feeling there is something more to be done within the workbench to make it compatible with the TNP fix.

shaise, 2025-08-21:
Hurray! thanks!

shaise, 2025-10-14:
using windows with this build:  FreeCAD_weekly-2025.10.13-Windows-x86_64-py311.7z 
what do you mean a compound with part?

drwho495, 2025-10-14:
@shaise [here](https://github.com/drwho495/FreeCAD/releases/tag/weekly-2025.10.14) is the new build.

drwho495, 2025-10-15:
> Testing the new version.
> Interesting: no more error on the first box made, but get an error as soon as I add a pad:
> ```
> >>> FreeCAD.ActiveDocument.Box.Shape.Faces
> [<Face object at 000002376617D3F0>, <Face object at 000002376617DFA0>, <Face object at 000002376617BB30>, <Face object at 000002376617E050>, <Face object at 000002376617E3C0>, <Face object at 000002376617D4A0>]
> >>> FreeCAD.ActiveDocument.Pad.Shape.Faces
> Traceback (most recent call last):
>   File "<input>", line 1, in <module>
> Base.FreeCADError: Unknown exception while reading attribute 'Faces' of object 'TopoShape'
> ```
> 
> [image]
> 
> file:
> [test-tnp5.zip](https://github.com/user-attachments/files/22921554/test-tnp5.zip)
> 

That means that this is definitely an issue with the mapping method that I call. I think what is happening is that the element map that c++ is trying to access is a nullptr and that's causing an access violation of something of the sort. Why this isn't happening on Linux is very odd though

maxwxyz, 2025-10-15:
@chennes maybe?

shaise, 2025-10-15:
I never had compiling issues before. so it must be something new.

drwho495, 2025-10-16:
> Here is a non-sheetmetal way to reproduce this error with a 2 line script:
> create a box inside a new body. 
> In the python console type:
> ```
> face = FreeCAD.ActiveDocument.Box.Shape.Faces[0]
> face.Edges
> ```
> 
> The second line will reproduce the error
> 

I was able to build the program on windows and now I see the error. I will try to get a fix out soon.

shaise, 2025-10-19:
@drwho495 , thanks, the fix seems to work.
Attached another sample similar to the first. when you move the rectangle inwards it generates an error:

[image]

```
09:56:39  pyException: Traceback (most recent call last):
  File "C:\Users\shai\AppData\Roaming\FreeCAD\v1-1\Mod\sheetmetal\.\SheetMetalCmd.py", line 1887, in execute
    thk, thkDir = sheet_thk(Main_Object, face[0])
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\shai\AppData\Roaming\FreeCAD\v1-1\Mod\sheetmetal\.\SheetMetalCmd.py", line 788, in sheet_thk
    selItem = MainObject.getElement(SheetMetalTools.getElementFromTNP(selFaceName))
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'ValueError'>: Invalid shape name ?Edge40
09:56:39  Bend001: Invalid shape name ?Edge40
```

[test-tnp6.zip](https://github.com/user-attachments/files/22988972/test-tnp6.zip)

Edit: it works under linkstage.

shaise, 2025-10-19:
Regarding LS3 I use a different file built from scratch as there is a big compatibility issue between both branches:

[test-tnp6-link.zip](https://github.com/user-attachments/files/22991869/test-tnp6-link.zip)

[image]
[image]
[image]
[image]

drwho495, 2025-10-19:
that model also works on main so i think there could be a difference in the models causing better stability

luzpaz, 2025-12-18:
soft bump

shaise, 2026-06-10:
I was waiting for the latest build with the #24632 fix. I will make some tests and get back.

ShamanTcler, 2026-10-07:
Let me chime in and say it is still an issue with 27.1 sheetmetal wb

drwho495, 2026-10-07:
> Let me chime in and say it is still an issue with 27.1 sheetmetal wb

In what way?

GS90, 2026-10-10:
Unfortunately, yes... All versions after 0.21 are still practically unusable for sheet-metal work.
Fixes are [available](https://github.com/FreeCAD/FreeCAD/pull/31040), and we’re eagerly waiting for them to be merged into the main branch.

shaise, 2026-11-10:
here is another sample file that works on realthunder's version but not main:

[BendToponamingDemo.zip](https://github.com/user-attachments/files/34620789/BendToponamingDemo.zip)

if you change the gap1 and gap2 properties of bend1, the hole jumps from the side to the bottom wall. On realthunder's version this is ok.

[image]